gazebosim/gz-sim #2610: KeyPress Plugin Can't Be Loaded from SDF File

akvavit01, 2024-09-10:
## Environment
* OS Version: Ubuntu 20.04.1 (noble) 
* Source or binary build? Binary build (gz gui 8.3.0, gz sim 8.6.0, gz sdf 14.5.0, gz plugin 2.0.3)
  
  
<details>

```
[Msg] Gazebo Sim GUI    v8.6.0
[Dbg] [gz.cc:166] Subscribing to [/gazebo/starting_world].
[Dbg] [gz.cc:168] Waiting for a world to be set from the GUI...
[Dbg] [Gui.cc:263] Waiting for subscribers to [/gazebo/starting_world]...
[Msg] Received world [test.sdf] from the GUI.
[Dbg] [gz.cc:172] Unsubscribing from [/gazebo/starting_world].
[Msg] Gazebo Sim Server v8.6.0
[Msg] Loading SDF world file[/home/leochen/ROS 2 Tutorial/gazebo_tutorial/test.sdf].
[Dbg] [Application.cc:96] Initializing application.
[Dbg] [Application.cc:170] Qt using OpenGL graphics interface
[GUI] [Dbg] [Application.cc:657] Create main window
[Msg] Serving entity system service on [/entity/system/add]
[Dbg] [Physics.cc:870] Loaded [gz::physics::dartsim::Plugin] from library [/opt/ros/jazzy/opt/gz_physics_vendor/lib/gz-physics-7/engine-plugins/libgz-physics-dartsim-plugin.so]
[Dbg] [SystemManager.cc:77] Loaded system [gz::sim::systems::Physics] for entity [1]
[Msg] Create service on [/world/Moving_robot/create]
[Msg] Remove service on [/world/Moving_robot/remove]
[Msg] Pose service on [/world/Moving_robot/set_pose]
[Msg] Pose service on [/world/Moving_robot/set_pose_vector]
[Msg] Light configuration service on [/world/Moving_robot/light_config]
[Msg] Physics service on [/world/Moving_robot/set_physics]
[Msg] SphericalCoordinates service on [/world/Moving_robot/set_spherical_coordinates]
[Msg] Enable collision service on [/world/Moving_robot/enable_collision]
[Msg] Disable collision service on [/world/Moving_robot/disable_collision]
[Msg] Material service on [/world/Moving_robot/visual_config]
[Msg] Material service on [/world/Moving_robot/wheel_slip]
[Dbg] [SystemManager.cc:77] Loaded system [gz::sim::systems::UserCommands] for entity [1]
[Dbg] [SystemManager.cc:77] Loaded system [gz::sim::systems::SceneBroadcaster] for entity [1]
[Dbg] [TriggeredPublisher.cc:689] TriggeredPublisher subscribed on /keyboard/keypress and publishing on /cmd_vel, 
[Dbg] [SystemManager.cc:77] Loaded system [gz::sim::systems::TriggeredPublisher] for entity [1]
[Dbg] [TriggeredPublisher.cc:689] TriggeredPublisher subscribed on /keyboard/keypress and publishing on /cmd_vel, 
[Dbg] [SystemManager.cc:77] Loaded system [gz::sim::systems::TriggeredPublisher] for entity [1]
[Dbg] [TriggeredPublisher.cc:689] TriggeredPublisher subscribed on /keyboard/keypress and publishing on /cmd_vel, 
[Dbg] [SystemManager.cc:77] Loaded system [gz::sim::systems::TriggeredPublisher] for entity [1]
[Dbg] [TriggeredPublisher.cc:689] TriggeredPublisher subscribed on /keyboard/keypress and publishing on /cmd_vel, 
[Dbg] [SystemManager.cc:77] Loaded system [gz::sim::systems::TriggeredPublisher] for entity [1]
[Err] [SystemLoader.cc:92] Failed to load system plugin [KeyPublisher] : Could not find shared library.
[Msg] DiffDrive subscribing to twist messages on [cmd_vel]
[Dbg] [SystemManager.cc:77] Loaded system [gz::sim::systems::DiffDrive] for entity [8]
[Msg] Loaded level [3]
[Msg] Serving world controls on [/world/Moving_robot/control], [/world/Moving_robot/control/state] and [/world/Moving_robot/playback/control]
[Msg] Serving GUI information on [/world/Moving_robot/gui/info]
[Msg] World [Moving_robot] initialized with [1ms] physics profile.
[Msg] Serving world SDF generation service on [/world/Moving_robot/generate_world_sdf]
[Msg] Serving world names on [/gazebo/worlds]
[Msg] Resource path add service on [/gazebo/resource_paths/add].
[Msg] Resource path get service on [/gazebo/resource_paths/get].
[Msg] Resource path resolve service on [/gazebo/resource_paths/resolve].
[Msg] Resource paths published on [/gazebo/resource_paths].
[Msg] Server control service on [/server_control].
[GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:413:17: QML ToolButton: Binding loop detected for property "implicitHeight"
[GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:309:21: QML Button: Binding loop detected for property "implicitHeight"
[GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:309:21: QML Button: Binding loop detected for property "implicitHeight"
[GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:413:17: QML ToolButton: Binding loop detected for property "implicitHeight"
[GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:309:21: QML Button: Binding loop detected for property "implicitHeight"
[GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:309:21: QML Button: Binding loop detected for property "implicitHeight"
[GUI] [Dbg] [PathManager.cc:68] Requesting resource paths through [/gazebo/resource_paths/get]
[GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:413:17: QML ToolButton: Binding loop detected for property "implicitHeight"
[GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:309:21: QML Button: Binding loop detected for property "implicitHeight"
[GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:309:21: QML Button: Binding loop detected for property "implicitHeight"
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [PathManager.cc:57] Received resource paths.
[GUI] [Dbg] [Gui.cc:401] Requesting GUI from [/world/Moving_robot/gui/info]...
[... 145 lines cut ...]
[GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model
[GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model
[GUI] [Wrn] [Component.hh:189] Trying to deserialize component with data type [N3sdf3v145WorldE], which doesn't have `operator>>`. Component will not be deserialized.
[GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model
[GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model
```

</details>

## Description
* Expected behavior: KeyPress plugin loaded when running gz sim with SDF 
* Actual behavior: KeyPress plugin is only able to be loaded via Gazebo GUI 

Basically, I followed [this](https://gazebosim.org/docs/latest/moving_robot/) tutorial, but I am unable to move my robot. Then, I try to add Key Publisher plugin via Gazebo GUI, and it works!

Then, I saved the world with Key Publisher plugin into an SDF file named `test.sdf.` After saving it, I load the SDF file with `gz sim test.sdf -v 4`. Then, the log clearly shows that KeyPublisher plugin can't be loaded.

## Steps to reproduce

1. Make a new SDF file from [this](https://justpaste.it/cgef5) and rename it `test.sdf`
2. Run `gz sim test.sdf -v 4`

## Output

[image]

akvavit01, 2024-09-10:
Hello.
Please let me know on how I can assist fixing this bug.

iche033, 2024-09-11:
the gui plugins are not saved when you save a world. The gui plugins can be saved using the `Save the client configuration as` option in the top left menu. Supposedly, users should be able to load the gui config back but I just ran into an issue with 3D scene not working when I tried this out. Ticketed https://github.com/gazebosim/gz-sim/issues/2617

akvavit01, 2024-09-27:
Fixed it. Turns out that I need to put it inside <gui> tag. Thanks

closed 2024-09-27 by akvavit01