gazebosim/gz-sim #2614: [QT] Failed to create OpenGL context for format QSurfaceFormat

Roboticistprogrammer, 2024-09-10:
## Environment
* OS Version: Ubuntu ionic
* Source or binary build? Source Installation on Docker
 

## Description
* Expected behavior: I have successfully build the packages (although I have to admit it requires huge RAM, otherwise it doesn't work even if use --parallel-works 1 flag). I sourced my environment and run "gz sim shapes.sdf" to start gazebo.
* Actual behavior: I get: [QT] Failed to create OpenGL context for format QSurfaceFormat

## Steps to reproduce

1.Build the Pakcage.
2.Source the environment
3.gz sim shapes.sdf

## Output
[image]

azeey, 2024-11-11:
@Roboticistprogrammer any updates?

azeey, 2024-11-11:
Might be worth looking at https://github.com/gazebosim/gz-sim/issues/2595 as well.