gazebosim/gz-sim #2620: Windows installation issue

avanmalleghem, 2024-09-12:
## Environment
* OS Version: Windows Server 2022 (I know it isn't a common setup but the purpose is to help on gazebo test cases for the tutorial party)
* Source or binary build?
  Source following https://gazebosim.org/docs/ionic/install_windows_src/
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: ogre2. 
      - [ ] Sensor rendering error. 
      - [x] GUI rendering error. 
    - EGL headless mode:
      - [ ] Running in EGL headless mode 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [ ] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [ ] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [ ] running on real hardware
      - [x] running in virtual machine
      - [ ] running in Docker/Singularity
      - [ ] running remotely (e.g. via SSH)
      - [x] running in a cloud
      - [ ] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
      - [ ] GPU is concurrently used for other tasks
        - [ ] desktop acceleration
        - [ ] video decoding (i.e. a playing Youtube video)
        - [ ] video encoding
        - [ ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [ ] multiple simulators running at the same time
      - [ ] other...
    - Rendering system info:
      - On Windows, run `dxdiag` and report the GPU-related information.
```
Name: NVIDIA Tesla P100-PCIE-16GB
Device ID: PCI\VEN_10DE&DEV_15F8&SUBSYS_118F10DE&REV_A1\3&13C0B0C5&0&20
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\NVWMI\nvPerfProvider.man, 2/22/2023 23:41:26, 14175 bytes
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\NVWMI\nvWmi.mof, 2/22/2023 23:41:26, 148921 bytes
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\NVWMI\nvWmi64.exe, 2.36.0000.0000 (English), 2/23/2023 00:42:28, 4511840 bytes
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\Display.NvContainer\MessageBus.dll, 1.22.2758.1620 (English), 2/23/2023 00:41:18, 7543184 bytes
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\Display.NvContainer\NVDisplay.Container.exe, 1.37.3103.4323 (English), 2/23/2023 00:41:20, 1015384 bytes
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\Display.NvContainer\NvContainerRecovery.bat, 2/22/2023 23:41:26, 1951 bytes
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\Display.NvContainer\dlsargs.xml, 2/22/2023 23:41:26, 544 bytes
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\Display.NvContainer\dlsnetparams.csv, 2/22/2023 23:41:26, 341661 bytes
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\Display.NvContainer\messagebus_client.conf, 2/22/2023 23:41:26, 57 bytes
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\Display.NvContainer\nvgwls.exe, 2/23/2023 00:41:22, 3941512 bytes
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\Display.NvContainer\nvtopps.db3, 2/22/2023 23:41:26, 126976 bytes
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\Display.NvContainer\plugins\LocalSystem\NvXDCore.dll, 31.00.0015.3114 (English), 2/23/2023 00:41:30, 1819272 bytes
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\Display.NvContainer\plugins\LocalSystem\NvcDispCorePlugin.dll, 31.00.0015.3114 (English), 2/23/2023 00:41:24, 523400 bytes
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\Display.NvContainer\plugins\LocalSystem\NvcDispWatchdog.dll, 1.37.3103.4323 (English), 2/23/2023 00:41:30, 739440 bytes
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\Display.NvContainer\plugins\LocalSystem\_DisplayDriverRAS.dll, 1.10.0000.0000 (English), 2/23/2023 00:41:36, 2714224 bytes
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\Display.NvContainer\plugins\LocalSystem\_NvMsgBusBroadcast.dll, 1.34.3030.7627 (English), 2/23/2023 00:41:38, 3192968 bytes
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\Display.NvContainer\plugins\LocalSystem\messagebus.conf, 2/22/2023 23:41:26, 57 bytes
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\Display.NvContainer\plugins\LocalSystem\wksServicePluginZ.dll, 31.00.0015.3114 (English), 2/23/2023 00:41:34, 267864 bytes
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\Display.NvContainer\plugins\Session\_NvGSTPlugin.dll, 31.00.0015.3114 (English), 2/23/2023 00:41:44, 2654832 bytes
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\Display.NvContainer\plugins\Session\_nvtopps.dll, 31.00.0015.3114 (English), 2/23/2023 00:41:46, 11936920 bytes
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\Display.NvContainer\plugins\Session\nvprofileupdaterplugin.dll, 31.00.0015.3114 (English), 2/23/2023 00:41:38, 2230920 bytes
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\Display.NvContainer\plugins\Session\nvxdsyncplugin.dll, 8.17.0015.3114 (English), 2/23/2023 00:41:40, 1830040 bytes
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\Display.NvContainer\plugins\Session\wksServicePlugin.dll, 31.00.0015.3114 (English), 2/23/2023 00:41:42, 309912 bytes
   Driver: C:\Windows\system32\DRIVERS\NVIDIA Corporation\Drs\dbInstaller.exe, 31.00.0015.3114 (English), 2/23/2023 00:34:10, 733320 bytes
   Driver: C:\Windows\system32\DRIVERS\NVIDIA Corporation\Drs\nvdrsdb.bin, 2/22/2023 23:41:26, 1893060 bytes
   Driver: C:\Windows\system32\nvcpl.dll, 8.17.0015.3114 (English), 2/23/2023 00:34:50, 5835888 bytes
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\nv3dappshext.dll, 6.14.0015.3114 (English), 2/23/2023 00:34:14, 1005720 bytes
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\nv3dappshextr.dll, 6.14.0015.3114 (English), 2/23/2023 00:34:16, 98968 bytes
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\nvcoproc.bin, 2/22/2023 23:41:26, 10321217 bytes
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\nvcpl.dll, 8.17.0015.3114 (English), 2/23/2023 00:34:50, 5835888 bytes
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\nvdevtools.dll, 31.00.0015.3114 (English), 2/23/2023 00:35:36, 4585096 bytes
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\nvdevtoolsr.dll, 31.00.0015.3114 (English), 2/23/2023 00:35:38, 245896 bytes
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\nvdisps.dll, 31.00.0015.3114 (English), 2/23/2023 00:35:42, 11649688 bytes
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\nvdispsr.dll, 31.00.0015.3114 (English), 2/23/2023 00:35:44, 11552408 bytes
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\nvgames.dll, 31.00.0015.3114 (English), 2/23/2023 00:36:24, 12479600 bytes
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\nvgamesr.dll, 31.00.0015.3114 (English), 2/23/2023 00:36:26, 13252720 bytes
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\nvlicensings.dll, 6.14.0015.3114 (English), 2/23/2023 00:36:52, 4551832 bytes
   Driver: C:\Windows\System32\DriverStore\FileRepository\nv_dispui.inf_amd64_c3352d3df1cf4d8c\nvlicensingsr.dll, 6.14.0015.3114 (English), 2/23/2023 00:37:00, 307296 bytes
[... cut ...]
14:45:10: Particle Affector Type 'Rotator' registered
14:45:10: Particle Affector Type 'DirectionRandomiser' registered
14:45:10: Particle Affector Type 'DeflectorPlane' registered
14:45:10: Plugin successfully installed
```

</details>

## Description
* Expected behavior: I would like to see the Gazebo Ionic GUI
* Actual behavior: The GUI open then close directly because of the issue.

## Steps to reproduce

1. Installation following https://gazebosim.org/docs/ionic/install_windows_src/
2. Crash at step "launch gui"

## Output

[image]

avanmalleghem, 2024-09-12:
I think the issue is related to OpenGL version. I found "Google's P100 provides OpenGL... but V1.1 only.". Will try with another GPU...

traversaro, 2024-09-12:
Indeed the relevant error in the ogre.log is: 

~~~
14:45:10: OGRE EXCEPTION(3:RenderingAPIException): WGL_ARB_create_context extension required for OpenGL 3.3. Update your graphics card driver, or your card doesn't support OpenGL 3.3 in Win32GLSupport::initialiseWGL at D:\bld\ogre-next_1724198948908\work\RenderSystems\GL3Plus\src\windowing\win32\OgreWin32GLSupport.cpp (line 628) 14:45:10: Loading library C:\Users\gazebouser\miniconda3\envs\gz-ws\Library\bin\OGRE-Next\Plugin_ParticleFX.dll 14:45:10: Installing plugin: ParticleFx
~~~