gazebosim/gz-sim #2624: Crash after adding model with detachable_joint and resetting

zhileiren, 2024-09-15:
## Environment
* OS Version: Ubuntu 24.04
* Source or binary build?
    source build
    gz-sim9 version: 3765579cfc017de813a5780c4dda2a39d979b6d4
    built with 
    gcc version 13.2.0 (Ubuntu 13.2.0-23ubuntu4)
    build options: -DCMAKE_BUILD_TYPE=Coverage
    also reproduced on gz-sim8 version: a1673123c49a39ae4622d64696c054a8574738f9

## Description
* Expected behavior: Gazebo doesn't crash
* Actual behavior: Gazebo crashes

## Steps to reproduce
1. `gz sim examples/worlds/detachable_joint.sdf -r &`
2. call /world/detachable_joint/create to add model extracted from the detachable_joint example

```bash
gz service --timeout 10000 -s /world/detachable_joint/create --reptype gz.msgs.Boolean --reqtype gz.msgs.EntityFactory --req 'sdf: "<sdf version=\"1.11\"><model name=\"vehicle_blue\">\n      <pose>0 2 0.325 0 -0 0</pose>\n      <self_collide>true</self_collide>\n\n      <link name=\"chassis\">\n        <pose>-0.151427 -0 0.175 0 -0 0</pose>\n        <inertial>\n          <mass>1.14395</mass>\n          <inertia>\n            <ixx>0.126164</ixx>\n            <ixy>0</ixy>\n            <ixz>0</ixz>\n            <iyy>0.416519</iyy>\n            <iyz>0</iyz>\n            <izz>0.481014</izz>\n          </inertia>\n        </inertial>\n        <visual name=\"visual\">\n          <geometry>\n            <box>\n              <size>2.01142 1 0.568726</size>\n            </box>\n          </geometry>\n          <material>\n            <ambient>0.5 0.5 1.0 1</ambient>\n            <diffuse>0.5 0.5 1.0 1</diffuse>\n            <specular>0.0 0.0 1.0 1</specular>\n          </material>\n        </visual>\n        <collision name=\"collision\">\n          <geometry>\n            <box>\n              <size>2.01142 1 0.568726</size>\n            </box>\n          </geometry>\n        </collision>\n      </link>\n\n      <link name=\"front_left_wheel\">\n        <pose>0.554283 0.625029 -0.025 -1.5707 0 0</pose>\n        <inertial>\n          <mass>2</mass>\n          <inertia>\n            <ixx>0.145833</ixx>\n            <ixy>0</ixy>\n            <ixz>0</ixz>\n            <iyy>0.145833</iyy>\n            <iyz>0</iyz>\n            <izz>0.125</izz>\n          </inertia>\n        </inertial>\n        <visual name=\"visual\">\n          <geometry>\n            <sphere>\n              <radius>0.3</radius>\n            </sphere>\n          </geometry>\n          <material>\n            <ambient>0.2 0.2 0.2 1</ambient>\n            <diffuse>0.2 0.2 0.2 1</diffuse>\n            <specular>0.2 0.2 0.2 1</specular>\n          </material>\n        </visual>\n        <collision name=\"collision\">\n          <geometry>\n            <sphere>\n              <radius>0.3</radius>\n            </sphere>\n          </geometry>\n          <surface>\n            <friction>\n              <ode>\n                <mu>1</mu>\n                <mu2>1</mu2>\n                <slip1>0.035</slip1>\n                <slip2>0</slip2>\n                <fdir1>0 0 1</fdir1>\n              </ode>\n              <bullet>\n                <friction>1</friction>\n                <friction2>1</friction2>\n                <rolling_friction>0.1</rolling_friction>\n              </bullet>\n            </friction>\n          </surface>\n        </collision>\n      </link>\n\n      <link name=\"rear_left_wheel\">\n        <pose>-0.957138 0.625029 -0.025 -1.5707 0 0</pose>\n        <inertial>\n          <mass>2</mass>\n          <inertia>\n            <ixx>0.145833</ixx>\n            <ixy>0</ixy>\n            <ixz>0</ixz>\n            <iyy>0.145833</iyy>\n            <iyz>0</iyz>\n            <izz>0.125</izz>\n          </inertia>\n        </inertial>\n        <visual name=\"visual\">\n          <geometry>\n            <sphere>\n              <radius>0.3</radius>\n            </sphere>\n          </geometry>\n          <material>\n            <ambient>0.2 0.2 0.2 1</ambient>\n            <diffuse>0.2 0.2 0.2 1</diffuse>\n            <specular>0.2 0.2 0.2 1</specular>\n          </material>\n        </visual>\n        <collision name=\"collision\">\n          <geometry>\n            <sphere>\n              <radius>0.3</radius>\n            </sphere>\n          </geometry>\n          <surface>\n            <friction>\n              <ode>\n                <mu>1</mu>\n                <mu2>1</mu2>\n                <slip1>0.035</slip1>\n                <slip2>0</slip2>\n                <fdir1>0 0 1</fdir1>\n              </ode>\n              <bullet>\n                <friction>1</friction>\n                <friction2>1</friction2>\n                <rolling_friction>0.1</rolling_friction>\n              </bullet>\n            </friction>\n          </surface>\n        </collision>\n      </link>\n\n      <link name=\"front_right_wheel\">\n        <pose>0.554282 -0.625029 -0.025 -1.5707 0 0</pose>\n        <inertial>\n          <mass>2</mass>\n          <inertia>\n            <ixx>0.145833</ixx>\n            <ixy>0</ixy>\n            <ixz>0</ixz>\n            <iyy>0.145833</iyy>\n            <iyz>0</iyz>\n            <izz>0.125</izz>\n          </inertia>\n        </inertial>\n        <visual name=\"visual\">\n          <geometry>\n            <sphere>\n              <radius>0.3</radius>\n            </sphere>\n          </geometry>\n          <material>\n            <ambient>0.2 0.2 0.2 1</ambient>\n            <diffuse>0.2 0.2 0.2 1</diffuse>\n            <specular>0.2 0.2 0.2 1</specular>\n          </material>\n        </visual>\n        <collision name=\"collision\">\n          <geometry>\n            <sphere>\n              <radius>0.3</radius>\n            </sphere>\n          </geometry>\n          <surface>\n            <friction>\n              <ode>\n                <mu>1</mu>\n                <mu2>1</mu2>\n                <slip1>0.035</slip1>\n                <slip2>0</slip2>\n                <fdir1>0 0 1</fdir1>\n              </ode>\n              <bullet>\n                <friction>1</friction>\n                <friction2>1</friction2>\n                <rolling_friction>0.1</rolling_friction>\n              </bullet>\n            </friction>\n          </surface>\n        </collision>\n      </link>\n\n      <link name=\"rear_right_wheel\">\n        <pose>-0.957138 -0.625029 -0.025 -1.5707 0 0</pose>\n        <inertial>\n          <mass>2</mass>\n          <inertia>\n            <ixx>0.145833</ixx>\n            <ixy>0</ixy>\n            <ixz>0</ixz>\n            <iyy>0.145833</iyy>\n            <iyz>0</iyz>\n            <izz>0.125</izz>\n          </inertia>\n        </inertial>\n        <visual name=\"visual\">\n          <geometry>\n            <sphere>\n              <radius>0.3</radius>\n            </sphere>\n          </geometry>\n          <material>\n            <ambient>0.2 0.2 0.2 1</ambient>\n            <diffuse>0.2 0.2 0.2 1</diffuse>\n            <specular>0.2 0.2 0.2 1</specular>\n          </material>\n        </visual>\n        <collision name=\"collision\">\n          <geometry>\n            <sphere>\n              <radius>0.3</radius>\n            </sphere>\n          </geometry>\n          <surface>\n            <friction>\n              <ode>\n                <mu>1</mu>\n                <mu2>1</mu2>\n                <slip1>0.035</slip1>\n                <slip2>0</slip2>\n                <fdir1>0 0 1</fdir1>\n              </ode>\n              <bullet>\n                <friction>1</friction>\n                <friction2>1</friction2>\n                <rolling_friction>0.1</rolling_friction>\n              </bullet>\n            </friction>\n          </surface>\n        </collision>\n      </link>\n\n\n      <joint name=\"front_left_wheel_joint\" type=\"revolute\">\n        <parent>chassis</parent>\n        <child>front_left_wheel</child>\n        <axis>\n          <xyz>0 0 1</xyz>\n          <limit>\n            <lower>-1.79769e+308</lower>\n            <upper>1.79769e+308</upper>\n          </limit>\n        </axis>\n      </joint>\n\n      <joint name=\"front_right_wheel_joint\" type=\"revolute\">\n        <parent>chassis</parent>\n        <child>front_right_wheel</child>\n        <axis>\n          <xyz>0 0 1</xyz>\n          <limit>\n            <lower>-1.79769e+308</lower>\n            <upper>1.79769e+308</upper>\n          </limit>\n        </axis>\n      </joint>\n\n      <joint name=\"rear_left_wheel_joint\" type=\"revolute\">\n        <parent>chassis</parent>\n        <child>rear_left_wheel</child>\n        <axis>\n          <xyz>0 0 1</xyz>\n          <limit>\n    [... cut ...]
#5    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x731058ea5a48, in std::terminate()
#4    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x731058ebae9b, in 
#3    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x731058ea5ffd, in 
#2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x73105e4288fe, in abort
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x73105e44526d, in gsignal
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x73105e49eb1c, in pthread_kill
Aborted (Signal sent by tkill() 22573 1000)
```

zhileiren, 2024-09-16:
Steps to reproduce could be simplified as:
1. `gz sim examples/worlds/detachable_joint.sdf`
2. remove the vehicle_blue model from GUI
3. hit the reset button from GUI

RPS98, 2024-10-01:
I have experienced the same issue. I noticed the strange behavior that it works with two models inside the SDF being spawned, but not with three.

## Environment
* OS Version: Ubuntu 22.04
* Binary: Gazebo Harmonic v8.6.0

## Description
* Expected behavior: Been able to reset Gazebo
* Actual behavior: Gazebo crash

## Steps to reproduce
Using the empty.sdf world:

1. Open Gazebo using the empty.sdf world

```bash
gazebo empty.sdf
```

2. Call the gazebo spawn service to spawn a model

```bash
gz service -s /world/empty/create --reqtype gz.msgs.EntityFactory --reptype gz.msgs.Boolean --timeout 1000 --req 'sdf_filename: "/home/rafa/test.sdf", name: "my_object"'
```

Using this test.sdf model:
```xml
<?xml version="1.0" ?>

<sdf version="1.6">
  <model name='box0'>
    <pose>0 0 0 0 0 0</pose>
    <link name="base_link">
      <visual name="visual">
        <geometry>
          <box>
            <size>0.2 0.2 0.2</size>
          </box>
        </geometry>
      </visual>
    </link>

    <model name="box1">
      <pose>0.3 0 0.5 0 0 0</pose>
      <link name="link1"></link>
    </model>

    <joint name="box1_joint" type="fixed">
      <parent>base_link</parent>
      <child>box1::link1</child>
    </joint>

    <model name="box2">
      <pose>0 -0.3 0.5 0 0 0</pose>
      <link name="link2"></link>
    </model>

    <joint name="box2_joint" type="fixed">
      <parent>base_link</parent>
      <child>box2::link2</child>
    </joint>

    <model name="box3">
      <pose>0 0 0.7 0 0 0</pose>
      <link name="link3"></link>
    </model>

    <joint name="box3_joint" type="fixed">
      <parent>base_link</parent>
      <child>box3::link3</child>
[... 67 lines cut ...]
#22   Object "/usr/lib/x86_64-linux-gnu/ruby/3.0.0/fiddle.so", at 0x7a52b1ea4d6b, in 
#21   Object "/lib/x86_64-linux-gnu/libffi.so.8", at 0x7a52b1e96492, in 
#20   Object "/lib/x86_64-linux-gnu/libffi.so.8", at 0x7a52b1e99e2d, in 
#19   Object "/usr/lib/x86_64-linux-gnu/libgz-sim8-gz.so.8.6.0", at 0x7a52b1ab3605, in runServer
#18   Object "/lib/x86_64-linux-gnu/libgz-sim8.so.8", at 0x7a52ac53d485, in 
#17   Object "/lib/x86_64-linux-gnu/libgz-sim8.so.8", at 0x7a52ac54d8a2, in gz::sim::v8::SimulationRunner::Run(unsigned long)
#16   Object "/lib/x86_64-linux-gnu/libgz-sim8.so.8", at 0x7a52ac54ce60, in gz::sim::v8::SimulationRunner::Step(gz::sim::v8::UpdateInfo const&)
#15   Object "/lib/x86_64-linux-gnu/libgz-sim8.so.8", at 0x7a52ac55fb16, in gz::sim::v8::SystemManager::Reset(gz::sim::v8::UpdateInfo const&, gz::sim::v8::EntityComponentManager&)
#14   Object "/usr/lib/x86_64-linux-gnu/gz-sim-8/plugins/libgz-sim-physics-system.so", at 0x7a52a2f20e15, in gz::sim::v8::systems::PhysicsPrivate::ResetPhysics(gz::sim::v8::EntityComponentManager&)
#13   Object "/usr/lib/x86_64-linux-gnu/gz-sim-8/plugins/libgz-sim-physics-system.so", at 0x7a52a2f19ad6, in gz::sim::v8::systems::PhysicsPrivate::RemovePhysicsEntities(gz::sim::v8::EntityComponentManager const&)
#12   Object "/usr/lib/x86_64-linux-gnu/gz-sim-8/plugins/libgz-sim-physics-system.so", at 0x7a52a2f8958c, in 
#11   Object "/usr/lib/x86_64-linux-gnu/gz-physics-7/engine-plugins/libgz-physics-dartsim-plugin.so", at 0x7a52761c76cf, in gz::physics::dartsim::EntityManagementFeatures::RemoveModel(gz::physics::Identity const&)
#10   Object "/usr/lib/x86_64-linux-gnu/gz-physics-7/engine-plugins/libgz-physics-dartsim-plugin.so", at 0x7a52761cec23, in gz::physics::dartsim::Base::RemoveModelImpl(unsigned long, unsigned long)
#9    Object "/usr/lib/x86_64-linux-gnu/gz-physics-7/engine-plugins/libgz-physics-dartsim-plugin.so", at 0x7a52761cebaf, in gz::physics::dartsim::Base::RemoveModelImpl(unsigned long, unsigned long)
#8    Object "/usr/lib/x86_64-linux-gnu/gz-physics-7/engine-plugins/libgz-physics-dartsim-plugin.so", at 0x7a52761cc94b, in std::__detail::_Map_base<unsigned long, std::pair<unsigned long const, std::shared_ptr<gz::physics::dartsim::ModelInfo> >, std::allocator<std::pair<unsigned long const, std::shared_ptr<gz::physics::dartsim::ModelInfo> > >, std::__detail::_Select1st, std::equal_to<unsigned long>, std::hash<unsigned long>, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<false, false, true>, true>::at(unsigned long const&)
#7    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7a52ad6a549f, in std::__throw_out_of_range(char const*)
#6    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7a52ad6ae4d7, in __cxa_throw
#5    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7a52ad6ae276, in std::terminate()
#4    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7a52ad6ae20b, in 
#3    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7a52ad6a2b9d, in 
#2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7a52b1c287f2, in abort
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7a52b1c42475, in raise
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7a52b1c969fc, in pthread_kill
Aborted (Signal sent by tkill() 17407 1000)
```

caguero, 2024-11-18:
According @iche033 , we need to implement some bookkeeping to undo the changes after combining the two models together.

PR #2686 (2024-11-26): Remove detachable joints when a model is removed

# 🦟 Bug fix

Partially fixes #2624

## Summary

Remove detachable joints from the ECM when a model is removed. The physics engine will then also remove any associated detachable joint physics objects. I also made a change in physics system to let physics engine remove the detachable joints first before models.

Now when you delete a model, the detachable joints should be deleted causing objects to be detached.

To test:

launch `detachable_joint.sdf` world

```
gz sim -v 4 -r ./src/gz-sim/examples/worlds/detachable_joint.sdf
```

Delete `vehicl [... cut]
closed 2024-11-26 by iche033