gazebosim/gz-sim #2637: High Thrust Calculation in Motor Model Plugin

Kametor, 2024-10-01:
## Environment
* OS Version: Ubuntu 22.04
* Gazebo Harmonic

## Description
* There may be a misunderstanding or misuse of the Gazebo motor model plugin parameters. Specifically, the “maxRotVelocity” parameter, which is described as the “Maximum rotational velocity command in units of rad/s”, seems to be leading to unintended results in the thrust calculation.
The thrust force is computed using the following equation:

thrust = realMotorVelocity * realMotorVelocity * this->motorConstant;
Here, the variable realMotorVelocity is defined as:

double realMotorVelocity = motorRotVel * this->rotorVelocitySlowdownSim;
Consequently, at full throttle, the resulting force calculation is:

(maxRotVelocity * 10)^2 * motorConstant.
Based on my calculations, the resulting force in Gazebo appears significantly higher than expected. I am uncertain if I am misinterpreting a specific aspect of the model or if there is an issue with the parameter setup. In my calculation the thrust force is very high. Also drone climbs very fast. I think there is a problem. Any guidance or insight into this discrepancy would be greatly appreciated.

Kametor, 2024-10-01:
As you can see my esc pwm values are near 250. My joystick is lower the half. But my drone climbs very fast. My custom parameters are:
    <plugin filename="gz-sim-multicopter-motor-model-system" name="gz::sim::systems::MulticopterMotorModel">
      <jointName>rotor_0_joint</jointName>
      <linkName>rotor_0</linkName>
      <turningDirection>ccw</turningDirection>
      <timeConstantUp>0.0125</timeConstantUp>
      <timeConstantDown>0.025</timeConstantDown>
      <maxRotVelocity>313.0</maxRotVelocity>
      <motorConstant>830.0e-06</motorConstant>
      <momentConstant>0.03</momentConstant>
      <commandSubTopic>command/motor_speed</commandSubTopic>
      <motorNumber>0</motorNumber>
      <rotorDragCoefficient>0.001</rotorDragCoefficient>
      <rollingMomentCoefficient>0.001</rollingMomentCoefficient>
      <rotorVelocitySlowdownSim>10</rotorVelocitySlowdownSim>
      <motorType>velocity</motorType>
    </plugin>
    Also my dron weight is 45 kg.
[image]

azeey, 2024-12-16:
This is my understanding:

## Joint velocity assignment 
1. Commanded rotor velocity is assigned to `refMotorInput` https://github.com/gazebosim/gz-sim/blob/83b30c2bfe8031977b536ef38851642e243f9bc6/src/systems/multicopter_motor_model/MulticopterMotorModel.cc#L528-L530
2. This velocity is filtered (`refMotorRotVel`) to fullfil the time constant parameters specified https://github.com/gazebosim/gz-sim/blob/83b30c2bfe8031977b536ef38851642e243f9bc6/src/systems/multicopter_motor_model/MulticopterMotorModel.cc#L673-L675
3. `refMotorRotVel/rotorVelocitySlowDownSim` is assigned as the actual commanded velocity on the joint https://github.com/gazebosim/gz-sim/blob/83b30c2bfe8031977b536ef38851642e243f9bc6/src/systems/multicopter_motor_model/MulticopterMotorModel.cc#L677-L680

## Thrust Calculation
1. The current joint velocity is obtained from the ECM and multiplied by `rotorVelocitySlowDownSim` to get the reference motor velocity (`realMotorVelocity`). In steady state, this should be the same as `refMotorInput`. In full throttle, this would be `motorRotVel`. https://github.com/gazebosim/gz-sim/blob/83b30c2bfe8031977b536ef38851642e243f9bc6/src/systems/multicopter_motor_model/MulticopterMotorModel.cc#L555-L565
2. Thrust is calculated as a constant times a signed `realMotorVelocity^2`. https://github.com/gazebosim/gz-sim/blob/83b30c2bfe8031977b536ef38851642e243f9bc6/src/systems/multicopter_motor_model/MulticopterMotorModel.cc#L570-L572

One difference in your analysis is 
> Consequently, at full throttle, the resulting force calculation is:

> (maxRotVelocity * 10)^2 * motorConstant.

I think it should be `(maxRotVelocity)^2 * motorConstant` since the velocity used in the thrust calculation is in the same order of magnitude as the input rotor velocity `refMotorInput`, i.e, `rotorVelocitySlowdownSim` cancels out when calculating `realMotorVelocity`.