gazebosim/gz-sim #2639: ROS2 Humble Gazebo Harmonic GUI is not responding

IsraaAbdelghany9, 2024-10-03:
I using Ubuntu 22.04 only Gazebo11 classic GUI is working neither ignition nor Hurmonic I have deleted everything related to the classic version to  see if it is the problem but nothing happend the GUI of the harmonic appear which you can choose whic sdf file you want and when I hit run it fails and print the GUI is not responding and if I run it from the terminal it stays a little bit printing that kind of message wait message then it prints also the GUI is not responding ....... I found someone talking about the firewall and locale setting but nothing , also adding that line to the .bashrc -> export GAZEBO_RESOURCE_PATH=/usr/share/gazebo-harmonic
but also nothing happeded do anyone have any idea what is the problem ... 

when I run the comand too:

gz service -s /gazebo/worlds --reqtype gz.msgs.Empty --reptype gz.msgs.StringMsg_V -r 'unused: true' --timeout 2000

it prints out service time out

azeey, 2024-10-03:
Have you tried following the networking step from the [troubleshooting](https://gazebosim.org/docs/harmonic/troubleshooting/#network-configuration-issue) guide?

azeey, 2024-12-16:
I'll close this since it seems like a setup problem. It's best to ask for help on https://robotics.stackexchange.com/. I encourage you to post your question there and if you find that there's a specific bug or feature request, come back and create a new issue here.

closed 2024-12-16 by azeey