gazebosim/gz-sim #2645: Build my own Lidar Sensor with custom scanning shape

nnln-lno, 2024-10-10:
Hi,

I don't have enough knowledge about rendering, modeling, etc.. 
But I tried my best to understand how gz sim Lidar emits ray from source code (e.g : Orge2GpuRays.cc, GpuRays.cc ..) and change some values to see what happens in the simulation.

As I understand, In this function (https://github.com/gazebosim/gzrendering/blob/c2e72ee51a7e4dba5156faa96c972c63ca5ab437/ogre2/src/Ogre2GpuRays.cc#L827-L950), gz makes 'cubeUVTexture'  using LiDAR spec in sdf parameter and mapping each direction vector to cube object. 

After I changed the value such as cubeUVTexture width and height, LiDAR ray showed a weird value than I expected.

Is there are instructions for creating custom ray shapes or understanding GpuRays ??

Thank you.

iche033, 2024-10-10:
> After I changed the value such as cubeUVTexture width and height, LiDAR ray showed a weird value than I expected.

We don't have documentation for creating custom ray shapes. Maybe you can describe what's the custom shape you're trying to create and we can see if it's possible to achieve it by making changes to that function

nnln-lno, 2024-10-12:
> After I changed the value such as cubeUVTexture width and height, LiDAR ray showed a weird value than I expected.
>
> We don't have documentation for creating custom ray shapes. Maybe you can describe [...]

@iche033  Thank you for reply my issue.

Currently, I want to make Livox Avia non-repetitive shape scan pattern with rotation each 0.1s

But In rendering perspective, I think, It sounds really hard ...

Nevertheless, thank you for doing your best in development.

iche033, 2024-10-14:
sounds like https://github.com/gazebosim/gz-sim/issues/1958. My answer would be the same as the comment: https://github.com/gazebosim/gz-sim/issues/1958#issuecomment-1499032276. Someone also posted in that issue another GPU based lidar implementation that you can check out.

closed 2024-12-16 by azeey