gazebosim/gz-sim #2650: Using body_link to move center of rotation in TrackedVehicle plugin

iandresolares, 2024-10-14:
## Environment
* OS Version: Ubuntu 22.04 
* Binary build: v8.6.0 (Gazebo Sim Harmonic)

## Description
Using [TrackedVehicle](https://gazebosim.org/api/gazebo/6/classignition_1_1gazebo_1_1systems_1_1TrackedVehicle.html) to simulate a tracked vehicle. I would like to modify the center of rotation to be aligned to a specific point on the robot instead of the default link. 
@peci1 suggested creating  a link for this and setting body_link tag in the plugin configuration but it is not working for me, vehicle is still turning in respect to the base_link. This is my current configuration. Note rotation_axis_link.
```   
    <plugin name="gz::sim::systems::TrackedVehicle" filename="gz-sim-tracked-vehicle-system">
        <body_link><link>rotation_axis_link</link></body_link>
        <left_track><link>left_track</link></left_track>
        <right_track><link>right_track</link></right_track>
        <tracks_separation>1.375</tracks_separation>
        <tracks_height>0.5</tracks_height>
        <steering_efficiency>0.5</steering_efficiency>
        <odometry_publish_frequency>10</odometry_publish_frequency>
        <child_frame_id>rotation_axis_link</child_frame_id>
        <linear_velocity>
          <min_velocity>-2.2</min_velocity>
          <max_velocity>2.2</max_velocity>
          <min_acceleration>-1.3</min_acceleration> 
          <max_acceleration>1.3</max_acceleration>
        </linear_velocity>
        <angular_velocity>
          <min_velocity>-1.0</min_velocity>
          <max_velocity>1.0</max_velocity>
          <min_acceleration>-1.1</min_acceleration>
          <max_acceleration>1.1</max_acceleration>
        </angular_velocity>
    </plugin>
```

peci1, 2024-10-14:
I've submitted PR https://github.com/gazebosim/gz-sim/pull/2652 that adds this tag to the system documentation (it was missing there for some reason).

iandresolares, 2024-10-14:
> That's wrong syntax. The plugin expects
> 
> ```
>         <body_link>rotation_axis_link</body_link>
> ```
> 
> If it still doesn't work, start Gazebo with `gz sim -v4` to get debug output and post the output here.

That's it, that works perfectly, thank you very much. I close the issue.

PR #2652 (2024-10-15): TrackedVehicle: Added docs for body_link option
# 🦟 Bug fix

Fixes #2650

## Summary

This PR adds forgotten documentation for the `<body_link>` tag in TrackedVehicle system.

The docs refer to source code at https://github.com/gazebosim/gz-sim/blob/gz-sim9/src/systems/tracked_vehicle/TrackedVehicle.cc#L238 and https://github.com/gazebosim/gz-sim/blob/gz-sim9/src/systems/tracked_vehicle/TrackedVehicle.cc#L475 .

## Checklist
- [x] Signed all commits for DCO
- [ ] Added tests
- [x] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python binding [... cut]
closed 2024-10-14 by iandresolares