gazebosim/gz-sim #2655: Model only reset Gazebo Fortress

paoloelle, 2024-10-23:
## Environment
* OS Version: Ubuntu 22.04
* Gazebo Fortress
* Rendering plugin: ogre2

## Description
* Expected behavior: Reset the world models to its starting position state
* Actual behavior: The world models doesn't reset to its starting position

## Steps to reproduce
I'm try to run the service by using ros_gz_bridge but the behavior is the same also without it.

1. First terminal: `ign gazebo arena3.sdf -v 4`
2. Second terminal: `ros2 run ros_gz_bridge parameter_bridge /world/arena/control@ros_gz_interfaces/srv/ControlWorld@ignition.msgs.WorldControl@ignition.msgs.Boolean`
3. Third terminal: `ros2 service call /world/arena/control ros_gz_interfaces/srv/ControlWorld "{world_control: {reset: {model_only: true}}}"`

## Output
`[Wrn] [SimulationRunner.cc:1131] Model only reset is not supported.`

azeey, 2024-12-16:
Model-only reset is supported in Gazebo Garden and later. There is no plan to backport this to Fortress as the changes required are substantial.

paoloelle, 2024-12-23:
@azeey, seems that I cannot run the command to reset the model only neither in Gazebo Harmonic. I made an Apptainer image with ROS2 Jazzy and Gazebo Harmonic but the result is again the following message: `[Wrn] [SimulationRunner.cc:1131] Model only reset is not supported.` 

[image]

iche033, 2025-01-10:
It does look like model-only reset is [still not supported](https://github.com/gazebosim/gz-sim/blob/d83d1352600022ea9da20906000a963580c7f258/src/SimulationRunner.cc#L1207). Maybe @azeey was referring to the full world reset (https://github.com/gazebosim/gz-sim/pull/1249)?