gazebosim/gz-sim #2659: C++ API for JointController

mogumbo, 2024-10-30:
## Desired behavior
C++ API for JointController, JointPositionController, and JointTrajectoryController similar to the API found in gazebo::physics::JointController from Gazebo Classic. Specific methods I have used from gazebo::physics::JointController are GetJoints, GetPositions, GetPositionPID, GetVelocities, GetVelocityPIDs, SetPositionPID, SetPositionTarget, SetVelocityPID, and SetVelocityTarget.

## Alternatives considered
You can interact with joint controllers using gz-transport, but results of these actions are not necessarily immediate and add complexity to your code if you need to receive a response.

## Additional context
I'm currently using gz-sim8 on Ubuntu 24.04.

scpeters, 2024-10-30:
I can imagine creating components like `JointPositionControlTarget`, `JointVelocityControlTarget` , `JointPositionControlPID`, `JointVelocityControlPID` that the joint control systems can try to read from the ECM in addition to gz-transport topics.

Would it also make sense to create a `JointController` C++ object similar to [gz::sim::Model](https://github.com/gazebosim/gz-sim/blob/gz-sim9/include/gz/sim/Model.hh) or [gz::sim::Joint](https://github.com/gazebosim/gz-sim/blob/gz-sim9/include/gz/sim/Joint.hh)?

azeey, 2025-01-06:
I support creating a `gz::sim::JointController` similar to `gz::sim::Joint` with all the PID functionality and having the current Joint controller plugins use that internally.

yaswanth1701, 2025-03-23:
Hi everyone, can I work on this issue ?

iche033, 2025-03-24:
yes please go ahead, I've assigned the issue to you

yaswanth1701, 2025-03-24:
thanks a lot.

yaswanth1701, 2025-03-28:
Hi, @iche033, which branch should I branch off from?

iche033, 2025-03-28:
Can you target `main` for this new feature? We'll then backport to other release versions once it's merged.

yaswanth1701, 2025-03-30:
Hi everyone. I want to ask a question regarding implementing the joint controller system. So, currently, the `targetVel` or `targetPos` are taken from the respective topics, but if it is set through the joint controller API for the same simulation step, which target value should be given preference, target topic value or `JointControllerVelocityTarget/JointControllePositionTarget` component value.

iche033, 2025-03-31:
If one cmd is received after another in different time step, I would expect that the last received cmd  overrides existing ones. However, it's in the same time step, I don't know if there is a strong preference for one type of cmd over another. We should print out an warning if this occurs and explicitly mention something like the JointController received both velocity and position commands in the same time step so position command will take precedence over velocity command (or the other way around)

yaswanth1701, 2025-04-03:
> If one cmd is received after another in different time step, I would expect that the last received cmd overrides existing ones. However, it's in the same time step, I don't know if there is a strong [...]
Should  joint controller class have callbacks for joint target commands ? I thought it's would used in conjunction with joint controller system and joint controller system will be using component's data from joint controller component.

iche033, 2025-04-03:
I think the existing joint controller system will keep the logic for subscribing to joint commands. In the joint controller system, it instantiates an object from the new joint controller class, and just pass those commands to the new joint controller object which will take care of any PID control calculations and setting data in the Joint*Cmd components.

yaswanth1701, 2025-04-03:
Thanks a lot for the clarification. I was also thinking of doing something similar. Thanks again.