gazebosim/gz-sim #2664: Add API for getting Joint limits

mogumbo, 2024-11-01:
## Desired behavior
Add methods Joint::VelocityLimits, Joint::EffortLimits, Joint::PositionLimits for getting limits, corresponding to the existing setter methods.

## Alternatives considered
I looked at getting these values from the ECM using components::JointVelocityLimitsCmd, components::JointEffortLimitsCmd, and components::JointPositionLimitsCmd. However, their headers all contain comments such as "It is expected that the physics plugin reads this component and sets the limit to the dynamics engine. After setting it, the data of this component will be cleared (i.e. the vector will have length zero)." So it is expected that these components will not contain the desired data.

## Additional context
I'm using gz-sim8 on Ubuntu 24.04.

azeey, 2025-01-06:
We'll need to add components to record the current limits obtained from the physics engine.

shawnkchan, 2025-10-31:
I'd like to work on this!