gazebosim/gz-sim #2665: using the MultiCopter motor Model control plugin causes a segmentation fault

torn8to, 2024-11-02:
## Environment
* OS Version:Ubuntu 24.04 
* ros Version: Jammy

* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: [ogre | ogre2]. 
      - [ ] Sensor rendering error. 
      - [ ] GUI rendering error. 
    - EGL headless mode:
      - [ ] Running in EGL headless mode 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [ ] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [ ] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [ ] running on real hardware
      - [ ] running in virtual machine
      - [ ] running in Docker/Singularity
      - [ ] running remotely (e.g. via SSH)
      - [ ] running in a cloud
      - [ ] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
      - [x ] GPU is concurrently used for other tasks
        - [ ] desktop acceleration
        - [ x] video decoding (i.e. a playing Youtube video)
        - [ ] video encoding
        - [x ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [ ] multiple simulators running at the same time
      - [ ] other...
    - Rendering system info:
  
      - On Linux, provide the outputs of the following commands:
          ```bash
          LANG=C lspci -nn | grep VGA  # might require installing pciutils
          echo "$DISPLAY"
          LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
          ps aux | grep Xorg
          sudo env LANG=C X -version  # if you don't have root access, try to tell the version of Xorg e.g. via package manager
          ```
ouput of above command
```b
1:00.0 VGA compatible controller [0300]: NVIDIA Corporation GA106M [GeForce RTX 3060 Mobile / Max-Q] [10de:2560] (rev a1)
05:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Rembrandt [Radeon 680M] [1002:1681] (rev c8)
:1
direct rendering: Yes
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: NVIDIA GeForce RTX 3060 Laptop GPU/PCIe/SSE2
OpenGL core profile version string: 4.6.0 NVIDIA 555.58.02
OpenGL core profile shading language version string: 4.60 NVIDIA
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL version string: 4.6.0 NVIDIA 555.58.02
OpenGL shading language version string: 4.60 NVIDIA
OpenGL context flags: (none)
OpenGL profile mask: (none)
OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 555.58.02
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
nathan      3754  1.8  1.5 26525984 247304 tty2  Sl+  08:16  15:14 /usr/lib/xorg/Xorg vt2 -displayfd 3 -auth /run/user/1000/gdm/Xauthority -nolisten tcp -background none -noreset -keeptty -novtswitch -verbose 3
nathan    100248  0.0  0.0   9468  2176 pts/3    S+   21:42   0:00 grep --color=auto Xorg
[sudo] password for nathan: 

X.Org X Server 1.21.1.11
X Protocol Version 11, Revision 0
Current Operating System: Linux nathan-Legion-5-15ARH7H 6.5.0-45-generic #45~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Jul 15 16:40:02 UTC 2 x86_64
Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.5.0-45-generic root=UUID=19cbb4a0-baff-4b69-8a22-c54371589d1b ro quiet splash vt.handoff=7
xorg-server 2:21.1.12-1ubuntu1.1 (For technical support please see http://www.ubuntu.com/support) 
Current version of pixman: 0.42.2
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
```
<details>

```bash
[... 231 lines cut ...]
		<timeConstantUp>0.0125</timeConstantUp>
		<timeConstantDown>0.025</timeConstantDown>
		<maxRotVelocity>800.0</maxRotVelocity>
		<motorConstant>8.54858e-06</motorConstant>
		<momentConstant>0.016</momentConstant>
		<commandSubTopic>gazebo/comannd/motor_speed</commandSubTopic>
		<actuator_number>1</actuator_number>
		<rotorDragCoefficient>8.06428e-05</rotorDragCoefficient>
		<rollingMomentCoefficient>1e-06</rollingMomentCoefficient>
		<motorSpeedPubTopic>motor_speed/1</motorSpeedPubTopic>
		<rotorVelocitySlowdownSim>10</rotorVelocitySlowdownSim>
		<motorType>velocity</motorType>
	    </plugin>
	    
    </include>
	</world>
</sdf>

2. launched using this command 
   ```bash
     ros2 launch ros_gz_sim gz_sim.launch.py  gz_args:=" -v 4 drone_sim.sdf"
```
4.

## Output

iche033, 2024-11-04:
Gazebo crashed because it couldn't find the specified link. Can you try changing the joint and link  and joint names to:

```
		<jointName>rotor_0_joint</jointName>
		<linkName>rotor_0</linkName>
```

There was a recent fix to prevent the crash but targeted at Ionic. I just backported it to Harmonic, which is used by Jazzy. https://github.com/gazebosim/gz-sim/pull/2666

torn8to, 2024-11-05:
ah big thanks thanks for the changes and the info

PR #2666 (2024-11-04): Backport fix for multicopter motor plugin crash

# 🦟 Bug fix

Fixes #2665 

## Summary

Backport #2653

## Checklist
- [x] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them)
- [ ] `codecheck` passed (See [contributing](https://gazebosim.org/docs/all/contributing#contributing-code))
- [ ] All tests passed (See [test coverage](https://gazebosim.org/docs/all/contributing#test-coverage))
- [ ] While waiting for a review on your PR, please help review [anothe [... cut]
closed 2024-12-13 by iche033