gazebosim/gz-sim #2670: Windows 11 WSLg Gazebo Black Screen

Ryanf55, 2024-11-07:
# Environment

* Host OS: Windows 11
* WSL2
* Lenovo Thinkpad X1 GEN12
* Gazebo Version: Harmonic
* Using the [WSLg workflow](https://community.gazebosim.org/t/wslg-with-gpu-support-available-on-latest-version-of-gazebo-garden-and-harmonic/2360)
* LIBGL_ALWAYS_SOFTWARE is NOT set ( I want GPU rendering)

# Expected behavior

```
gz sim shapes.sdf`
```

On another (beefier) Windows 10 computer ( Dell precision 7770, NVIDIA RTX A3000 GPU) , this works in WSL2 

# Actual Behavior

The Lenovo Thinkpad starts gazebo, but the renderer is black. 

# Logs 

Coming soon

Ryanf55, 2024-11-07:
# Working Logs from my Dell
[dell_gz_harmonic_shapes_v4.txt](https://github.com/user-attachments/files/17668439/dell_gz_harmonic_shapes_v4.txt)

Ryanf55, 2024-11-07:
Edit: Today, it works. Uploading logs, closing the issue unless it pops back up.

Here's the thinkpad X1 carbon logs

[gz_harmonic_thinkpad_logs.txt](https://github.com/user-attachments/files/17668547/gz_harmonic_thinkpad_logs.txt)

nesserine-laribi, 2025-01-12:
hello i have the same problem as you did you solve it please ?

Ryanf55, 2025-01-13:
It's unresolved.

traversaro, 2025-01-15:
Sorry, I forgot to reply. The actual error from ogre2.log seems:

~~~
15:27:37: Added resource location '/usr/share/gz/gz-rendering8/media/materials/textures/' of type 'FileSystem' to resource group 'General'
15:27:37: Can't assign material [Default] because this Material does not exist. Have you forgotten to define it in a .material script?
15:27:37: Can't assign material scene::Material(65501) because this Material does not exist. Have you forgotten to define it in a .material script?
15:27:37: Can't assign material scene::Material(65495) because this Material does not exist. Have you forgotten to define it in a .material script?
15:27:37: Can't assign material scene::Material(65489) because this Material does not exist. Have you forgotten to define it in a .material script?
15:27:37: Can't assign material scene::Material(65483) because this Material does not exist. Have you forgotten to define it in a .material script?
15:27:37: Can't assign material scene::Material(65476) because this Material does not exist. Have you forgotten to define it in a .material script?
15:27:37: Can't assign material scene::Material(65483) because this Material does not exist. Have you forgotten to define it in a .material script?
15:27:37: Can't assign material scene::Material(65465) because this Material does not exist. Have you forgotten to define it in a .material script?
15:27:37: OGRE EXCEPTION(3:RenderingAPIException): Fixed Function pipeline is no longer allowed nor supported. The material scene::Material(65518) must use shaders in HlmsLowLevel::calculateHashFor at ./OgreMain/src/OgreHlmsLowLevel.cpp (line 160)
15:27:37: OGRE EXCEPTION(3:RenderingAPIException): Fixed Function pipeline is no longer allowed nor supported. The material scene::Material(65518) must use shaders in HlmsLowLevel::calculateHashFor at ./OgreMain/src/OgreHlmsLowLevel.cpp (line 160)
15:27:37: Couldn't apply datablock '[Hash 0x1dc5a795]' to this renderable. Using default one. Check previous log messages to see if there's more information.
15:27:37: OGRE EXCEPTION(3:RenderingAPIException): Fixed Function pipeline is no longer allowed nor supported. The material scene::Material(65518) must use shaders in HlmsLowLevel::calculateHashFor at ./OgreMain/src/OgreHlmsLowLevel.cpp (line 160)
15:27:37: OGRE EXCEPTION(3:RenderingAPIException): Fixed Function pipeline is no longer allowed nor supported. The material scene::Material(65518) must use shaders in HlmsLowLevel::calculateHashFor at ./OgreMain/src/OgreHlmsLowLevel.cpp (line 160)
15:27:37: Couldn't apply datablock '[Hash 0x1dc5a795]' to this renderable. Using default one. Check previous log messages to see if there's more information.
15:27:37: Can't assign material scene::Material(65458) because this Material does not exist. Have you forgotten to define it in a .material script?
15:27:37: OGRE EXCEPTION(3:RenderingAPIException): Fixed Function pipeline is no longer allowed nor supported. The material Default/TransGreen must use shaders in HlmsLowLevel::calculateHashFor at ./OgreMain/src/OgreHlmsLowLevel.cpp (line 160)
15:27:37: OGRE EXCEPTION(3:RenderingAPIException): Fixed Function pipeline is no longer allowed nor supported. The material Default/TransGreen must use shaders in HlmsLowLevel::calculateHashFor at ./OgreMain/src/OgreHlmsLowLevel.cpp (line 160)
15:27:37: Couldn't apply datablock '[Hash 0x1a9fb58f]' to this renderable. Using default one. Check previous log messages to see if there's more information.
15:27:37: Vertex Shader: 100000001VertexShader_vs
Fragment Shader: 100000001PixelShader_ps
 GLSL validation result : 
active samplers with a different type refer to the same texture image unit
15:27:38: WARNING: Deleting mapped buffer without having it unmapped. This is often sign of a resource leak or a bad pattern. Umapping the buffer for you...
15:27:38: WARNING: Deleting mapped buffer without having it unmapped. This is often sign of a resource leak or a bad pattern. Umapping the buffer for you...
15:27:38: WARNING: Deleting mapped buffer without having it unmapped. This is often sign of a resource leak or a bad pattern. Umapping the buffer for you...
15:27:38: WARNING: Deleting mapped buffer without having it unmapped. This is often sign of a resource leak or a bad pattern. Umapping the buffer for you...
~~~

It would be interesting to get the `ogre2.log` in case of a working installation to compare the two.

traversaro, 2025-01-15:
Note that mesa 24 had a lot of fixes to `d3d12` (see https://github.com/gazebosim/gz-sim/issues/920#issuecomment-1862553317), so it may be worth to give it a shot to update mesa and see how it goes with a recent mesa.

jaehho, 2025-02-26:
this fixed it for me
```bash
export LIBGL_ALWAYS_SOFTWARE=1
```

from [Gazebo Community](https://community.gazebosim.org/t/wslg-with-gpu-support-available-on-latest-version-of-gazebo-garden-and-harmonic/2360) and [closed issue](https://github.com/gazebosim/gz-sim/issues/920#issue-946786820)

Ryanf55, 2025-02-26:
> this fixed it for me
> 
> export LIBGL_ALWAYS_SOFTWARE=1
> from [Gazebo Community](https://community.gazebosim.org/t/wslg-with-gpu-support-available-on-latest-version-of-gazebo-garden-and-harmonic/2360) and [closed issue](https://github.com/gazebosim/gz-sim/issues/920#issue-946786820)

That's a workaround, not a fix. WSL has a GPU, this would prevent us using the GPU. Only the simplest of simulations run with software rendering only.

knmcguire, 2025-03-24:
Hi! Just tried it out on my WSL2 Ubuntu 24.04 on Windows 11 home. No black screen for me actually, and I didn't even had to set `LIBGL_ALWAYS_SOFTWARE=1`

[image]

Seems to use my standard GPU (but not my second Nvidia one but that is a different issue)

But I do definitely remember that I had to do the above to make it work on my previous machine... and now I've done a full fresh install of Windows a month ago. That would be a pretty nuclear fix... And I'm not sure if WSL is different with the Ubuntu versions.

Anyway, thought this might help you to further debug your problem.

traversaro, 2025-03-24:
> Seems to use my standard GPU (but not my second Nvidia one but that is a different issue)

Did you tried to set MESA_D3D12_DEFAULT_ADAPTER_NAME? See https://github.com/microsoft/wslg/wiki/GPU-selection-in-WSLg .

traversaro, 2025-03-24:
> But I do definitely remember that I had to do the above to make it work on my previous machine... and now I've done a full fresh install of Windows a month ago. That would be a pretty nuclear fix... And I'm not sure if WSL is different with the Ubuntu versions.

If you have another machine affected by this I guess it is more probable that this is a mesa version issue or the windows host GPU driver (or GPU itself) not supporting some features required by the d3d12 mesa driver rather then something that can be fixed by a Windows reinstall.

knmcguire, 2025-03-24:
@traversaro ah no sorry for the confusion. Same machine (Lenovo Legion S7), but just a full fresh install :) 

I haven't tried with the MESA setting to enable it for the Nvidia GPU, but that is something else that is non-gazebo related and the AMD GPU is plenty for now to try out this usecase.

traversaro, 2025-03-25:
> [@traversaro](https://github.com/traversaro) ah no sorry for the confusion. Same machine (Lenovo Legion S7), but just a full fresh install :)

In that case my hunch is that either you changed the GPU driver version, or the mesa version used inside Linux (that provides the `d3d12` driver that acts as a bridge between OpenGL in WSL and Direct3D12 on Linux).

knmcguire, 2025-03-30:
Gotcha! sorry for the delay but here are the driver versions for reference
- AMD Radeon Graphics (2023-03-23): 31.0.12044.47003
- Mesa 24.2.8-1

changqingjiaotaimao, 2025-04-10:
I encountered a similar issue of a black screen during 3D rendering. If you're using VMware Ubuntu and haven't changed any settings, try disabling the "3D Acceleration" option in your VMware configuration (as shown in the image) and restart Ubuntu. The problem arises due to a compatibility conflict between VMware's rendering settings and the system's graphics libraries.

[image]