gazebosim/gz-sim #2688: Visualization of frustum

BA-Utkarsh, 2024-11-27:
Hello Team,

We are looking for the visualization of the frustum plugin in latest gazebo repo.
We could see it was present in gazebo classic and from gazebo garden onwards the plugin/feature is not available.

If there any scope of adding a new feature please let us know.
Moreover, we have found a similar thread, https://github.com/gazebosim/gz-sim/issues/2401 where @scpeters (after discussion with @iche033) mentioned about we can do similar way by adding a gui plugin similar to visualize_lidar plugin.
Ref - https://github.com/gazebosim/gz-sim/issues/2401#issuecomment-2183287236

Please let us know if someone in community actively working on this thread? We can support.
**Or we can initiate the contribution from our side.**

Regards,

ahcorde, 2024-11-27:
@BA-Utkarsh There is no open PR for this feature, I would say nobody is working on this. 

We will be happy to review your PR. I can assign you this issue

BA-Utkarsh, 2024-11-27:
> @BA-Utkarsh There is no open PR for this feature, I would say nobody is working on this.
> 
> We will be happy to review your PR. I can assign you this issue

Oh ok. that's great.
Will proceed accordingly.

Ryanf55, 2024-11-28:
Yea, this would be great if you wanted to contribute it!

BA-Utkarsh, 2024-12-26:
@ahcorde @azeey @Ryanf55 
I have the things ready. I shall create the PR in gz-sim, gz-sensors and gz-rendering.

BA-Utkarsh, 2024-12-26:
@ahcorde @azeey @Ryanf55
Created respective PRs as below,
- https://github.com/gazebosim/gz-sim/pull/2707
- https://github.com/gazebosim/gz-sensors/pull/491
- https://github.com/gazebosim/gz-rendering/pull/1095

Also created ticket in respective repos as below,
- https://github.com/gazebosim/gz-sensors/issues/492
- https://github.com/gazebosim/gz-rendering/issues/1096

BA-Utkarsh, 2025-01-30:
@azeey , seems all the PRs are merged. We can close this issue.
Thank you @iche033 , @ahcorde , @azeey & @ntfshard for your support. :)

ahcorde, 2025-01-30:
Thank you for your contribution! Hopefully we can see more in the future!

I will close this issue because the task is completed!

closed 2025-01-30 by ahcorde

Ryanf55, 2025-01-31:
Awesome work @BA-Utkarsh !!