gazebosim/gz-sim #2698: GZ Harmonic: Client get empty window when connects to remote server

rom341, 2024-12-12:
## Environment
* OS Version: Ubuntu 24.04
* Binary build: Harmonic 8.7.0
  
  
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: ogre2 
      - [ ] Sensor rendering error. 
      - [X] GUI rendering error. 
    - EGL headless mode:
      - [ ] Running in EGL headless mode 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [ ] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [ ] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [X] running on real hardware
      - [ ] running in virtual machine
      - [ ] running in Docker/Singularity
      - [X] running remotely (e.g. via SSH)
      - [X] running in a cloud
      - [ ] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
      - [ ] GPU is concurrently used for other tasks
        - [ ] desktop acceleration
        - [ ] video decoding (i.e. a playing Youtube video)
        - [ ] video encoding
        - [ ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [ ] multiple simulators running at the same time
      - [ ] other...

    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
          ```bash
          LANG=C lspci -nn | grep VGA  # might require installing pciutils
          echo "$DISPLAY"
          LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
          ps aux | grep Xorg
          sudo env LANG=C X -version  # if you don't have root access, try to tell the version of Xorg e.g. via package manager
          ```
          ```bash
          roman@roman:~$ LANG=C lspci -nn | grep VGA
          00:02.0 VGA compatible controller [0300]: Intel Corporation HD Graphics 620 [8086:5916] (rev 02)

          roman@roman:~$ echo "$DISPLAY"
          :1

         roman@roman:~$ LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  
        direct rendering: Yes
            Preferred profile: core (0x1)
            Max core profile version: 4.6
            Max compat profile version: 4.6
            Max GLES1 profile version: 1.1
            Max GLES[23] profile version: 3.2
        OpenGL vendor string: Intel
        OpenGL renderer string: Mesa Intel(R) HD Graphics 620 (KBL GT2)
        OpenGL core profile version string: 4.6 (Core Profile) Mesa 24.0.9-0ubuntu0.2
        OpenGL core profile shading language version string: 4.60
        OpenGL core profile context flags: (none)
        OpenGL core profile profile mask: core profile
        OpenGL version string: 4.6 (Compatibility Profile) Mesa 24.0.9-0ubuntu0.2
        OpenGL shading language version string: 4.60
        OpenGL context flags: (none)
        OpenGL profile mask: compatibility profile
        OpenGL ES profile version string: OpenGL ES 3.2 Mesa 24.0.9-0ubuntu0.2
        OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20

          roman@roman:~$ ps aux | grep Xorg
          roman       2826  7.4  1.5 7403828 188816 tty2   Sl+  12:41   2:44 /usr/lib/xorg/Xorg vt2 -displayfd 3 -auth /run/user/1000/gdm/Xauthority -nolisten tcp -background none -noreset -keeptty -novtswitch -verbose 3
          roman      10417  0.0  0.0  18136  2176 pts/4    S+   13:17   0:00 grep --color=auto Xorg

          roman@roman:~$ sudo env LANG=C X -version 

          X.Org X Server 1.21.1.11
          X Protocol Version 11, Revision 0
          Current Operating System: Linux roman 6.8.0-50-generic #51-Ubuntu SMP PREEMPT_DYNAMIC Sat Nov  9 17:58:29 UTC 2024 x86_64
          Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.8.0-50-generic root=UUID=d34a680e-7c4c-49f1-bf4b-528073bb9073 ro quiet splash vt.handoff=7
          xorg-server 2:21.1.12-1ubuntu1.1 (For technical support please see http://www.ubuntu.com/support) 
          Current version of pixman: 0.42.2
	          Before reporting problems, check http://wiki.x.org
	          to make sure that you have the latest version.

          ```
    - [X] Please, attach the ogre.log or ogre2.log file from  `~/.gz/rendering` 

<details>

[client_ogre2.log](https://github.com/user-attachments/files/18109135/client_ogre2.log)

</details>

## Description
I need to run 'gz -s' on aws or computer in the other network, connect to it with OpenVPN in tap mode (udp multicast can pass through) and operate simulation by running 'gz sim -g' from my computer.

* Expected behavior: 
When running 'gz -g' it connects to server, and shows world and entities 

* Actual behavior: 
When running 'gz -g' it connects to server, but shows empty window (no scene, no entity tree, etc)
[image]

But, if world contains camera, i can use this camera to see the world
[image]

## Steps to reproduce

1. On server (AWS EC2 Ubuntu 24.04 ip = 10.8.0.1): 
          ```
           GZ_IP=10.8.0.1 GZ_PARTITION=1 GZ_VERBOSE=1 GZ_TRANSPORT_RCVHWM=100000 GZ_TRANSPORT_SNDHWM=100000 GZ_TRANSPORT_TOPIC_STATISTICS=0 gz sim -s -v4
          ```

2. On client (Real PC Ubuntu 24.04 ip =10.8.0.2)
          ```
           GZ_IP=10.8.0.2 GZ_PARTITION=1 GZ_VERBOSE=1 GZ_TRANSPORT_RCVHWM=100000 GZ_TRANSPORT_SNDHWM=100000 GZ_TRANSPORT_TOPIC_STATISTICS=0 gz sim -g -v4
          ```

##Extra
1. Ping between server and client +- 150ms
2. GZ_RELAY - have no effect
3. If these steps are repeated when the server and client are on the same Wi-Fi network, everything works correctly; the client can see and interact with the world.
4. if use comand 
'''
sudo tc qdisc add dev lo root netem delay 17ms
'''
it will create 34ms ping on localhost and gz client will see gray windoa as at the first screenshot, but if change this ping to 16ms or lower, all work correctly

## Output

[gz_client_console.txt](https://github.com/user-attachments/files/18109109/gz_client_console.txt)

[gz_server_console.txt](https://github.com/user-attachments/files/18109175/gz_server_console.txt)
</details>

rom341, 2024-12-13:
but why gui does not receive that service call, if ping is more than 17ms?
all work well, if do it in wifi newtwork, or with docker or VM, where ping is minimal, but if i use 
`sudo tc qdisc add dev <network interface> root netem delay 17ms`
gui breaks even with wifi or docker

iche033, 2024-12-13:
> but why gui does not receive that service call, if ping is more than 17ms?

that I'm not sure. Maybe @caguero has some ideas about the impact of latency on gz-transport service requests?

arjo129, 2024-12-16:
Is the [scene broadcaster](https://github.com/gazebosim/gz-sim/blob/f0cde0574b357fba3cb9ee3f22df0108bda44db1/examples/worlds/empty.sdf#L76-L79) plugin enabled in your world?

rom341, 2024-12-16:
Yes, scene broadcaster
<https://github.com/gazebosim/gz-sim/blob/f0cde0574b357fba3cb9ee3f22df0108bda44db1/examples/worlds/empty.sdf#L76-L79>
plugin is enabled. I am using default worlds. (empty.sdf; shapes.sdf; etc.)

rom341, 2024-12-17:
Is it even possible to run Gazebo Harmonic or other version (except classic) on different computers and connect them via the internet (vpn tunnel for example)?

I have successfully launched gazebo classic through a vpn tunnel, but cant do it with new versions

r-bt, 2025-01-26:
@rom341 Did you have any luck getting this to work? I'm also interested in having a Gazebo server on one computer and connecting to it from another

rom341, 2025-01-27:
@r-bt Yes. I successfully ran `gz sim -s` on aws and my friend can successfully connect to it using `gz sim -g` through vpn network.
but my friend has ping to server +- 30ms, so the condition described above (less than 17ms in one direction) is met.
mine is +-70ms, so I still see that "blank" window. The only way to connect a client is to use remote desktop to a computer that has lower ping to your server.

used comands:
server and client connected with OpenVPN network in tun mode (udp multicast can`t be used), so client must use GZ_RELAY

    On server (AWS EC2 Ubuntu 24.04 ip = 10.8.0.1)
    GZ_IP=10.8.0.1 GZ_PARTITION=1 GZ_VERBOSE=1 GZ_TRANSPORT_RCVHWM=100000 GZ_TRANSPORT_SNDHWM=100000 GZ_TRANSPORT_TOPIC_STATISTICS=0 gz sim -s -v4

    On client (Real PC Ubuntu 24.04 ip = 10.8.0.2)
    GZ_IP=10.8.0.2 GZ_PARTITION=1 GZ_RELAY=10.8.0.1 GZ_VERBOSE=1 GZ_TRANSPORT_RCVHWM=100000 GZ_TRANSPORT_SNDHWM=100000 GZ_TRANSPORT_TOPIC_STATISTICS=0 gz sim -g -v4