gazebosim/gz-sim #2704: Interface to get pointcloud from SDF

MA-Wenhui, 2024-12-16:
## Desired behavior
Need an interface to get the world's all collisions and generate the 3d map(pointcloud). 

## Alternatives considered

I tried to parse sdf file to extract dae/stl/geometry, but it was too complicated since the sdf structures vary.

## Implementation suggestion

There is a RayQuery based solution for gazebo-classic:

https://github.com/[arshadlab/gazebo_map_creator](https://github.com/arshadlab/gazebo_map_creator)

I didn't find the equivalent class for this usage in gz harmonic.

## Additional context

arjo129, 2024-12-19:
I don't know if this should be a core part of gazebo. The buoyancy plugin for instance takes the [collision mesh](https://github.com/gazebosim/gz-sim/blob/9d6230a562c5d463306070b07269b135f9ecd134/src/systems/buoyancy/Buoyancy.cc#L297) and iterates through the individual meshes. 

Perhaps a more refined question would be we should expose an easy to use `RayQuery` API since this is something that is relatively fundamental to a lot of robotics software.

azeey, 2025-01-07:
https://github.com/gazebosim/gz-sim/pull/2514 could be useful here.