gazebosim/gz-sim #2708: Camera sensor images intermittently turn to black.

david-wb, 2025-01-06:
## Environment
* OS Version: Ubuntu 22.04
* Binary version 6.16.0
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: [ogre2]. 
      - [x] Sensor rendering error. 
      - [ ] GUI rendering error. 
    - EGL headless mode:
      - [ ] Running in EGL headless mode 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [ ] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [ ] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [ ] running on real hardware
      - [ ] running in virtual machine
      - [ ] running in Docker/Singularity
      - [ ] running remotely (e.g. via SSH)
      - [ ] running in a cloud
      - [ ] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
      - [ ] GPU is concurrently used for other tasks
        - [ ] desktop acceleration
        - [ ] video decoding (i.e. a playing Youtube video)
        - [ ] video encoding
        - [ ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [ ] multiple simulators running at the same time
      - [ ] other...
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
          ```bash
          LANG=C lspci -nn | grep VGA  # might require installing pciutils
          echo "$DISPLAY"
          LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
          ps aux | grep Xorg
          sudo env LANG=C X -version  # if you don't have root access, try to tell the version of Xorg e.g. via package manager
          ```
      - On Windows, run `dxdiag` and report the GPU-related information.
      - On Mac OS, open a terminal and type `system_profiler SPDisplaysDataType`. Copy the output here.
        
    - [x] Please, attach the ogre.log or ogre2.log file from  `~/.gz/rendering` 

<details>

```
0000:00:02.0 VGA compatible controller [0300]: Intel Corporation Device [8086:a788] (rev 04)
0000:01:00.0 VGA compatible controller [0300]: NVIDIA Corporation Device [10de:27ba] (rev a1)
:1
direct rendering: Yes
    Preferred profile: core (0x1)
    Max core profile version: 4.6
    Max compat profile version: 4.6
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
OpenGL vendor string: Intel
OpenGL renderer string: Mesa Intel(R) Graphics (RPL-S)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 23.2.1-1ubuntu3.1~22.04.2
OpenGL core profile shading language version string: 4.60
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL version string: 4.6 (Compatibility Profile) Mesa 23.2.1-1ubuntu3.1~22.04.2
OpenGL shading language version string: 4.60
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 23.2.1-1ubuntu3.1~22.04.2
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
chroma      2700  3.0  0.2 29702672 178052 tty2  Sl+  08:24  10:33 /usr/lib/xorg/Xorg vt2 -displayfd 3 -auth /run/user/1001/gdm/Xauthority -nolisten tcp -background none -noreset -keeptty -novtswitch -verbose 3
chroma    275067  0.0  0.0  11760  2304 pts/0    S+   14:07   0:00 grep --color=auto Xorg
[sudo] password for chroma: 
```

```
14:11:06: Creating resource group General
[... 552 lines cut ...]
active samplers with a different type refer to the same texture image unit
14:11:09: WARNING: Deleting mapped buffer without having it unmapped. This is often sign of a resource leak or a bad pattern. Umapping the buffer for you...
14:11:34: Vertex Shader: 100000003VertexShader_vs
Fragment Shader: 100000003PixelShader_ps
 GLSL validation result : 
active samplers with a different type refer to the same texture image unit
idation result : 
active samplers with a different type refer to the same texture image unit
```

</details>

## Description
* Expected behavior: Gazebo Fortress camera sensor should function properly and not return black/empty images.
* Actual behavior: The images are intermittently black/empty. Please see screenshots below.

## Steps to reproduce

1. Create simulation of a robot with a camera sensor.
2. Move the robot around the environment.
3. Watch the camera output and notice that it turns to black. 

## Output
[image]
[image]

azeey, 2025-01-07:
This might be an issue with the graphics driver. I noticed that you have both an Intel and NVIDIA graphics cards, but you are currently using the Intel. Have you tried with the NVIDIA? Another thing to try would be to update your Mesa drivers (https://launchpad.net/~kisak/+archive/ubuntu/kisak-mesa). Also, can you fill out the "Generally, mention all circumstances that might affect rendering capabilities:" section on the issue template?

david-wb, 2025-01-07:
Here is the rendering system info.

```bash
$ LANG=C lspci -nn | grep VGA  # might require installing pciutils
echo "$DISPLAY"
LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
ps aux | grep Xorg
sudo env LANG=C X -version  # if you don't have root access, try to tell the version of Xorg e.g. via package manager
0000:00:02.0 VGA compatible controller [0300]: Intel Corporation Device [8086:a788] (rev 04)
0000:01:00.0 VGA compatible controller [0300]: NVIDIA Corporation Device [10de:27ba] (rev a1)
:1
direct rendering: Yes
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: NVIDIA RTX 4000 Ada Generation Laptop GPU/PCIe/SSE2
OpenGL core profile version string: 4.6.0 NVIDIA 535.183.01
OpenGL core profile shading language version string: 4.60 NVIDIA
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL version string: 4.6.0 NVIDIA 535.183.01
OpenGL shading language version string: 4.60 NVIDIA
OpenGL context flags: (none)
OpenGL profile mask: (none)
OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 535.183.01
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
chroma      2941  3.5  0.3 27704240 201680 tty2  Sl+  10:21   0:32 /usr/lib/xorg/Xorg vt2 -displayfd 3 -auth /run/user/1001/gdm/Xauthority -nolisten tcp -background none -noreset -keeptty -novtswitch -verbose 3
chroma     21134  0.0  0.0  11760  2304 pts/0    S+   10:36   0:00 grep --color=auto Xorg

X.Org X Server 1.21.1.4
X Protocol Version 11, Revision 0
Current Operating System: Linux chroma 6.8.0-50-generic #51~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Nov 21 12:03:03 UTC 2 x86_64
Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.8.0-50-generic root=UUID=ad140d1c-624f-4fe2-b717-212eeb39b772 ro quiet splash vt.handoff=7
xorg-server 2:21.1.4-2ubuntu1.7~22.04.12 (For technical support please see http://www.ubuntu.com/support) 
Current version of pixman: 0.40.0
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
```

I tried using the nvidia gpu vs the intel one and it makes no difference. The gazebo log and ogre2 log are attached. One interesting thing is this problem does not occur when using `ogre` in place of `ogre2`.

[server_console.log](https://github.com/user-attachments/files/18330532/server_console.log)
[ogre2.log](https://github.com/user-attachments/files/18330534/ogre2.log)

Ddddddzy, 2026-01-22:
I encountered a similar issue as well. When I select Ogre2, the camera flickers and occasionally turns completely black, and the screen goes black when the vehicle stops moving. However, when I switch to Ogre, the camera works properly, but the GPU LiDAR malfunctions: all LiDAR readings remain stuck at the minimum value and cannot be used normally.