gazebosim/gz-sim #2711: No odometry data available mecanum drive plugin using gazebo topic echo CLI

TZECHIN6, 2025-01-09:
## Environment
* OS Version:  Ubuntu 22.04.5 LTS
* Source or binary build? Binary, Ignition Fortress
* ROS Version: Humble

<details>

```
# paste log here
```

</details>

## Description
* Expected behavior: 
Expected to have all the topic message being echo to confirm the dataflow is ready to pipe to ROS2 env.
* Actual behavior: 
Not all the topic is shown, and the `odom` topic from the mecanum drive plugin has no data at all.

## Steps to reproduce

1. The sdf file is mostly copied from the example. You can create a sdf and copy below code for testing.
2. `ign gazebo mecanum_drive_custom.sdf` (depends what you have named your sdf)
3. Start a new terminal, `ign topic -e -t /odom`
4. No message being printed out

## Output

Empty...

## Used sdf file for testing
```
<?xml version="1.0" ?>
<!--
  Ignition Gazebo Mecanum drive plugin demo

  Try sending commands:

    ign topic -t "/model/vehicle_blue/cmd_vel" -m ignition.msgs.Twist -p "linear: {x: 0.5, y: 0.5}"

    ign topic -t "/model/vehicle_blue/cmd_vel" -m ignition.msgs.Twist -p "linear: {y: 0.5}, angular: {z: 0.2}"

-->
<sdf version="1.9">
  <world name="mecanum_drive">

    <physics name="1ms" type="ignored">
      <max_step_size>0.001</max_step_size>
      <real_time_factor>1.0</real_time_factor>
    </physics>
    <plugin
      filename="ignition-gazebo-physics-system"
      name="ignition::gazebo::systems::Physics">
    </plugin>
    <plugin
      filename="ignition-gazebo-user-commands-system"
      name="ignition::gazebo::systems::UserCommands">
    </plugin>
    <plugin
      filename="ignition-gazebo-scene-broadcaster-system"
      name="ignition::gazebo::systems::SceneBroadcaster">
    </plugin>
    <plugin
      filename="ignition-gazebo-sensors-system"
      name="gz::sim::systems::Sensors">
      <render_engine>ogre2</render_engine>
    </plugin>
    <plugin 
      filename="ignition-gazebo-imu-system"
      name="ignition::gazebo::systems::Imu">
[... 374 lines cut ...]
        <child>base_scan</child>
      </joint>

      <plugin
        filename="ignition-gazebo-mecanum-drive-system"
        name="ignition::gazebo::systems::MecanumDrive">
        <front_left_joint>front_left_wheel_joint</front_left_joint>
        <front_right_joint>front_right_wheel_joint</front_right_joint>
        <back_left_joint>rear_left_wheel_joint</back_left_joint>
        <back_right_joint>rear_right_wheel_joint</back_right_joint>
        <wheel_separation>1.25</wheel_separation>
        <wheelbase>1.511</wheelbase>
        <wheel_radius>0.3</wheel_radius>
        <min_acceleration>-5</min_acceleration>
        <max_acceleration>5</max_acceleration>
        <odom_topic>odom</odom_topic> <!-- from <odometry_topic> -->
        <frame_id>odom</frame_id> <!-- from <odometry_frame> -->
        <child_frame_id>chassis</child_frame_id> <!-- from <robot_base_frame> -->
      </plugin>

    </model>

  </world>
</sdf>
```

azeey, 2025-01-09:
This is being implemented in https://github.com/gazebosim/gz-sim/pull/2297. Once that's merged, it might be possible to backport it.

TZECHIN6, 2025-01-10:
Thanks for the reply, sorry I am very new to the new gazebo. So, by your statement, my current build has no odom and tf data available right? So even I us gz-bridge to bridge up ROS2 topic, there will be still no data at all for odom and tf.

(I have followed your turtlebot3 example, and want to create a mecanum drive robot with Nav2)

TZECHIN6, 2025-01-27:
@azeey Hi, any update for the backport? As I am working on a school project, if the foreseeable time might take a while, does the implementation compatible with `ign-gazebo-6.17.0`?