gazebosim/gz-sim #2713: AddLinkForce alternative

ZenBubble, 2025-01-11:
## Desired behavior
In old gazebo, this function was used to apply force relative to a link frame, which made it optimal to use for thrusters for AUVs and any underwater vehicle. 

## Alternatives considered
Tried using AddWorldForce but the force is applied relative to the world

ZenBubble, 2025-01-12:
My current workaround is by taking the world pose of the link and multiplying its quaternion by the desired vector force

azeey, 2025-01-13:
Yes, `AddWorldForce` applies the force at the center of mass of the link, but the force is expressed in world coordinates. An API where the force is expressed in the link frame has not been implemented yet. Would you be interested in contributing a PR?

arjo129, 2025-03-07:
Feel free to open a PR with the new function pointing to main.

hezek94, 2025-04-17:
@ZenBubble I believe this issue is still open, I would like to pick it up and I will be also be needing some up to understand some information about what is needed, I would Like to know if this is link [[AddLinkForce](https://github.com/gazebosim/gz-sim/pull/2816/files)] is part of this issue?

Vedant87, 2025-04-17:
> @ZenBubble I believe this issue is still open, I would like to pick it up and I will be also be needing some up to understand some information about what is needed, I would Like to know if this is link [[AddLinkForce](https://github.com/gazebosim/gz-sim/pull/2816/files)] is part of this issue?
> 

The PR introduces an API similar to the one mentioned in the issue, it introduces an API for applying a force expressed in terms of the inertial frame at the inertial or at an offset from the inertial.

hezek94, 2025-04-17:
Thanks for this. If I understand correctly, the issue is about implementing a function that makes the applies force at the link’s center of mass and expresses it in the link frame, rather than in the world frame, which has already been implemented?

hezek94, 2025-04-17:
> The PR introduces an API similar to the one mentioned in the issue, it introduces an API for applying a force expressed in terms of the inertial frame at the inertial or at an offset from the [...]

 If I understand correctly, the issue is about implementing a function that makes the applies force at the link’s center of mass and expresses it in the link frame, rather than in the world frame, which has already been implemented?

hezek94, 2025-04-17:
I’m not entirely sure of the best approach, but I’ve forked [this repository](https://github.com/hezek94/gz-simAddLinkForce) and created a new branch to work from. Should the issue be officially assigned to me, or would it be better if I linked the forked branch here? @Vedant87 ?

PR #2816 (2025-05-12): Link::AddForceInInertialFrame APIs added

# 🎉 New feature

Closes #2713 

## Summary
Function added: to apply force relative to the center of mass of the link frame, the function converts the force to a force with respect to the world coordinates. It uses the AddWorldForce method to apply force either at the center of mass of the link or at an offset from the center of mass, based on user input.

## Checklist
- [x] Signed all commits for DCO
- [x] Added tests
- [ ] Added example and/or tutorial
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [x] Consider upd [... cut]