gazebosim/gz-sim #2715: https://robotics.stackexchange.com/questions/114261/failed-to-load-system-plugin-gzsimsystemsdiffdrive

abdvswmdr, 2025-01-12:
I'm using ROS2 Jazzy on Ubuntu 24.04. I'm encountering an error when launching the diff drive plugin in Gazebo for my robot. The error message on the terminal after ros2 launch is

[image]

I'm also using the joystick as a teleop, which doesn't work, meaning the robot cannot move. The topic /cmd_vel is also not published, only /cmd_vel_joy shows up on "ros2 topic list". I setup my diff drive control plugin as below in ros2_control.xacro 

`  
<gazebo>
    <plugin name="gz_ros2_control::GazeboSimROS2ControlPlugin" filename="libgz_ros2_control-system.so">
`

The gazebo_control.xacro is set as below: 

[image]

azeey, 2025-01-13:
The error messages shows you're trying to load `gz::sim::Systems::DiffDrive`, but it should be `gz::sim::systems::DiffDrive` with a lowercase `s` in `systems`.

abdvswmdr, 2025-01-13:
> The error messages shows you're trying to load `gz::sim::Systems::DiffDrive`, but it should be `gz::sim::systems::DiffDrive` with a lowercase `s` in `systems`.

Thanks! and yes that was part of the problem. I already fixed the problem and it works fine now. Just a note, I realized that it's actually called **Gazebo Sim** (version 8.7.0) and not **Gazebo Harmonic**. This was very confusing since I ended up installing wrong libraries. 

For diff-drive plugins, these are the ones which work now
`<plugin filename="libgz-sim-diff-drive-system.so" name="gz::sim::v8::systems::DiffDrive">`

I also had a message type mismatch and that's why I couldn't control the robot with my Xbox 360 joystick, they joy node will pass in Twist but then in the gz_bridge.yaml, we must convert it to TwistStamped to pass it to gz, otherwise it won't work. 

`- ros_topic_name: "diff_cont/cmd_vel_unstamped"`
     `  gz_topic_name: "cmd_vel"`
   `  ros_type_name: "geometry_msgs/msg/TwistStamped"`
   `  gz_type_name: "gz.msgs.Twist"`
   `  direction: ROS_TO_GZ`

azeey, 2025-01-14:
Glad it's working now. Did you follow the https://gazebosim.org/docs/harmonic/ros_installation/ ? You don't have to install Gazebo directly if you're using it with ROS.

> <plugin filename="libgz-sim-diff-drive-system.so" name="gz::sim::v8::systems::DiffDrive">

It's better to drop the `v8` so that you can use the same SDF/URDF file with newer versions of Gazebo in the future. It's also better to drop the `lib` and `.so` from filename to keep the file compatible with macOS and Windows. This should work
```xml
<plugin filename="gz-sim-diff-drive-system" name="gz::sim::systems::DiffDrive">
```

I'll close this issue since it's not a bug.

closed 2025-01-14 by azeey

abdvswmdr, 2025-01-14:
> Glad it's working now. Did you follow the https://gazebosim.org/docs/harmonic/ros_installation/ ? You don't have to install Gazebo directly if you're using it with ROS.
> 
> > 
> 
> It's better to drop the `v8` so that you can use the same SDF/URDF file with newer versions of Gazebo in the future. It's also better to drop the `lib` and `.so` from filename to keep the file compatible with macOS and Windows. This should work
> 
> ```
> <plugin filename="gz-sim-diff-drive-system" name="gz::sim::systems::DiffDrive">
> ```
> 
> I'll close this issue since it's not a bug.

Sure, thanks for the tip. I'll maintain without `v8`
I think I installed it directly and that's why it wasn't that clear to set up. All good!