gazebosim/gz-sim #2719: Segmentation fault (Address not mapped to object [0x18])

abdvswmdr, 2025-01-14:
On Ubuntu 24.04 with Gazebo Sim (Harmonic), I am trying to load a `.dae` file into Gazebo and save the world as an `.sdf` file. I have followed every step with regards to the .dae creation in Blender and the model.sdf. When I load the custom `.dae` into Gazebo Sim (version 8.7.0 Harmonic), I encounter segmentation faults and mesh loading errors. 

`libEGL warning: egl: failed to create dri2 screen`
`libEGL warning: egl: failed to create dri2 screen`
`libEGL warning: egl: failed to create dri2 screen`
`[Err] [SystemPaths.cc:425] Unable to find file with URI [model://fyp_projectlab2/meshes/fyp_projectlab2.dae]`
`[Err] [SystemPaths.cc:525] Could not resolve file [model://fyp_projectlab2/meshes/fyp_projectlab2.dae]`
`[Err] [MeshManager.cc:211] Unable to find file[model://fyp_projectlab2/meshes/fyp_projectlab2.dae]`
`Segmentation fault (Address not mapped to object [0x18])`
`[GUI] [Err] [GuiFileHandler.cc:108] Service call for generating world SDFormat timed out`
`Unknown error occurred when saving the world. Please check the console output of gz-sim`

[image]

I exported my model (.dae) file path into Gazebo Sim and the model loads correctly with the right materials, appearance and everything. Please help fix this.

azeey, 2025-01-14:
Can you please follow the issue [template](https://github.com/gazebosim/gz-sim/blob/gz-sim9/.github/ISSUE_TEMPLATE/bug_report.md) and fill out the relevant parts including the full logs in text format rather than an image?

abdvswmdr, 2025-01-14:
> Can you please follow the issue [template](https://github.com/gazebosim/gz-sim/blob/gz-sim9/.github/ISSUE_TEMPLATE/bug_report.md) and fill out the relevant parts including the full logs in text format rather than an image?

Sure, I've filled the template you can view. Some things I left blank since I didn't know what to fill

azeey, 2025-01-14:
Sorry, but there is not enough information here to diagnose the problem. Can you copy the full console output from Gazebo (not an image)? Can you share the `.dae` file that's causing the issue? Also, can you create a step-by-step instruction so that someone else can reproduce the issue?