gazebosim/gz-sim #2720: GZ sim crashes: libprotobuf ERROR google/protobuf/descriptor_database.cc:175

DisCoLabIITK, 2025-01-15:
## Environment
* OS Version: 
* Source or binary build?
  
  
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: [ogre | ogre2]. 
      - [ ] Sensor rendering error. 
      - [ *] GUI rendering error. 
    - EGL headless mode:
      - [ ] Running in EGL headless mode 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [ *] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [ ] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [ ] running on real hardware
      - [ ] running in virtual machine
      - [ ] running in Docker/Singularity
      - [ ] running remotely (e.g. via SSH)
      - [ ] running in a cloud
      - [ ] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
      - [ ] GPU is concurrently used for other tasks
        - [ ] desktop acceleration
        - [ ] video decoding (i.e. a playing Youtube video)
        - [ ] video encoding
        - [ ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [ ] multiple simulators running at the same time
      - [ ] other...
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
          ```bash
          LANG=C lspci -nn | grep VGA  # might require installing pciutils
          echo "$DISPLAY"
          LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
          ps aux | grep Xorg
          sudo env LANG=C X -version  # if you don't have root access, try to tell the version of Xorg e.g. via package manager
          ```
      - On Windows, run `dxdiag` and report the GPU-related information.
      - On Mac OS, open a terminal and type `system_profiler SPDisplaysDataType`. 
      -
      - 
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation GA107M [GeForce RTX 3050 Ti Mobile] [10de:25a0] (rev a1)
05:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Rembrandt [1002:1681] (rev c8)
$ echo "$DISPLAY"
:0
$ LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)' 
Command 'glxinfo' not found, but can be installed with:
sudo apt install mesa-utils
$ ps aux | grep Xorg
     19745  0.0  0.0   9720  2488 pts/2    S+   23:27   0:00 grep --color=auto Xorg
$ sudo env LANG=C X -version 

X.Org X Server 1.21.1.4
X Protocol Version 11, Revision 0
Current Operating System: Linux blackbox 6.8.0-51-generic #52~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Dec  9 15:00:52 UTC 2 x86_64
Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.8.0-51-generic root=UUID=9c98fd32-57c7-455a-9c1b-f0a56d16c74f ro quiet splash vt.handoff=7
xorg-server 2:21.1.4-2ubuntu1.7~22.04.12 (For technical support please see http://www.ubuntu.com/support) 
Current version of pixman: 0.40.0
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.

        
    - [ ] Please, attach the ogre.log or ogre2.log file from  `~/.gz/rendering` 

<details>

```
23:02:12: OGRE EXCEPTION(3:RenderingAPIException): Fixed Function pipeline is no longer allowed nor supported. The material highlight_material must use shaders in HlmsLowLevel::calculateHashFor at ./OgreMain/src/OgreHlmsLowLevel.cpp (line 160)
23:02:12: OGRE EXCEPTION(3:RenderingAPIException): Fixed Function pipeline is no longer allowed nor supported. The material highlight_material must use shaders in HlmsLowLevel::calculateHashFor at ./OgreMain/src/OgreHlmsLowLevel.cpp (line 160)
23:02:12: Couldn't apply datablock '[Hash 0x537893c9]' to this renderable. Using default one. Check previous log messages to see if there's more information.
23:02:12: OGRE EXCEPTION(3:RenderingAPIException): Fixed Function pipeline is no longer allowed nor supported. The material highlight_material must use shaders in HlmsLowLevel::calculateHashFor at ./OgreMain/src/OgreHlmsLowLevel.cpp (line 160)
[... 44 lines cut ...]
#23   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7ea0d80906c7, in dlopen
#22   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7ea0d809012d, in 
#21   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7ea0d8174b62, in _dl_catch_error
#20   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7ea0d8174a97, in _dl_catch_exception
#19   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7ea0d809063b, in 
#18   Object "/lib64/ld-linux-x86-64.so.2", at 0x7ea0d884d34d, in 
#17   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7ea0d8174a97, in _dl_catch_exception
#16   Object "/lib64/ld-linux-x86-64.so.2", at 0x7ea0d884cff5, in 
#15   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7ea0d8174af4, in _dl_catch_exception
#14   Object "/lib64/ld-linux-x86-64.so.2", at 0x7ea0d8845567, in 
#13   Object "/lib64/ld-linux-x86-64.so.2", at 0x7ea0d884547d, in 
#12   Object "/lib/x86_64-linux-gnu/libgz-sim7.so.7", at 0x7ea0a2ab826b, in 
#11   Object "/lib/x86_64-linux-gnu/libprotobuf.so.23", at 0x7ea0d1f81951, in google::protobuf::internal::AddDescriptors(google::protobuf::internal::DescriptorTable const*)
#10   Object "/lib/x86_64-linux-gnu/libprotobuf.so.23", at 0x7ea0d1f81966, in google::protobuf::internal::AddDescriptors(google::protobuf::internal::DescriptorTable const*)
#9    Object "/lib/x86_64-linux-gnu/libprotobuf.so.23", at 0x7ea0d1e92c35, in 
#8    Object "/lib/x86_64-linux-gnu/libgcc_s.so.1", at 0x7ea0d7ea72dc, in _Unwind_Resume
#7    Object "/lib/x86_64-linux-gnu/libgcc_s.so.1", at 0x7ea0d7ea6883, in 
#6    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7ea0d3cad958, in __gxx_personality_v0
#5    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7ea0d3cad1e8, in 
#4    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7ea0d3cae20b, in 
#3    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7ea0d3ca2b9d, in 
#2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7ea0d80287f2, in abort
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7ea0d8042475, in raise
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7ea0d80969fc, in pthread_kill
Aborted (Signal sent by tkill() 17879 1000)

azeey, 2025-01-15:
What OS version are you using? It looks like there's an issue with the protobuf version. Do you have a newer protobuf version installed than what comes in Ubuntu by default?

DisCoLabIITK, 2025-01-15:
Hi. I am using Ubuntu 22.04.5 LTS x86_64. I have narrowed down the issue to two libraries: libSurface.so and libWaveVisual.so. If i disable these two the models load and sim doesn't crash.

azeey, 2025-01-15:
Taking a closer look at your error logs, it looks like you have both Gazebo Harmonic (gz-sim8) and Garden (gz-sim7) installed:
```
#26   Object "/lib/x86_64-linux-gnu/libgz-sim8.so.8", at 0x7ea0d2b5ed58, in 
...
#12   Object "/lib/x86_64-linux-gnu/libgz-sim7.so.7", at 0x7ea0a2ab826b, in 
...
```

While it's possible to use them side by side, it might complicate things a bit. Can you try uninstalling Garden? Depending on how you installed things, the following might work

```
sudo apt-get remove gz-garden
sudo apt-get autoremove
```

You'll then have to **clean** and rebuild your workspace to make sure it's using Harmonic.

DisCoLabIITK, 2025-01-16:
I did the above mentioned steps. I noticed a few things. If i build after i delete the logs, build and install folder using "colcon build" the sim loads except for it cannot find the plugins error:
```Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[Err] [SystemLoader.cc:92] Failed to load system plugin [libPublisherPlugin.so] : Could not find shared library.
[Err] [SystemLoader.cc:92] Failed to load system plugin [libWaveVisual.so] : Could not find shared library.
[Err] [SystemLoader.cc:92] Failed to load system plugin [libSurface.so] : Could not find shared library.
[Err] [SystemLoader.cc:92] Failed to load system plugin [libSurface.so] : Could not find shared library.
Warning [Utils.cc:132] [/sdf/plugin[@name="maritime::WaveVisual"]:<data-string>:L1]: XML Element[plugin], child of element[sdf], not defined in SDF. Copying[plugin] as children of [sdf].
[GUI] [Err] [SystemLoader.cc:92] Failed to load system plugin [libWaveVisual.so] : Could not find shared library.
Warning [Utils.cc:132] [/sdf/plugin[@name="maritime::WaveVisual"]:<data-string>:L1]: XML Element[plugin], child of element[sdf], not defined in SDF. Copying[plugin] as children of [sdf].
```
However if i repeat the steps but instead build with "colcon build --merge-install" and start the sim, it crashes with the same protobuf error.