gazebosim/gz-sim #2733: Feature request: ability to updating link mass during runtime.

JOHNI1, 2025-01-25:
I need change the link's mass during runtime. This was possible in gazebo classic through SetMass and UpdateMass functions but I can't find such ability in gz sim.

https://stackoverflow.com/questions/70864861/gazebo-how-can-i-change-model-weigth-during-runtime

https://answers.gazebosim.org/question/19201/

<br/>
<br/>
Additionally I tried to make a plugin for this purpose that updates the link's mass from ros topic but it doesn't work...

CMakeLists.txt:
```cmake
cmake_minimum_required(VERSION 3.10.2 FATAL_ERROR)

project(LinkMassChanger)  #says that doesn't follow naming convention...I don't care
#WARNING: Package name "LinkMassChanger" does not follow the naming conventions. 
#It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.

# Default to C99
if(NOT CMAKE_C_STANDARD)
  set(CMAKE_C_STANDARD 99)
endif()

# Default to C++14
if(NOT CMAKE_CXX_STANDARD)
  set(CMAKE_CXX_STANDARD 14)
endif()

# Find dependencies
find_package(ament_cmake REQUIRED)
find_package(rclcpp REQUIRED)
find_package(std_msgs REQUIRED)
find_package(gz-plugin2 REQUIRED COMPONENTS register)
find_package(gz-sim8 REQUIRED)

# Add plugin library
add_library(${PROJECT_NAME} SHARED
  LinkMassChanger.cc
)

target_include_directories(${PROJECT_NAME} PRIVATE
  ${CMAKE_CURRENT_SOURCE_DIR}
)

ament_target_dependencies(${PROJECT_NAME}
  rclcpp
  std_msgs
  gz-plugin2
  gz-sim8
)

# Install the plugin
install(
  TARGETS ${PROJECT_NAME}
  LIBRARY DESTINATION lib
)

# Register the package
ament_package()
```

LinkMassChanger.hh:
```cpp
#ifndef SYSTEM_PLUGIN_LINKMASSCHANGER_HH_
#define SYSTEM_PLUGIN_LINKMASSCHANGER_HH_

#include <gz/sim/System.hh>
#include <gz/sim/Entity.hh>
#include <gz/transport/Node.hh>

[... 358 lines cut ...]

package.xml:
```xml
<?xml version="1.0"?>
<package format="3">
  <name>LinkMassChanger</name> <!--says that doesn't follow naming convention...I don't care-->
  <!-- WARNING: Package name "LinkMassChanger" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes. -->
  <version>0.0.0</version>
  <description>Ground Collision Gazebo Harmonic Plugin</description>

  <maintainer email="takahashiyoni@gmail.com">takahashi yoni</maintainer>
  <license>Apache 2.0</license>

  <buildtool_depend>ament_cmake</buildtool_depend>

  <depend>rclcpp</depend>
  <depend>std_msgs</depend>
  <depend>gazebo_dev</depend>
  <depend>gazebo_plugins</depend>

  <export>
    <build_type>ament_cmake</build_type>
  </export>
</package>
```

azeey, 2025-01-31:
There is no API to do update the mass of a link at runtime. The pattern of using components to update values typically starts with creating a `<Property>Cmd` component and having one of the core systems like the `Physics` system be responsible for handling that component. For example, to change the force applied by a joint, there's a `JointForceCmd` (https://github.com/gazebosim/gz-sim/blob/1739bdbeaff9118813cd447fcb442e80ec0f846a/include/gz/sim/components/JointForceCmd.hh#L1) component, which other systems can set to update the joint force (https://github.com/gazebosim/gz-sim/blob/1739bdbeaff9118813cd447fcb442e80ec0f846a/src/systems/joint_controller/JointController.cc#L327-L338), and the `Physics` system takes care of applying the force in the physics engine https://github.com/gazebosim/gz-sim/blob/1739bdbeaff9118813cd447fcb442e80ec0f846a/src/systems/physics/Physics.cc#L2317-L2335 

So, if you're interested in working on this, I'd recommend starting by creating `InertialCmd` component and updating the `Physics` system to apply the new inertial in the physics engine.

JOHNI1, 2025-02-03:
Thank you for your response and guidance.

From what I understand, updating a link’s mass at runtime requires an InertialCmd component, similar to JointForceCmd. However, since the Physics system doesn’t process InertialCmd by default, I’d need to modify it to apply these updates to the physics engine. This is why changes to the Physics system are necessary, and I assume that's why this issue was moved to Core development. Essentially, inertial data remains static after being loaded, so modifying Physics is required to enable runtime updates.

I appreciate your insights and would love to hear your thoughts.

JOHNI1, 2025-02-03:
This problem also reminds me when I tried to move a model's position that I set to be static in sdf, as I changing its position, the changes did reflect in the gui (even when I set to show the collider of the model) but the actual collider that was registered to the physics system did not update the position and the the invisible static object stayed in the same place and other models would collide with it. So I guess the same solution is needed for this problem too, to make a Cmd that updates the Component values in Physics System.

Mat198, 2026-10-19:
Is there anyone working on this? I would like this feature to simulate an autonomous sprayer tank dynamics. I think I can help

azeey, 2026-10-19:
I don't hink anyone is working on it.

anirudh12032008, 2026-10-21:
Hi @azeey, @Mat198, I'd like to work on this. Mat198, you mentioned interest on Oct 19; I haven't seen a branch or PR yet, so happy to split it: I can take the gz-physics feature and you the gz-sim side (or the other way round), whichever you prefer.

I looked at the code: gz-physics has no inertial setter on Link (only `AddedMass::SetAddedMass`), and gz-sim's `Physics.cc` has no `InertialCmd` handling. Plan: 1) gz-physics: a new link feature (like `AddedMass.hh`) with `SetInertial`, implemented in dartsim using the existing `AssignInertialToBody` in `Base.hh` (and updating `LinkInfo::inertial`); 2) gz-sim: an `InertialCmd` component (`math::Inertiald`), consumed in the Physics system and removed afterwards like `JointVelocityCmd`; 3) a test in `physics_system.cc`.

Questions: should the physics system also update the `Inertial` component so other systems see the new value? And should `InertialCmd` carry a full `Inertiald`, or is mass-only enough?

Mat198, 2026-10-21:
@anirudh12032008, I'm working on this feature and I'm near of finishing it. I'm doing the testing right now. I'm glad if you can help with the review process. I think it's more productive if you work on other issue in the meanwhile.
 
I should have commented that I was working on it but I thought other people wouldn't have interest since it was stale for more than a year

anirudh12032008, 2026-10-22:
Great, go ahead then, I'll leave it to you. Thanks for the update.