gazebosim/gz-sim #2737: Decoupling the server from the client libraries

j-rivero, 2025-01-27:
While working on https://github.com/gazebo-release/gz-sim8-release/issues/8 I found some difficulties into splitting the packaging from client/server in gz-sim. Note that we mainly want to split the GUI libraries (specially those that needs Qt) into a separate package.

Reviewing the situation: there are two libraries that links against Qt:
```
/usr/lib/x86_64-linux-gnu/libgz-sim8-gui.so.8.6.0
	libQt5Widgets.so.5 => /lib/x86_64-linux-gnu/libQt5Widgets.so.5 (0x000074453e000000)
	libQt5Quick.so.5 => /lib/x86_64-linux-gnu/libQt5Quick.so.5 (0x000074453da00000)
	libQt5Qml.so.5 => /lib/x86_64-linux-gnu/libQt5Qml.so.5 (0x000074453d400000)
	libQt5Gui.so.5 => /lib/x86_64-linux-gnu/libQt5Gui.so.5 (0x000074453cc00000)
	libQt5Core.so.5 => /lib/x86_64-linux-gnu/libQt5Core.so.5 (0x000074453c600000)
	libQt5QmlModels.so.5 => /lib/x86_64-linux-gnu/libQt5QmlModels.so.5 (0x000074453d892000)
	libQt5Network.so.5 => /lib/x86_64-linux-gnu/libQt5Network.so.5 (0x000074453b454000)
...
/usr/lib/x86_64-linux-gnu/libgz-sim8-gz.so.8.6.0
	libQt5Core.so.5 => /lib/x86_64-linux-gnu/libQt5Core.so.5 (0x0000792777000000)
	libQt5Widgets.so.5 => /lib/x86_64-linux-gnu/libQt5Widgets.so.5 (0x0000792776000000)
	libQt5Quick.so.5 => /lib/x86_64-linux-gnu/libQt5Quick.so.5 (0x0000792775a00000)
	libQt5Qml.so.5 => /lib/x86_64-linux-gnu/libQt5Qml.so.5 (0x0000792775400000)
	libQt5Gui.so.5 => /lib/x86_64-linux-gnu/libQt5Gui.so.5 (0x0000792774c00000)
	libQt5QmlModels.so.5 => /lib/x86_64-linux-gnu/libQt5QmlModels.so.5 (0x0000792776aba000)
	libQt5Network.so.5 => /lib/x86_64-linux-gnu/libQt5Network.so.5 (0x0000792773c54000)
```

While the gz-sim8-gui is expected to use the Qt symbols and we can package it in a separate GUI package the 
libgz-sim8-gz library is probably use by the calls to server and client at the same time.

If I'm not wrong the related code is:
https://github.com/gazebosim/gz-sim/blob/4c2797d335c3d44128923c1315a051a5c922ebaa/src/CMakeLists.txt#L212-L229

Question to answer is: is there a real need to link the gz CLI to these GUI libraries or we can remove the linking? If there is a real need, why we can not run the `gz sim -s` without the Qt libraries?

azeey, 2025-01-31:
I think there are two issues:
- There's a `runGui` command inside `gz.cc` that launches the GUI (https://github.com/gazebosim/gz-sim/blob/1739bdbeaff9118813cd447fcb442e80ec0f846a/src/gz.cc#L439). This brings in all the Qt libraries.
-  the CLI target is linked against the core library (https://github.com/gazebosim/gz-sim/blob/1739bdbeaff9118813cd447fcb442e80ec0f846a/src/CMakeLists.txt#L183), which itself is linked against gz-gui (https://github.com/gazebosim/gz-sim/blob/1739bdbeaff9118813cd447fcb442e80ec0f846a/src/CMakeLists.txt#L220)

I think the best option is to accelerate https://github.com/gazebosim/gz-tools/issues/7, which will give us the option to have separate binaries for the server and the GUI. @sauk2 is already working on this, so it'd be great if we focus our efforts on supporting him.

j-rivero, 2025-05-16:
After merging https://github.com/gazebosim/gz-sim/pull/2849 we have the ability to skip the GUI building at compilation time using `WITH_GUI` flag. In the standard build we are still using the gui component to built and link against it in the gz-sim-main binary:

```
libexec/gz/sim10 ❯ ldd gz-sim-main  | grep gui
	libgz-sim-gui.so.10 => /home/jrivero/code/gz/ws_standalone/install/lib/libgz-sim-gui.so.10 (0x000075cae5f38000)
	libgz-gui.so.10 => /home/jrivero/code/gz/ws_standalone/install/lib/libgz-gui.so.10 (0x000075cae5547000)

libexec/gz/sim10 ❯ nm -D gz-sim-main   | grep gui
                 U _ZN2gz3sim3v103gui6runGuiERiPPcPKcS7_iS7_S7_
```