gazebosim/gz-sim #2746: Extend Visualize Frustum GUI plugin to support camera sensors

iche033, 2025-01-30:
Visualize Frustum GUI plugin (https://github.com/gazebosim/gz-sim/pull/2707) adds frustum visualization for logical camera sensors by subscribing to the logical camera sensor's frustum topic, e.g. /logical_cam_sensor/frustum (https://github.com/gazebosim/gz-sensors/pull/491). One improvement is to extend the plugin to support visualization of other types of camera sensors, starting with a regular RGB camera sensor.

## Desired behavior
The GUI plugin allows users to select other camera topics for visualizing frustums 

## Alternatives considered
Another option for enabling/disabling frustum visualization is to parse the [//sensor/visualize](http://sdformat.org/spec?ver=1.12&elem=sensor#sensor_visualize) sdf tag

## Implementation suggestion
In the topic drop down list, we can also support [gz.msgs.CameraInfo](https://github.com/gazebosim/gz-msgs/blob/gz-msgs11/proto/gz/msgs/camera_info.proto) msg types. On callback, we should be able to compute camera frustum info (near, far, hfov, etc) from the intrinsic params.

Mehul0x, 2025-04-03:
Hi! 
Can I be alloted this issue?

Mehul0x, 2025-04-06:
Hi!
I have gone through the structure of Gazebo and worked out the details, as to what I have to change to add frustum to other cameras but I have ran into 2 problems

1. 

[image]

I can't visualize the frustum for logical camera 

2.  I can't open examples from cmd line

[image]

I already have  added these lines to /. bashrc 
 
```
GZ_CONFIG_PATH=/home/aerosheesh-24/workspace/install/share/gz/

source /home/aerosheesh-24/workspace/install/setup.sh 
```

iche033, 2025-04-07:
just double checking, for 1), having you tried running the simulation by clicking the Play button on the bottom left of the window?

As for 2), what's the directory that you're in from you run that cmd? Looking at the path you specified in the cmd line, you'll need to run that from the root of the `gz-sim` directory, which is probably `/home/aerosheesh-24/workspace/src/gz-sim`

shashank1300, 2025-04-12:
Hey Ian,
I was thinking of working on this issue (already started).