gazebosim/gz-sim #2762: [ruby $(which gz) sim-1] [Err] [UserCommands.cc:1152] Error Code 9: Msg: The norm of the xyz vector cannot be zero

siddarth09, 2025-02-08:
[INFO] [launch]: All log files can be found below /home/siddarth/.ros/log/2025-02-08-03-10-57-199975-maverick-14437
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ruby $(which gz) sim-1]: process started with pid [14440]
[INFO] [create-2]: process started with pid [14442]
[INFO] [parameter_bridge-3]: process started with pid [14445]
[INFO] [image_bridge-4]: process started with pid [14448]
[INFO] [relay-5]: process started with pid [14450]
[INFO] [robot_state_publisher-6]: process started with pid [14452]
[create-2] [INFO] [1739002258.571289859] [ros_gz_sim]: Requesting list of world names.
[parameter_bridge-3] [INFO] [1739002258.628556374] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/joint_states (gz.msgs.Model) -> /joint_states (sensor_msgs/msg/JointState)] (Lazy 0)
[parameter_bridge-3] [INFO] [1739002258.667718258] [ros_gz_bridge]: Creating ROS->GZ Bridge: [/joint_states (sensor_msgs/msg/JointState) -> /joint_states (gz.msgs.Model)] (Lazy 0)
[robot_state_publisher-6] [INFO] [1739002258.806099044] [robot_state_publisher]: got segment LF_FOOT
[robot_state_publisher-6] [INFO] [1739002258.806550931] [robot_state_publisher]: got segment LF_calf
[robot_state_publisher-6] [INFO] [1739002258.806594851] [robot_state_publisher]: got segment LF_fut
[robot_state_publisher-6] [INFO] [1739002258.806616995] [robot_state_publisher]: got segment LF_hip
[robot_state_publisher-6] [INFO] [1739002258.806637585] [robot_state_publisher]: got segment LF_thigh
[robot_state_publisher-6] [INFO] [1739002258.806656041] [robot_state_publisher]: got segment LH_FOOT
[robot_state_publisher-6] [INFO] [1739002258.806674472] [robot_state_publisher]: got segment LH_calf
[robot_state_publisher-6] [INFO] [1739002258.806720791] [robot_state_publisher]: got segment LH_fut
[robot_state_publisher-6] [INFO] [1739002258.806757043] [robot_state_publisher]: got segment LH_hip
[robot_state_publisher-6] [INFO] [1739002258.806791100] [robot_state_publisher]: got segment LH_thigh
[robot_state_publisher-6] [INFO] [1739002258.806819582] [robot_state_publisher]: got segment RF_FOOT
[robot_state_publisher-6] [INFO] [1739002258.806839024] [robot_state_publisher]: got segment RF_calf
[robot_state_publisher-6] [INFO] [1739002258.806856491] [robot_state_publisher]: got segment RF_fut
[robot_state_publisher-6] [INFO] [1739002258.806874353] [robot_state_publisher]: got segment RF_hip
[robot_state_publisher-6] [INFO] [1739002258.806893131] [robot_state_publisher]: got segment RF_thigh
[robot_state_publisher-6] [INFO] [1739002258.806910990] [robot_state_publisher]: got segment RH_FOOT
[robot_state_publisher-6] [INFO] [1739002258.806929165] [robot_state_publisher]: got segment RH_calf
[robot_state_publisher-6] [INFO] [1739002258.806946883] [robot_state_publisher]: got segment RH_fut
[robot_state_publisher-6] [INFO] [1739002258.806965643] [robot_state_publisher]: got segment RH_hip
[robot_state_publisher-6] [INFO] [1739002258.806983809] [robot_state_publisher]: got segment RH_thigh
[robot_state_publisher-6] [INFO] [1739002258.807001807] [robot_state_publisher]: got segment base
[robot_state_publisher-6] [INFO] [1739002258.807020339] [robot_state_publisher]: got segment bl_fibula
[robot_state_publisher-6] [INFO] [1739002258.807038929] [robot_state_publisher]: got segment br_fibula
[robot_state_publisher-6] [INFO] [1739002258.807056261] [robot_state_publisher]: got segment camera_link
[robot_state_publisher-6] [INFO] [1739002258.807085797] [robot_state_publisher]: got segment camera_link_optical
[robot_state_publisher-6] [INFO] [1739002258.807118642] [robot_state_publisher]: got segment fl_fibula
[robot_state_publisher-6] [INFO] [1739002258.807149914] [robot_state_publisher]: got segment fr_fibula
[robot_state_publisher-6] [INFO] [1739002258.807184481] [robot_state_publisher]: got segment front_left_blade
[robot_state_publisher-6] [INFO] [1739002258.807217830] [robot_state_publisher]: got segment front_right_blade
[robot_state_publisher-6] [INFO] [1739002258.807248183] [robot_state_publisher]: got segment hussain_imu
[robot_state_publisher-6] [INFO] [1739002258.807279823] [robot_state_publisher]: got segment rear_left_blade
[robot_state_publisher-6] [INFO] [1739002258.807313871] [robot_state_publisher]: got segment rear_right_blade
[robot_state_publisher-6] [INFO] [1739002258.807347040] [robot_state_publisher]: got segment root
[robot_state_publisher-6] [INFO] [1739002258.807370116] [robot_state_publisher]: got segment unitree_imu
[create-2] [INFO] [1739002263.572379383] [ros_gz_sim]: Requesting list of world names.
[create-2] [INFO] [1739002264.290940733] [ros_gz_sim]: Waiting messages on topic [robot_description].
[create-2] [INFO] [1739002264.302492913] [ros_gz_sim]: Requested creation of entity.
[create-2] [INFO] [1739002264.302536126] [ros_gz_sim]: OK creation of entity.
[INFO] [create-2]: process has finished cleanly [pid 14442]
[ruby $(which gz) sim-1] [Err] [UserCommands.cc:1152] Error Code 9: Msg: The norm of the xyz vector cannot be zero
[ruby $(which gz) sim-1] [Err] [UserCommands.cc:1152] Error Code 9: Msg: The norm of the xyz vector cannot be zero
[ruby $(which gz) sim-1] [Err] [UserCommands.cc:1152] Error Code 9: Msg: The norm of the xyz vector cannot be zero
[ruby $(which gz) sim-1] [Err] [UserCommands.cc:1152] Error Code 9: Msg: The norm of the xyz vector cannot be zero
[ruby $(which gz) sim-1] [Err] [UserCommands.cc:1152] Error Code 9: Msg: The norm of the xyz vector cannot be zero
[ruby $(which gz) sim-1] [Err] [UserCommands.cc:1152] Error Code 9: Msg: The norm of the xyz vector cannot be zero

siddarth09, 2025-02-11:
Can someone help me with this ?

arjo129, 2025-02-11:
Can you at least give which version of gazebo you are using? Sounds like you are not filling in the `JointAxis` parameter correctly in some SDF model. I suspect the error comes from here:
https://github.com/gazebosim/sdformat/blob/2b583cbc3f40ac4f1af27613a49b475f2d0d99b7/src/JointAxis.cc#L285

siddarth09, 2025-02-11:
I am currently trying to simulate a custom quadruped on gazebo harmonic. I am using a urdf and launching using python. 

https://github.com/siddarth09/husky/tree/husky_update (you can check the urdf file here, hoping that gives clarity).

arjo129, 2025-02-11:
Is it possible for you to provide a smaller reproducible snippet? I'm guessing somewhere along the way one of the `<joint><axis></axis></joint>` tags is not being set correctly.