gazebosim/gz-sim #2765: All rendering are aborted using gz sim in wsl2

j3anm1rc, 2025-02-11:
I follow the installation tutorial and try to run defaults model in gz sim from :
#<https://gazebosim.org/api/sim/9/install.html>
I run an ubuntu os inside wsl2. Any rendering are aborted with the following error:
```
: CommandLine Error: Option 'spirv-expand-step' registered more than once!
LLVM ERROR: inconsistency in registered CommandLine options
```
## Environment
* OS Version:  Ubuntu 22.04.5 LTS  in wsl2  2.4.10
* Source or binary build?

- gz-sim9-cli/unknown,now 9.0.0-1~jammy amd64 [installed,automatic]
- libgz-sim9-dev/unknown,now 9.0.0-1~jammy amd64 [installed]
- libgz-sim9/unknown,now 9.0.0-1~jammy amd64 [installed,automatic]
- python3-gz-sim9/unknown,now 9.0.0-1~jammy amd64 [installed,automatic]

* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: [ogre | ogre2]. 
      - [ ] Sensor rendering error. 
      - [ ] GUI rendering error. 
    - EGL headless mode:
      - [ ] Running in EGL headless mode 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [ ] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [ ] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [ ] running on real hardware
      - [X ] running in virtual machine
      - [ ] running in Docker/Singularity
      - [ ] running remotely (e.g. via SSH)
      - [ ] running in a cloud
      - [ ] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
      - [ ] GPU is concurrently used for other tasks
        - [ ] desktop acceleration
        - [ ] video decoding (i.e. a playing Youtube video)
        - [ ] video encoding
        - [ ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [ ] multiple simulators running at the same time
      - [ ] other...
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
          ```bash
redgy@local:~$ echo "$DISPLAY"
:0
redgy@local:~$ glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'
direct rendering: Yes
    Preferred profile: core (0x1)
    Max core profile version: 4.1
    Max compat profile version: 4.1
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.0
OpenGL vendor string: Microsoft Corporation
OpenGL renderer string: D3D12 (Intel(R) UHD Graphics)
OpenGL core profile version string: 4.1 (Core Profile) Mesa 23.2.1-1ubuntu3.1~22.04.3
OpenGL core profile shading language version string: 4.10
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL version string: 4.1 (Compatibility Profile) Mesa 23.2.1-1ubuntu3.1~22.04.3
OpenGL shading language version string: 4.10
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile
OpenGL ES profile version string: OpenGL ES 3.0 Mesa 23.2.1-1ubuntu3.1~22.04.3
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.00
redgy@local:~$ ps aux | grep Xorg
redgy      15745  0.0  0.0   4032  2004 pts/0    S+   11:59   0:00 grep --color=auto Xorg
redgy@local:~$ sudo env LANG=C X -version
env: ‘X’: No such file or directory
redgy@local:~$
          ```

<details>

```
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
: CommandLine Error: Option 'spirv-expand-step' registered more than once!
LLVM ERROR: inconsistency in registered CommandLine options
Stack trace (most recent call last):
#31   Object "/usr/lib/wsl/lib/libd3d12core.so", at 0x7f3526c7ce3f, in
#30   Object "/usr/lib/wsl/lib/libd3d12core.so", at 0x7f3526c7b488, in
#29   Object "/usr/lib/wsl/lib/libd3d12core.so", at 0x7f3526c7a87f, in
#28   Object "/usr/lib/wsl/lib/libd3d12core.so", at 0x7f3526c7ab2b, in
#27   Object "/usr/lib/wsl/lib/libd3d12core.so", at 0x7f3526d4b6a7, in
#26   Object "/usr/lib/wsl/lib/libd3d12core.so", at 0x7f3526d5f780, in
#25   Object "/usr/lib/wsl/drivers/iigd_dch.inf_amd64_1822c4fce94b7cb9/libigd12umd64.so", at 0x7f35275e435e, in OpenAdapter12
#24   Object "/usr/lib/wsl/drivers/iigd_dch.inf_amd64_1822c4fce94b7cb9/libigd12umd64.so", at 0x7f35275e456d, in
#23   Object "/usr/lib/wsl/drivers/iigd_dch.inf_amd64_1822c4fce94b7cb9/libigd12um64xel.so", at 0x7f351b41282c, in
#22   Object "/usr/lib/wsl/drivers/iigd_dch.inf_amd64_1822c4fce94b7cb9/libigd12um64xel.so", at 0x7f351b48a942, in
#21   Object "/usr/lib/wsl/drivers/iigd_dch.inf_amd64_1822c4fce94b7cb9/libigd12um64xel.so", at 0x7f351b4b29fc, in
#20   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f35444446c7, in dlopen
#19   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f354444412d, in
#18   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f3544528b62, in _dl_catch_error
#17   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f3544528a97, in _dl_catch_exception
#16   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f354444463b, in
#15   Object "/lib64/ld-linux-x86-64.so.2", at 0x7f354497634d, in
#14   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f3544528a97, in _dl_catch_exception
#13   Object "/lib64/ld-linux-x86-64.so.2", at 0x7f3544975ff5, in
#12   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f3544528af4, in _dl_catch_exception
#11   Object "/lib64/ld-linux-x86-64.so.2", at 0x7f354496e567, in
#10   Object "/lib64/ld-linux-x86-64.so.2", at 0x7f354496e47d, in
#9    Object "/usr/lib/wsl/drivers/iigd_dch.inf_amd64_1822c4fce94b7cb9/libigc.so", at 0x7f350987ff82, in
#8    Object "/usr/lib/wsl/drivers/iigd_dch.inf_amd64_1822c4fce94b7cb9/libigc.so", at 0x7f3509c65a1a, in
#7    Object "/usr/lib/wsl/drivers/iigd_dch.inf_amd64_1822c4fce94b7cb9/libLLVM-14.so", at 0x7f351909bbdf, in llvm::cl::Option::addArgument()
#6    Object "/usr/lib/wsl/drivers/iigd_dch.inf_amd64_1822c4fce94b7cb9/libLLVM-14.so", at 0x7f351909bb5a, in
#5    Object "/usr/lib/wsl/drivers/iigd_dch.inf_amd64_1822c4fce94b7cb9/libLLVM-14.so", at 0x7f351909b80e, in
#4    Object "/usr/lib/wsl/drivers/iigd_dch.inf_amd64_1822c4fce94b7cb9/libLLVM-14.so", at 0x7f35190adf9b, in llvm::report_fatal_error(char const*, bool)
#3    Object "/usr/lib/wsl/drivers/iigd_dch.inf_amd64_1822c4fce94b7cb9/libLLVM-14.so", at 0x7f35190adee8, in llvm::report_fatal_error(llvm::Twine const&, bool)
#2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f35443dc7f2, in abort
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f35443f6475, in raise
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f354444a9fc, in pthread_kill
Aborted (Signal sent by tkill() 10511 0)

```

</details>

## Description
* Expected behavior: rendering defaults worlds after following the installation step
* Actual behavior: rendering aborted 

## Steps to reproduce

1. follow the installation describe by #<https://gazebosim.org/api/sim/9/install.html> inside a wsl 2 with ubuntu
2.  launch gz sim
3. try to run any world

## Output

My LLVM pakages
```
libllvmspirvlib12/jammy,now 12.0.0-3 amd64 [installed,automatic]
libllvmspirvlib13/jammy,now 13.0.0-3 amd64 [installed,automatic]
libllvmspirvlib14/jammy-updates,now 14.0.0-3~ubuntu0.22.04.1 amd64 [installed,automatic]
libllvmspirvlib15/jammy-updates,jammy-security,now 15.0.0-1~ubuntu0.22.04.2 amd64 [installed,automatic]
```

WSL config
`[wsl2]

localhostforwarding=true

debugConsole=true

# Sets additional kernel parameters, in this case enabling older Linux base images such as Centos 6
kernelCommandLine = vsyscall=emulate
firewall=false
networkingMode=mirrored
[experimental]
hostAddressLoopback=true
bestEffortDnsParsing=true`

traversaro, 2025-02-11:
What happens if you try to force software rendering?

j3anm1rc, 2025-02-12:
> What happens if you try to force software rendering?

What I understand is that gazebo uses dart as physical rendering  and ogre2 as engine for 3d rendering.

I have these libs, they are maybe inconsistent ?
``bash
#/usr/lib/x86_64-linux-gnu/gz-physics-8/engine-plugins/libgz-physics-dartsim-plugin.so
 export GZ_SIM_PHYSICS_ENGINE_PATH=/usr/lib/x86_64-linux-gnu/gz-physics-8/engine-plugins/

#/usr/lib/x86_64-linux-gnu/gz-rendering-9/engine-plugins/libgz-rendering-ogre.so
export GZ_SIM_RENDER_ENGINE_PATH=/usr/lib/x86_64-linux-gnu/gz-rendering-9/engine-plugins/

```
I tried  to launch the gz sim defining any renderer options:

```
$ sudo gz sim  --verbose --render-engine-api-backend opengl
(2025-02-12 11:33:07.338) [info] [Gui.cc:332] Gazebo Sim GUI    v9.0.0
(2025-02-12 11:33:07.556) [info] [Gui.cc:246] Gazebo Sim Quick start dialog
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
(2025-02-12 11:33:13.261) [info] [gz.cc:170] Received world [empty.sdf] from the GUI.
(2025-02-12 11:33:13.263) [info] [gz.cc:337] Gazebo Sim Server v9.0.0
(2025-02-12 11:33:13.269) [warning] [Application.cc:933] [GUI] [QT] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
(2025-02-12 11:33:13.292) [info] [Server.cc:145] Loading SDF world file[/usr/share/gz/gz-sim9/worlds/empty.sdf].
: CommandLine Error: Option 'spirv-expand-step' registered more than once!
LLVM ERROR: inconsistency in registered CommandLine options
(2025-02-12 11:33:13.375) [error] [SignalHandler.cc:178] [GUI] 4 Interrupted system call
Stack trace (most recent call last):
#31   Object "/usr/lib/wsl/lib/libd3d12core.so", at 0x7f346949de3f, in
#30   Object "/usr/lib/wsl/lib/libd3d12core.so", at 0x7f346949c488, in
#29   Object "/usr/lib/wsl/lib/libd3d12core.so", at 0x7f346949b87f, in
#28   Object "/usr/lib/wsl/lib/libd3d12core.so", at 0x7f346949bb2b, in
#27   Object "/usr/lib/wsl/lib/libd3d12core.so", at 0x7f346956c6a7, in
#26   Object "/usr/lib/wsl/lib/libd3d12core.so", at 0x7f3469580780, in
#25   Object "/usr/lib/wsl/drivers/iigd_dch.inf_amd64_1822c4fce94b7cb9/libigd12umd64.so", at 0x7f3469e0135e, in OpenAdapter12
#24   Object "/usr/lib/wsl/drivers/iigd_dch.inf_amd64_1822c4fce94b7cb9/libigd12umd64.so", at 0x7f3469e0156d, in
#23   Object "/usr/lib/wsl/drivers/iigd_dch.inf_amd64_1822c4fce94b7cb9/libigd12um64xel.so", at 0x7f344a74b82c, in
#22   Object "/usr/lib/wsl/drivers/iigd_dch.inf_amd64_1822c4fce94b7cb9/libigd12um64xel.so", at 0x7f344a7c3942, in
#21   Object "/usr/lib/wsl/drivers/iigd_dch.inf_amd64_1822c4fce94b7cb9/libigd12um64xel.so", at 0x7f344a7eb9fc, in
#20   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f347ec6a6c7, in dlopen
#19   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f347ec6a12d, in
#18   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f347ed4eb62, in _dl_catch_error
#17   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f347ed4ea97, in _dl_catch_exception
#16   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f347ec6a63b, in
#15   Object "/lib64/ld-linux-x86-64.so.2", at 0x7f347f19d34d, in
#14   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f347ed4ea97, in _dl_catch_exception
#13   Object "/lib64/ld-linux-x86-64.so.2", at 0x7f347f19cff5, in
#12   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f347ed4eaf4, in _dl_catch_exception
#11   Object "/lib64/ld-linux-x86-64.so.2", at 0x7f347f195567, in
#10   Object "/lib64/ld-linux-x86-64.so.2", at 0x7f347f19547d, in
#9    Object "/usr/lib/wsl/drivers/iigd_dch.inf_amd64_1822c4fce94b7cb9/libigc.so", at 0x7f3446f8cf82, in
#8    Object "/usr/lib/wsl/drivers/iigd_dch.inf_amd64_1822c4fce94b7cb9/libigc.so", at 0x7f3447372a1a, in
#7    Object "/usr/lib/wsl/drivers/iigd_dch.inf_amd64_1822c4fce94b7cb9/libLLVM-14.so", at 0x7f3444c17bdf, in llvm::cl::Option::addArgument()
#6    Object "/usr/lib/wsl/drivers/iigd_dch.inf_amd64_1822c4fce94b7cb9/libLLVM-14.so", at 0x7f3444c17b5a, in
#5    Object "/usr/lib/wsl/drivers/iigd_dch.inf_amd64_1822c4fce94b7cb9/libLLVM-14.so", at 0x7f3444c1780e, in
#4    Object "/usr/lib/wsl/drivers/iigd_dch.inf_amd64_1822c4fce94b7cb9/libLLVM-14.so", at 0x7f3444c29f9b, in llvm::report_fatal_error(char const*, bool)
#3    Object "/usr/lib/wsl/drivers/iigd_dch.inf_amd64_1822c4fce94b7cb9/libLLVM-14.so", at 0x7f3444c29ee8, in llvm::report_fatal_error(llvm::Twine const&, bool)
#2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f347ec027f2, in abort
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f347ec1c475, in raise
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f347ec709fc, in pthread_kill
Aborted (Signal sent by tkill() 7962 0)
(2025-02-12 11:33:13.452) [info] [SystemManager.cc:54] Serving entity system service on [/entity/system/add]
(2025-02-12 11:33:13.586) [info] [UserCommands.cc:664] Create service on [/world/empty/create]
(2025-02-12 11:33:13.586) [info] [UserCommands.cc:671] Remove service on [/world/empty/remove]
(2025-02-12 11:33:13.586) [info] [UserCommands.cc:678] Pose service on [/world/empty/set_pose]
(2025-02-12 11:33:13.586) [info] [UserCommands.cc:686] Pose service on [/world/empty/set_pose_vector]
(2025-02-12 11:33:13.586) [info] [UserCommands.cc:693] Light configuration service on [/world/empty/light_config]
(2025-02-12 11:33:13.586) [info] [UserCommands.cc:719] Physics service on [/world/empty/set_physics]
(2025-02-12 11:33:13.586) [info] [UserCommands.cc:727] SphericalCoordinates service on [/world/empty/set_spherical_coordinates]
[... cut ...]
(2025-02-12 11:34:24.238) [info] [SceneBroadcaster.cc:680] Publishing entity deletions on [/world/nao_world/scene/deletion]
(2025-02-12 11:34:24.238) [info] [SceneBroadcaster.cc:689] Publishing state changes on [/world/nao_world/state]
(2025-02-12 11:34:24.238) [info] [SceneBroadcaster.cc:700] Publishing pose messages on [/world/nao_world/pose/info]
(2025-02-12 11:34:24.238) [info] [SceneBroadcaster.cc:711] Publishing dynamic pose messages on [/world/nao_world/dynamic_pose/info]
```

same behaviours ...

traversaro, 2025-02-12:
Actually I meant to set `LIBGL_ALWAYS_SOFTWARE=1` env variable so that we understand if the issue is in your hardware-accelerated graphics rendering. However, I am a bit confused about your settings:
* Why are you running `gz sim` under `sudo`?  You should be able to just run `gz sim`.
* Why are you explicitly setting `GZ_SIM_PHYSICS_ENGINE_PATH`  ? The default physics engine should work out of the box.

More in general, is this a vanilla WSL2 Ubuntu 22.04 or there is something strange about it?

j3anm1rc, 2025-02-12:
> Actually I meant to set `LIBGL_ALWAYS_SOFTWARE=1` env variable so that we understand if the issue is in your hardware-accelerated graphics rendering. 
 
With  `LIBGL_ALWAYS_SOFTWARE=1` gz sim does run properly despite some warnings  

```
$ gz sim --verbose
(2025-02-12 14:46:50.403) [info] [Gui.cc:332] Gazebo Sim GUI    v9.0.0
(2025-02-12 14:46:50.639) [info] [Gui.cc:246] Gazebo Sim Quick start dialog
QStandardPaths: wrong permissions on runtime directory /run/user/1000/, 0755 instead of 0700
(2025-02-12 14:46:56.224) [info] [gz.cc:170] Received world [empty.sdf] from the GUI.
(2025-02-12 14:46:56.225) [info] [gz.cc:337] Gazebo Sim Server v9.0.0
(2025-02-12 14:46:56.230) [warning] [Application.cc:933] [GUI] [QT] QStandardPaths: wrong permissions on runtime directory /run/user/1000/, 0755 instead of 0700
(2025-02-12 14:46:56.234) [info] [Server.cc:145] Loading SDF world file[/usr/share/gz/gz-sim9/worlds/empty.sdf].
(2025-02-12 14:46:56.387) [info] [SystemManager.cc:54] Serving entity system service on [/entity/system/add]
(2025-02-12 14:46:56.465) [info] [UserCommands.cc:664] Create service on [/world/empty/create]
(2025-02-12 14:46:56.465) [info] [UserCommands.cc:671] Remove service on [/world/empty/remove]
(2025-02-12 14:46:56.465) [info] [UserCommands.cc:678] Pose service on [/world/empty/set_pose]
(2025-02-12 14:46:56.465) [info] [UserCommands.cc:686] Pose service on [/world/empty/set_pose_vector]
(2025-02-12 14:46:56.466) [info] [UserCommands.cc:693] Light configuration service on [/world/empty/light_config]
(2025-02-12 14:46:56.466) [info] [UserCommands.cc:719] Physics service on [/world/empty/set_physics]
(2025-02-12 14:46:56.466) [info] [UserCommands.cc:727] SphericalCoordinates service on [/world/empty/set_spherical_coordinates]
(2025-02-12 14:46:56.466) [info] [UserCommands.cc:736] Enable collision service on [/world/empty/enable_collision]
(2025-02-12 14:46:56.466) [info] [UserCommands.cc:745] Disable collision service on [/world/empty/disable_collision]
(2025-02-12 14:46:56.466) [info] [UserCommands.cc:754] Material service on [/world/empty/visual_config]
(2025-02-12 14:46:56.466) [info] [UserCommands.cc:762] Material service on [/world/empty/wheel_slip]
(2025-02-12 14:46:56.468) [info] [LevelManager.cc:641] Loaded level [default]
(2025-02-12 14:46:56.468) [info] [SimulationRunner.cc:260] Serving world controls on [/world/empty/control], [/world/empty/control/state] and [/world/empty/playback/control]
(2025-02-12 14:46:56.468) [info] [SimulationRunner.cc:286] Serving GUI information on [/world/empty/gui/info]
(2025-02-12 14:46:56.468) [info] [SimulationRunner.cc:289] World [empty] initialized with [1ms] physics profile.
(2025-02-12 14:46:56.469) [info] [SimulationRunner.cc:296] Serving world SDF generation service on [/world/empty/generate_world_sdf]
(2025-02-12 14:46:56.469) [info] [ServerPrivate.cc:334] Serving world names on [/gazebo/worlds]
(2025-02-12 14:46:56.469) [info] [ServerPrivate.cc:347] Resource path add service on [/gazebo/resource_paths/add].
(2025-02-12 14:46:56.469) [info] [ServerPrivate.cc:360] Resource path get service on [/gazebo/resource_paths/get].
(2025-02-12 14:46:56.469) [info] [ServerPrivate.cc:375] Resource path resolve service on [/gazebo/resource_paths/resolve].
(2025-02-12 14:46:56.469) [info] [ServerPrivate.cc:389] Resource paths published on [/gazebo/resource_paths].
(2025-02-12 14:46:56.469) [info] [ServerPrivate.cc:402] Server control service on [/server_control].
(2025-02-12 14:46:56.469) [info] [SimulationRunner.cc:743] Found no publishers on /stats, adding root stats topic
(2025-02-12 14:46:56.469) [info] [SimulationRunner.cc:777] Found no publishers on /clock, adding root clock topic
(2025-02-12 14:46:56.471) [info] [SceneBroadcaster.cc:634] Serving scene information on [/world/empty/scene/info]
(2025-02-12 14:46:56.471) [info] [SceneBroadcaster.cc:643] Serving graph information on [/world/empty/scene/graph]
(2025-02-12 14:46:56.471) [info] [SceneBroadcaster.cc:654] Serving full state on [/world/empty/state]
(2025-02-12 14:46:56.471) [info] [SceneBroadcaster.cc:663] Serving full state (async) on [/world/empty/state_async]
(2025-02-12 14:46:56.471) [info] [SceneBroadcaster.cc:671] Publishing scene information on [/world/empty/scene/info]
(2025-02-12 14:46:56.471) [info] [SceneBroadcaster.cc:680] Publishing entity deletions on [/world/empty/scene/deletion]
(2025-02-12 14:46:56.471) [info] [SceneBroadcaster.cc:689] Publishing state changes on [/world/empty/state]
(2025-02-12 14:46:56.471) [info] [SceneBroadcaster.cc:700] Publishing pose messages on [/world/empty/pose/info]
(2025-02-12 14:46:56.471) [info] [SceneBroadcaster.cc:711] Publishing dynamic pose messages on [/world/empty/dynamic_pose/info]
(2025-02-12 14:46:56.713) [info] [Application.cc:766] [GUI] Added plugin [3D View] to main window
(2025-02-12 14:46:56.714) [info] [Application.cc:652] [GUI] Loaded plugin [MinimalScene] from path [/usr/lib/x86_64-linux-gnu/gz-gui-9/plugins/libMinimalScene.so]
(2025-02-12 14:46:56.717) [info] [EntityContextMenu.cc:134] [GUI] Currently tracking topic on [/gui/currently_tracked]
(2025-02-12 14:46:56.722) [warning] [Application.cc:933] [GUI] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:52:3: QML RenderWindowOverlay: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[... cut ...]
#5    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f5b8da7c9c9, in
#4    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f5b8da7acfb, in
#3    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f5b8da63676, in
#2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f5b8da027f2, in abort
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f5b8da1c475, in raise
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f5b8da709fc, in pthread_kill
Aborted (Signal sent by tkill() 68102 1000)
Escalating to SIGKILL on [Gazebo Sim Server]
```

j3anm1rc, 2025-02-12:
> Are you experiencing the same problem with other OpenGL-programs like `glxgears` ? If yes, probably the problem is not gz-sim but something in your WSL2 setup, see for example https://stackoverflow.com/questions/78987897/error-running-paraview-on-wsl-spirv-expand-step-registered-more-than-once .

I think it's not directly cause by wsl config. I did remove the docker -desktop from wsl list as your post suggest , nevertheless it didn't change anything.   
 
I ran also nvidia-smi from wls bash  that dont give any warning :
```bash
redgy@local:/mnt/c/Users/jm4$ nvidia-smi
Wed Feb 12 15:40:52 2025
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 570.86.09              Driver Version: 571.96         CUDA Version: 12.8     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA T600 Laptop GPU         On  |   00000000:01:00.0 Off |                  N/A |
| N/A   54C    P8              2W /   40W |       0MiB /   4096MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+

+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI              PID   Type   Process name                        GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|  No running processes found                                                             |
+-----------------------------------------------------------------------------------------+
```

I try to test 3d rendering with #<https://www.simplified.guide/linux/3d-acceleration-check>
And the result seems fine :
```bash
redgy@local:/mnt/c/Users/jm4$ glxinfo | grep "direct rendering"
direct rendering: Yes
redgy@local:/mnt/c/Users/jm4$ glxgears
508 frames in 5.0 seconds = 101.545 FPS
483 frames in 5.0 seconds = 96.569 FPS
223 frames in 5.6 seconds = 39.477 FPS
489 frames in 5.0 seconds = 97.699 FPS
529 frames in 5.0 seconds = 105.657 FPS
525 frames in 5.0 seconds = 104.937 FPS
X connection to :0 broken (explicit kill or server shutdown).
```

but indeed I suppose there is some inconsistency inside the WSL ubuntu packages.  I will try to remove all the package locally and reinstall gz from source .

j3anm1rc, 2025-02-12:
Eventually it turns out that gz sim -s and gs sim -g works , even if we don't have access to the quick start dialog this way.