gazebosim/gz-sim #2780: Kinematic tag does not affect model in sim

fnndyl, 2025-02-18:
## Environment
* OS Version: Ubuntu 22.04
* Source or binary build? Gazebo Harmonic 8.8.0

## Description
* Expected behavior: The kinematic tag should make an object able to interact (collide) with other physics objects, but be otherwise unaffected by forces/torques/etc. 
* Actual behavior: The kinematic tag seems to do nothing. An object given the kinematic tag still falls with gravity, for example.

## Steps to reproduce
Model SDF:
<?xml version="1.0" encoding="UTF-8"?>
<sdf version='1.9'>
  <model name='barge'>
    <static>true</static>
    
    <pose>1 0 -1 0 0 1.5708</pose>
    <link name='base'>
      
      <kinematic>true</kinematic>

      <visual name='base_visual'>
        <geometry>
          <box>
            <size>0.5 0.5 0.05</size>
          </box>
        </geometry>
        <material>
          <diffuse>1 1 1 1</diffuse>
          <specular>0.4 0.4 0.4 1</specular>
          <pbr>
            <metal>
              <albedo_map>model://arucotag/arucotag.png</albedo_map>
            </metal>
          </pbr>
        </material>
      </visual>
      <collision name='base_collision'>
        <geometry>
          <box>
            <size>0.5 0.5 0.05</size>
          </box>
        </geometry>
        <surface>
          <friction>
            <ode>
              <mu>1.0</mu>
              <mu2>1.0</mu2>
            </ode>
          </friction>
          <bounce>
            <restitution_coefficient>0.1</restitution_coefficient>
            <threshold>1.0</threshold>
          </bounce>
        </surface>
      </collision>
    </link>
  </model>
</sdf>

I can also provide the world's sdf file if that would be helpful.

jasonimercer, 2025-04-01:
Any updates on this bug? We are trying to port our Gazebo Classic sims over to gzSim and this issue is impacting us as well.

fnndyl, 2025-04-02:
Unfortunately I have not heard back on this issue - it might be that the kinematic tag is simply not supported in gzSim yet - I tried two other versions (Ionic and... Garden?) and neither had a functioning kinematic property.

iche033, 2025-04-02:
it's not supported yet in gz sim. There's initial effort on this in https://github.com/gazebosim/gz-physics/pull/618 but there's still more work that needs to be done