gazebosim/gz-sim #2801: INTEGRATION_mesh_inertia_calculation test is flaky on windows on gz-sim8

scpeters, 2025-02-28:
## Environment
* OS Version: Windows with conda
* Source or binary build? source builds of gz-sim8, first observed with d6b107b108693f24628cc98deabcf780786feeb5

## Description
* Expected behavior: Test should pass
* Actual behavior: Test fails

## Steps to reproduce

1. Run https://build.osrfoundation.org/view/gz-harmonic/job/gz_sim-8-clowin

## Output

### console output from build 10

* https://build.osrfoundation.org/view/gz-harmonic/job/gz_sim-8-clowin/10/testReport/junit/(root)/INTEGRATION_mesh_inertia_calculation/test_ran/

~~~
171: Test command: C:\J\workspace\gz_sim-8-clowin\ws\build\gz-sim8\bin\Release\INTEGRATION_mesh_inertia_calculation.exe "--gtest_output=xml:C:/J/workspace/gz_sim-8-clowin/ws/build/gz-sim8/test_results/INTEGRATION_mesh_inertia_calculation.xml"
171: Working Directory: C:/J/workspace/gz_sim-8-clowin/ws/build/gz-sim8/test/integration
171: Environment variables: 
171:  GZ_SIM_INSTALL_PREFIX=C:/J/workspace/gz_sim-8-clowin/ws/install/gz-sim8
171: Test timeout computed to be: 240
171: DynamicFactory(): Unable to parse descriptor set from [C:/J/workspace/gz_sim-8-clowin/ws/install/gz-msgs10\share\gz\protos\gz-msgs10.gz_desc]
171: Running main() from C:\J\workspace\gz_sim-8-clowin\ws\gz-sim\test\gtest_vendor\src\gtest_main.cc
171: [==========] Running 2 tests from 1 test suite.
171: [----------] Global test environment set-up.
171: [----------] 2 tests from MeshInertiaCalculationTest
171: [ RUN      ] MeshInertiaCalculationTest.CylinderColladaMeshInertiaCalculation
171/250 Test #171: INTEGRATION_mesh_inertia_calculation ...................Exit code 0xc0000374
***Exception:   3.01 sec
~~~

### console output from build 15

* https://build.osrfoundation.org/view/gz-harmonic/job/gz_sim-8-clowin/15/testReport/junit/(root)/INTEGRATION_mesh_inertia_calculation/test_ran/

~~~
171: Test command: C:\J\workspace\gz_sim-8-clowin\ws\build\gz-sim8\bin\Release\INTEGRATION_mesh_inertia_calculation.exe "--gtest_output=xml:C:/J/workspace/gz_sim-8-clowin/ws/build/gz-sim8/test_results/INTEGRATION_mesh_inertia_calculation.xml"
171: Working Directory: C:/J/workspace/gz_sim-8-clowin/ws/build/gz-sim8/test/integration
171: Environment variables: 
171:  GZ_SIM_INSTALL_PREFIX=C:/J/workspace/gz_sim-8-clowin/ws/install/gz-sim8
171: Test timeout computed to be: 240
171: DynamicFactory(): Unable to parse descriptor set from [C:/J/workspace/gz_sim-8-clowin/ws/install/gz-msgs10\share\gz\protos\gz-msgs10.gz_desc]
171: Running main() from C:\J\workspace\gz_sim-8-clowin\ws\gz-sim\test\gtest_vendor\src\gtest_main.cc
171: [==========] Running 2 tests from 1 test suite.
171: [----------] Global test environment set-up.
171: [----------] 2 tests from MeshInertiaCalculationTest
171: [ RUN      ] MeshInertiaCalculationTest.CylinderColladaMeshInertiaCalculation
171: [       OK ] MeshInertiaCalculationTest.CylinderColladaMeshInertiaCalculation (2824 ms)
171: [ RUN      ] MeshInertiaCalculationTest.CylinderColladaMeshWithNonCenterOriginInertiaCalculation
171/250 Test #171: INTEGRATION_mesh_inertia_calculation ...................Exit code 0xc0000374
***Exception:   4.02 sec
~~~

### console output from build 18

* https://build.osrfoundation.org/view/gz-harmonic/job/gz_sim-8-clowin/18/testReport/junit/(root)/INTEGRATION_mesh_inertia_calculation/test_ran/

~~~
171: Test command: C:\J\workspace\gz_sim-8-clowin\ws\build\gz-sim8\bin\Release\INTEGRATION_mesh_inertia_calculation.exe "--gtest_output=xml:C:/J/workspace/gz_sim-8-clowin/ws/build/gz-sim8/test_results/INTEGRATION_mesh_inertia_calculation.xml"
171: Working Directory: C:/J/workspace/gz_sim-8-clowin/ws/build/gz-sim8/test/integration
171: Environment variables: 
171:  GZ_SIM_INSTALL_PREFIX=C:/J/workspace/gz_sim-8-clowin/ws/install/gz-sim8
171: Test timeout computed to be: 240
171: DynamicFactory(): Unable to parse descriptor set from [C:/J/workspace/gz_sim-8-clowin/ws/install/gz-msgs10\share\gz\protos\gz-msgs10.gz_desc]
171: Running main() from C:\J\workspace\gz_sim-8-clowin\ws\src\gz-sim\test\gtest_vendor\src\gtest_main.cc
171: [==========] Running 3 tests from 1 test suite.
171: [----------] Global test environment set-up.
171: [----------] 3 tests from MeshInertiaCalculationTest
171: [ RUN      ] MeshInertiaCalculationTest.CylinderColladaMeshInertiaCalculation
171: [       OK ] MeshInertiaCalculationTest.CylinderColladaMeshInertiaCalculation (8136 ms)
171: [ RUN      ] MeshInertiaCalculationTest.CylinderColladaMeshWithNonCenterOriginInertiaCalculation
171/250 Test #171: INTEGRATION_mesh_inertia_calculation ...................Exit code 0xc0000374
***Exception:   8.87 sec
~~~

scpeters, 2025-02-28:
I haven't seen this fail on gz-sim9

* https://build.osrfoundation.org/view/gz-ionic/job/gz_sim-9-clowin/
* https://build.osrfoundation.org/view/gz-ionic/job/gz_sim-9-clowin/lastCompletedBuild/testReport/(root)/MeshInertiaCalculationTest/history/

iche033, 2025-03-04:
Disabled test on Windows in  #2799. Test is still flaky, and found that crash happens when exiting the test. So there seems to be an issue shutting down the server. The `ServerPrivate` destructor does get called though.