gazebosim/gz-sim #2803: Allow OdometryPublisher to work with non-fixed link

Morgantiz, 2025-02-28:
Allow the OdometryPublisher to publish odometry for a non-fixed link of the robot model. Introduce the possibility to publish the odometry of a point after a non-fixed joint. 

Solves StackExchange question:
https://robotics.stackexchange.com/questions/112266/how-to-get-the-global-position-of-a-link-in-gazebo-sim

Morgantiz, 2025-02-28:
Solution implemented in: https://github.com/Morgantiz/gz-sim/pull/1

iche033, 2025-03-14:
actually to resolve the original stack exchange question regarding getting pose of links, there is a PosePublisher system that can be used, example world here: https://github.com/gazebosim/gz-sim/blob/gz-sim9/examples/worlds/pose_publisher.sdf

Morgantiz, 2025-03-17:
Thank you very much for your reply and support. Using pose publisher is it also possible to publish the location of a fixed link that is removed within the gazebo model, but present in the URDF? I'm working on the odometry publisher because it allows you to consider a fixed xyz and rpy offset. Furthermore the odometry publisher computes the velocity with respect to the body frame.

iche033, 2025-03-18:
> Using pose publisher is it also possible to publish the location of a fixed link that is removed within the gazebo model, but present in the URDF?

Do you mean that a link connected to another link with 'fixed joint' is missing in gazebo? If that's the case, you can set the `< preserveFixedJoint>` tag in URDF (see this [page](http://sdformat.org/tutorials?tut=sdformat_urdf_extensions&cat=specification&#fixed-joint-lumping)) which preserves the link. The pose publisher system doesn't publish velocities so may not fit your use case.