gazebosim/gz-sim #2804: Camera Sensor Lens Orthographic not working

Jak253, 2025-02-28:
Hello,

I'm trying to use the camera sensor with the orthographic lens to generate a 2D map for navigation.
However, I can't change my lens type. It always stays stereographic.
Every value I write in the <type> field is ignored.

The relevant SDF Parts look like this:
`    
<?xml version="1.0" encoding="UTF-8"?>
<sdf version="1.7">
  <world name="warehouse">

    <physics type="ode">
      <max_step_size>0.003</max_step_size>
      <real_time_factor>1.0</real_time_factor>
    </physics>
    <plugin name="gz::sim::systems::Physics" filename="libgz-sim8-physics-system.so"/>
    <plugin name="gz::sim::systems::UserCommands" filename="libgz-sim8-user-commands-system.so"/>
    <plugin name="gz::sim::systems::SceneBroadcaster" filename="libgz-sim8-scene-broadcaster-system.so"/>
    <plugin name="gz::sim::systems::Sensors" filename="libgz-sim8-sensors-system.so">
      <render_engine>ogre2</render_engine>
    </plugin>

    <model name="My_Ortho_Cam">
      <pose>170 -200 400 0 1.57 0</pose>
      <static>true</static>
      <link name="camera_link">
        <sensor name="my_camera" type="camera">
          <topic>/map_camera</topic>
          <update_rate>1</update_rate>
	  <always_on>1</always_on>
          <visualize>true</visualize>
          <camera>
	    <save enabled="true">
		    <path>/home/ros/camera</path>
            </save>
            <image>
              <width>10000</width>
              <height>10000</height>
              <format>R8G8B8</format>
            </image>
            <clip>
              <near>0.1</near>
              <far>1000</far>
            </clip>
	    <lens>
              <type>orthographic</type>
	    </lens>
          </camera>
        </sensor>
      </link>
    </model>
`

I noticed that if I check my camera model with `gz model -l -m My_Ortho_Cam -s` all my properties are correctly set, except the lens type.
The output of said `gz model -l -m My_Ortho_Cam -s`
`
Requesting state for world [warehouse]...

- Link [5]
  - Name: camera_link
  - Parent: My_Ortho_Cam [4]
  - Mass (kg): 1.000000
  - Inertial Pose [ XYZ (m) ] [ RPY (rad) ]:
    [0.000000 0.000000 0.000000]
    [0.000000 -0.000000 0.000000]
  - Inertial Matrix (kg.m^2):
    [1.000000 0.000000 0.000000]
    [0.000000 1.000000 0.000000]
    [0.000000 0.000000 1.000000]
  - Pose [ XYZ (m) ] [ RPY (rad) ]:
    [0.000000 0.000000 0.000000]
    [0.000000 -0.000000 0.000000]
  - Sensor [6]
    - Name: my_camera
    - Parent: My_Ortho_Cam [4]
    - Pose [ XYZ (m) ] [ RPY (rad) ]:
      [0.000000 0.000000 0.000000]
      [0.000000 -0.000000 0.000000]
    - Horizontal field of view (rad): 1.047
    - Image width (px): 10000
    - Image height (px): 10000
    - Near clip (m): 0.1
    - Far clip (m): 1000
    - Pixel format: RGB_INT8
    - Save frames: 1
    - Save frames path: /home/ros/camera
    - Image noise:
      - Mean: 0
      - Bias mean: 0
      - Standard deviation: 0
      - Bias standard deviation: 0
      - Precision: 0
      - Dynamic bias standard deviation: 0
      - Dynamic bias correlation time (s): 0
    - Distortion K1: 0
    - Distortion K2: 0
    - Distortion K3: 0
    - Distortion P1: 0
    - Distortion P2: 0
    - Distortion center: 0.5 0.5
    - Lens type: stereographic
    - Lens scale to horizontal field of view (rad): 1
    - Lens C1: 1
    - Lens C2: 1
    - Lens C3: 0
    - Lens focal length (m): 1
    - Lens function: tan
    - Lens cutoff angle (rad): 1.5708
    - Lens texture size: 256
    - Lens intrinsics Fx: 277
    - Lens intrinsics Fy: 277
    - Lens intrinsics Cx: 160
    - Lens intrinsics Cy: 120
    - Lens intrinsics skew: 0
    - Visibility mask: 4294967295
  `

I'm unsure if it's because I did something wrong with the implementation of the SDF-file, or gazebo does not support different lens types, or the rendering engine does not support it.

Thanks for your help and input in advance.

iche033, 2025-03-05:
we currently don't support setting a camera to orthographic projection from the sdf. The `//lens/type` sdf tag is currently only used by wide angle cameras but not regular cameras.

To support this, I think we would need to parse the `//lens/type` SDF element in [CameraSensor::CreateCamera](https://github.com/gazebosim/gz-sensors/blob/6bfff2bd47916e5e4a937b6ffbd9e26fc74c05e1/src/CameraSensor.cc#L135) and set the projection type (by calling [camera->SetProjectionType](https://github.com/gazebosim/gz-rendering/blob/597d8892d7478b89f7f794c176f737578dd9c40a/include/gz/rendering/Camera.hh#L233))

Jak253, 2025-03-10:
Hello,

Thanks for the information!
I cloned the gz-sensor repository and made my changes. However, I can't get Gazebo to use my modified version.

Is there an easy way to integrate my changes in gz-sensor into Gazebo?

I have three options in mind, but I'm not sure which is the best or fastest one:
1) Clone gz-sensor, make changes, build it, and let Gazebo use my modified version instead of the default one.
2) Create a plugin, apply the changes there, and import the plugin in the SDF file.
3) Clone the entire gz-sim project along with all its submodules (gz-gui, gz-launch, etc.), modify gz-sensor, and rebuild Gazebo from scratch.
I would prefer the first option if it's feasible since it seems to be the simplest approach.
However, I couldn't find a guide on how to make Gazebo use a modified submodule like gz-sensor instead of the default version.

Could you maybe point me in the right direction?

iche033, 2025-03-10:
you should be able to go with 1 in a colcon workspace

Clone and build gz-sensors:

```bash
mkdir -p ~/colcon_ws/src
cd ~/colcon_ws/src
git clone git@github.com:gazebosim/gz-sensors.git

# make your local changes then build
cd ~/colcon_ws
colcon build --merge-install
```

Source the workspace and run gz

```bash
cd ~/colcon_ws
. install/setup.bash
gz sim -v 4 <your_sdf_world>
```