gazebosim/gz-sim #2805: Gazebo always loads with the same world

nino192, 2025-03-04:
## Environment
* OS Version: Ubuntu 24.04 Noble
* Source or binary build?
Gazebo Harmonic
* Graphics card (integrated): VGA compatible controller: Intel Corporation WhiskeyLake-U GT2 [UHD Graphics 620] (rev 02)
* NVIDIA driver: Driver Version: 550.120 
    - Rendering plugin: ogre2

## Description
* Expected behavior: Gazebo loads an empty world when empty is clicked and ran
* Actual behavior: Gazebo always loads "Tugbot in warehouse" world no matter what is selected in the GUI

## Steps to reproduce

1. Open gz sim and select "Tugbot in warehouse"
2. After running, save the world and exit gazebo
3. Run gz sim again and try selecting any other world from the GUI. Gazebo always loads "Tugbot in warehouse".

arjo129, 2025-03-04:
Have you tried restarting? It's possible that there was a server running in the background that has not died.

nino192, 2025-03-04:
After restarting the issue goes away.  Still, pretty weird that it behaved like this and I cannot reproduce it again.. Thanks

closed 2025-03-04 by nino192