gazebosim/gz-sim #2806: Consider warning users if there is already a Gazebo server running in the background

arjo129, 2025-03-04:
## Desired behavior
When users close the Gazebo GUI, the server can sometimes continue running in the background. This, combined with the ability to launch multiple Gazebo simulations, leads to confusion when a new gz sim instance is started. To improve usability, the Gazebo quick start screen should detect existing running servers and present them to the user before launching a new one. Additionally, enhancing the gz sim -g command to accept a world name argument would allow users to target specific running simulations.

## Alternatives considered

## Implementation suggestion

## Additional context
See #2805

GauravKumar9920, 2025-03-13:
Hi @arjo129, I have a potential code ready to fix this. But first, I would need to test if It works. How do I test it ?
I am using Mac and running ubuntu 24.04 through docker, which has gz installation script right in its image.

arjo129, 2025-03-13:
I recommend spinning up a second instance of gazebo sim in the background as a server only process.

```
gz sim -s some_world.sdf
```

Then spin up gz sim

```
gz sim
```

GauravKumar9920, 2025-03-13:
Hi @arjo129 , I'm using ubuntu 24.04(on a different machine(not docker)), and previously I had gz harmonic installed. 
for this issue I installed gazebo ionic(gz-sim9 branch) from source and tried removing gz harmonic.
but when I run `gz sim --version ` it shows -
`Gazebo Sim, version 8.9.0`

Is this the correct version, If I am not wrong it's the harmonic version right ?

arjo129, 2025-03-13:
I recommend following the source install instructions [here](
https://gazebosim.org/docs/latest/install_ubuntu_src/).

Make sure to source the correct directory and don't have a gazebo classic installation. You can use docker on ubuntu as well, personally I use [distrobox](https://distrobox.it/) to create isolated docker containers so that my host system is not tainted. However its up to you to choose what you'd like to use.

GauravKumar9920, 2025-03-17:
I am running into a issue where - 
I have the run `gz sim -s shapes.sdf `and `gz sim -s default.sdf` in seprate termianls. when I go ahead and run `gz sim -g shapes.sdf` why does the sometimes worldName printed is shapes and sometimes default ?

I’ve figured out why this is happening. Both servers are running at the same time, and the client (gz sim -g) chooses the server that responds first. currently I have added a functionality to accept -g "file name" as an argument and only launch the give file name(previously it didn't use to consider 'file name' as an argument.) though while doing so, it is sometimes able to launch the passed gui and sometimes it doesn't because There’s no way to tell the client which server to connect to when there are multiple instances active, so it sometimes connects to the “shapes” world and sometimes to the “default” one.

any ideas on how do I fix that ?

iche033, 2025-03-17:
> currently I have added a functionality to accept -g "file name" 

I think there is an edge case that if the user launches multiple gz sim servers with the same world name, e.g. shapes.sdf, they would still run into the same issue though

GauravKumar9920, 2025-03-18:
> > currently I have added a functionality to accept -g "file name"
> 
> I think there is an edge case that if the user launches multiple gz sim servers with the same world name, e.g. shapes.sdf, they would still run into the same issue though

Indeed, I had observed that. I believe that determining how to receive the appropriate server name feedback from the service call would be crucial in resolving this issue. I am currently looking into this matter. 

p.s - Please note that I have created a pull request to demonstrate the modifications I have made thus far. However, it is not currently in a stage where it can be merged.

arjo129, 2025-03-19:
The diff you've submitted is really big for a relatively small change. Couple of things I'd suggest are to breakdown this problem into smaller chunks. 

1. To deal with the case which @iche033 I recommend a really simple fix. First when a server [starts a world](https://github.com/gazebosim/gz-sim/blob/d9024d8aa1420aba56dc80ca46bd9456befdba36/src/SimulationRunner.cc#L100), I recommend using gz-transport to check if there are other worlds running. If there are print a `gzwarn`/`gzerr` saying "Another world of the same name is running".
2. On the client side, I don't recommend using file names but world names. It is possible to create worlds without files. TBF, I'd hold the work of on this till #2809 is complete. Or build on #2809. 
3. On the [startup screen](https://github.com/gazebosim/gz-sim/blob/d9024d8aa1420aba56dc80ca46bd9456befdba36/src/gui/QuickStartHandler.cc#L18) again using gz-transport query for the existence of worlds. If they exist consider injecting a banner that says: "Warning: we detected an already running world on start".

GauravKumar9920, 2025-03-19:
> Additionally, enhancing the gz sim -g command to accept a world name argument would allow users to target specific running simulations.

Thank you @arjo129 for clarifying, I was previously trying to fix the issues relating to gz-sim '-s' and '-g' commands, I will help with #2809 once done with this : )

Since the scope of this pr is just to warn the user of multiple running servers I will make the changes accordingly..

PR #2828 (2025-04-14): Detect running server on startup
# 🦟 Bug fix and New feature

Fixes #2806 

## Summary
Previously, when launching multiple GZ Sim servers, for instance, using the commands `gz sim -s shapes.sdf` and `gz sim -s default.sdf`,

If we had used the command `gz sim -g ‘fileName’`,
It would not have considered the `fileName` parameter and would have launched any of the servers at random. 

This pr introduces a new feature that allows you to - 
- Launch a file by passing the file name as an argument. For instance, you can utilize the command `gz sim -g ‘fileName’` to launch the GUI, provided that the  [... cut]