gazebosim/gz-sim #2824: Invisible button

EmmanuelMess, 2025-03-16:
## Environment
* OS Version: docker image ghcr.io/sloretz/ros:jazzy-simulation
* Source or binary build?
Gazebo Sim, version 8.8.0

    - Rendering plugin: ogre2
      - [ ] Sensor rendering error. 
      - [ ] GUI rendering error. 
    - EGL headless mode:
      - [ ] Running in EGL headless mode 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [x] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [x] running in Docker/Singularity
      - [x] GPU is concurrently used for other tasks
        - [x] desktop acceleration
        - [x] video decoding (i.e. a playing Youtube video)
        - [x] video encoding
        - [ ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [ ] multiple simulators running at the same time
      - [ ] other...
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
          ```bash
          LANG=C lspci -nn | grep VGA  # might require installing pciutils
          03:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Navi 23 [Radeon RX 6600/6600 XT/6600M] [1002:73ff] (rev c7)
          0b:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Picasso/Raven 2 [Radeon Vega Series / Radeon Vega Mobile Series] [1002:15d8] (rev c9)

          echo "$DISPLAY"
          :0

          LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
          _amdgpu_device_initialize: amdgpu_get_auth (1) failed (-1)
          amdgpu: amdgpu_device_initialize failed.
          glx: failed to create dri3 screen
          failed to load driver: radeonsi
          _amdgpu_device_initialize: amdgpu_get_auth (1) failed (-1)
          amdgpu: amdgpu_device_initialize failed.
          glx: failed to create dri3 screen
          failed to load driver: radeonsi
          direct rendering: Yes
              Preferred profile: core (0x1)
              Max core profile version: 4.5
              Max compat profile version: 4.5
              Max GLES1 profile version: 1.1
              Max GLES[23] profile version: 3.2
          OpenGL vendor string: Mesa
          OpenGL renderer string: llvmpipe (LLVM 19.1.1, 256 bits)
          OpenGL core profile version string: 4.5 (Core Profile) Mesa 24.2.8-1ubuntu1~24.04.1
          OpenGL core profile shading language version string: 4.50
          OpenGL core profile context flags: (none)
          OpenGL core profile profile mask: core profile
          OpenGL version string: 4.5 (Compatibility Profile) Mesa 24.2.8-1ubuntu1~24.04.1
          OpenGL shading language version string: 4.50
          OpenGL context flags: (none)
          OpenGL profile mask: compatibility profile
          OpenGL ES profile version string: OpenGL ES 3.2 Mesa 24.2.8-1ubuntu1~24.04.1
          OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20

          ps aux | grep Xorg
          root        1806  0.0  0.0   3956  2048 pts/0    S+   11:40   0:00 grep --color=auto Xorg

          sudo env LANG=C X -version  # if you don't have root access, try to tell the version of Xorg e.g. via package manager
          env: ‘X’: No such file or directory
          ```
      - On Windows, run `dxdiag` and report the GPU-related information.
      - On Mac OS, open a terminal and type `system_profiler SPDisplaysDataType`. Copy the output here.
        
    - [x] Please, attach the ogre.log or ogre2.log file from  `

<details>

[ogre2.log](https://github.com/user-attachments/files/19272557/ogre2.log)

</details>

## Description
* Expected behavior: Button has icon
* Actual behavior: Button doesn't have icon

## Steps to reproduce

Open new world.

## Output

[image]

iche033, 2025-03-20:
oh interesting, I couldn't reproduce it. Tested with gz-sim8 on jammy, and gz-sim9 on noble. No docker. 

Maybe someone else can also give it a try

[image]

EmmanuelMess, 2025-03-20:
> oh interesting, I couldn't reproduce it. Tested with gz-sim 8.9.0 on jammy, and gz-sim 9.1.0 on noble.
> 
> Maybe someone else can also give it a try
> <img alt="Image" width="441" src="https://private-user-images.githubusercontent.com/4000684/425142205-89ac203f-12db-4c9e-9c5b-1bc67bb55929.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDI1MDkwNDcsIm5iZiI6MTc0MjUwODc0NywicGF0aCI6Ii80MDAwNjg0LzQyNTE0MjIwNS04OWFjMjAzZi0xMmRiLTRjOWUtOWM1Yi0xYmM2N2JiNTU5MjkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDMyMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAzMjBUMjIxMjI3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZDAxMzg5ODRmYTc2NDNjOWQ1YzYxZTVjMGYzNGVkZDkyODU2NzJmNGY0Mjg4ODY0MTRkYjZlNmY2ZDc4NDJhNCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.mgZ5mJK-Zy3KJlU65uvjfLcCVpUYC9rlLYkio3FGCPU">

did you run it in a docker? I can replicate in two computers with the command `rocker --x11 --nvidia auto ghcr.io/sloretz/ros:jazzy-simulation`.