gazebosim/gz-sim #2868: Could not find the example of OpenRobotics.Quadcopter teleop

TZECHIN6, 2025-04-19:
I am studying to simulate a drone in gazebo using vel_cmd to control it. I found the example "Quadcopter teleop" in the thumbnails is what I want to achieve. Therefore I am searching for its source code for implementation.

The quadcopter.sdf (https://github.com/gazebosim/gz-sim/blob/ign-gazebo6/examples/worlds/quadcopter.sdf) only allow motor velocity, which not a direct control via vel_cmd like the one in teleop. And also the echoing the topic, seem its bugged out and no related topic is shown.

I am using Ubuntu22 with ign-gazebo-6.17.0

iche033, 2025-04-22:
This is the example world: https://app.gazebosim.org/OpenRobotics/fuel/worlds/Quadcopter%20teleop. You should also be able to find the downloaded world file in your fuel cache: `~/.gz/fuel/fuel.gazebosim.org/openrobotics/worlds/quadcopter\ teleop/1/world.sdf`

It uses the `MulticopterVelocityControl` plugin instead of `MulticopterMotorModel`.

TZECHIN6, 2025-04-22:
@iche033 Thanks buddy, I am using ignition, but still able find the path to the sdf file.

closed 2025-04-22 by TZECHIN6