gazebosim/gz-sim #2873: Gaazebo ignition

UavYuriyS, 2025-04-23:
When I run a world with meshes in WSL2, I experience memory leak in the `gz sim gui` process which can be tracked both from inside the VM with `htop` and from outside with the Task Manager

## Environment
* OS Version: WSL Ubuntu-22.04
* Source or binary build?
  * Binary build, `Gazebo Sim, version 8.9.0`, Harmonic
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
* There are warnings `[GUI] [Wrn] [ColladaLoader.cc:2395] Triangle input semantic: 'COLOR' is currently not supported` and `[GUI] [Wrn] [Ogre2Heightmap.cc:125] Heightmap final sampling should be 2^n`
- [ ] running on a dual GPU machine (integrated GPU + discrete GPU)
- [ ] running on a multi-GPU machine (it has multiple discrete GPUs)
- [ ] running on real hardware
- [x] running in virtual machine: WSLg
- [x] running in Docker/Singularity, No, but the issue persists in Doсker too.
- [ ] running remotely (e.g. via SSH)
- [ ] running in a cloud
- [ ] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
- [x] GPU is concurrently used for other tasks
  - [x] desktop acceleration
  - [x] video decoding (i.e. a playing Youtube video)
  - [ ] video encoding
  - [ ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
  - [ ] multiple simulators running at the same time
- [ ] other...
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
          ```bash
          LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'
          direct rendering: Yes
          Preferred profile: core (0x1)
          Max core profile version: 4.2
          Max compat profile version: 4.2
          Max GLES1 profile version: 1.1
          Max GLES[23] profile version: 3.1
          OpenGL vendor string: Microsoft Corporation
          OpenGL renderer string: D3D12 (NVIDIA GeForce RTX 4070 SUPER)
          OpenGL core profile version string: 4.2 (Core Profile) Mesa 23.2.1-1ubuntu3.1~22.04.3
          OpenGL core profile shading language version string: 4.20
          OpenGL core profile context flags: (none)
          OpenGL core profile profile mask: core profile
          OpenGL version string: 4.2 (Compatibility Profile) Mesa 23.2.1-1ubuntu3.1~22.04.3
          OpenGL shading language version string: 4.20
          OpenGL context flags: (none)
          OpenGL profile mask: compatibility profile
          OpenGL ES profile version string: OpenGL ES 3.1 Mesa 23.2.1-1ubuntu3.1~22.04.3
          OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.10

          kotik    3680518  0.0  0.0   4024  2032 pts/4    S+   15:38   0:00 grep --color=auto Xorg
          ```
      - On Windows, run `dxdiag` and report the GPU-related information.
      - On Mac OS, open a terminal and type `system_profiler SPDisplaysDataType`. Copy the output here.
        
    - [ ] Please, attach the ogre.log or ogre2.log file from  `~/.gz/rendering` 

## Description
* Expected behavior: Gazebo works and does not hog memory
* Actual behavior: Gazebo slowly eats all the memory, starts swapping and makes the machine unusable. Setting LIBGL_ALWAYS_SOFTWARE to 1 helps, but it is a no-solution because it disables GPU and makes everything excruciatingly slow

This is the world I'm using to test
[drone_scene_repro.txt](https://github.com/user-attachments/files/19868952/drone_scene_repro.txt)

Here is my ogre2 log
[ogre2.log](https://github.com/user-attachments/files/19868990/ogre2.log)

And dxdiag to boot
[DxDiag.txt](https://github.com/user-attachments/files/19869008/DxDiag.txt)

## Steps to reproduce

1. Work on a machine with WSL2 and NVIDIA GPU installed
2. Launch the world I provided with `LIBGL_ALWAYS_SOFTWARE=0 gz sim drone_scene_repro.txt -v`, observe memory leak.
3. Launch the world I provided with `LIBGL_ALWAYS_SOFTWARE=1 gz sim drone_scene_repro.txt -v`, and observe NO memory leak

## Output

traversaro, 2025-04-23:
If there the leak is only when using the Nvidia driver, could it be possible that the leak is in the Nvidia driver itself?

UavYuriyS, 2025-04-24:
> Perhaps you can try with a newer mesa version to get a newer version of d3d12?

Sure! 
I tried to install a newer mesa (25.0.4) from kisak-mesa, but it did not work well with WSLg—the renderer kept defaulting to llvmpipe. So I spun up a fresh Ubuntu-24.04, which has 24.2.8 preinstalled
```
LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'
direct rendering: Yes
    Preferred profile: core (0x1)
    Max core profile version: 4.6
    Max compat profile version: 4.6
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.1
    OpenGL vendor string: Microsoft Corporation
    OpenGL renderer string: D3D12 (NVIDIA GeForce RTX 4070 SUPER)
    OpenGL core profile version string: 4.6 (Core Profile) Mesa 24.2.8-1ubuntu1~24.04.1
    OpenGL core profile shading language version string: 4.60
    OpenGL core profile context flags: (none)
    OpenGL core profile profile mask: core profile
    OpenGL version string: 4.6 (Compatibility Profile) Mesa 24.2.8-1ubuntu1~24.04.1
    OpenGL shading language version string: 4.60
    OpenGL context flags: (none)
    OpenGL profile mask: compatibility profile
    OpenGL ES profile version string: OpenGL ES 3.1 Mesa 24.2.8-1ubuntu1~24.04.1
    OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.10
```

I then installed Gazebo Harmonic, and unfortunately, the leak is still there. 

I might as well try to install the latest Gazebo on 24.04 and see what happens

traversaro, 2025-04-24:
Thanks for the in-depth analysis. So I guess you also tried with an empty world, and this does not happen? 

In your world you have two models:

~~~xml
        <include>
            <uri>
                https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Fortress island
            </uri>
        </include>
            
        <include>
            <uri>
                https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Fortress heightmap
            </uri>
        </include>
~~~

Did you tried to only load one or another, and check if the behavior is still there?

UavYuriyS, 2025-04-24:
I did some more digging, removing various parts of the world here and there and that's what I found:

* The effect seems to be cumulative, so both models contribute to the leak but at different rates though. 
* The heightmap model which represents mountains around the island contributes a little bit, while the island itself contributes a lot
  * Island is at about 60 MB/s, heightmap is much smaller, around 0.6MB/s but it is there -- if left to its own devices it will eat up ram
  * Numbers are precise up to the order of magnitude, I just eyeballed them from the Task Manager
* Removing various parts of island model -- like chairs, benches and other stuff -- decreases the mem leak rate
* Removing textures greatly reduces the rate (2.65 MB/s with full world) but that's yeah, not the best solution 

I'll try with an empty world and upd my comment

UavYuriyS, 2025-04-25:
I've also tried to create a world with my own obj files manually; and it also leaks memory, unfortunately

zyeemf, 2026-10-03:
<html>
<body>
<h3>Additional quantitative data: the leak is d3d12-specific, scales with rendering load, and reproduces without any application code</h3><p>I can reproduce this on a very similar setup and would like to add numbers, plus a minimal
reproduction that involves no third-party application code at all.</p><p><strong>Environment</strong></p><ul><li>Windows 11 + WSL2 (WSLg), Ubuntu 24.04</li></ul><ul><li>GPU: NVIDIA RTX 5060 Laptop (plus an integrated GPU), <code>MESA_D3D12_DEFAULT_ADAPTER_NAME=NVIDIA</code></li></ul><ul><li>Gazebo Sim 8.x, Mesa <code>25.2.8-0ubuntu0.24.04.2</code></li></ul><p><strong>Method</strong></p><p>Every 30 s for 540 s I sampled <code>/proc/&lt;pid&gt;/smaps_rollup</code> (<code>Rss</code>, <code>Anonymous</code>, <code>Private_Dirty</code>) of
the <code>gz sim gui</code> and <code>gz sim server</code> processes, discarding the first 120 s as startup transient.
Growth was strictly linear in every run.</p>
Backend | Scene | gz sim gui | gz sim server | gui CPU
-- | -- | -- | -- | --
d3d12 | app scene, 0 models | +7.05 MB/min | +0.00 | ~1 core
d3d12 | app scene + water + 3 models | +12.07 MB/min | +0.00 | ~1 core
llvmpipe | app scene, 0 models | −0.23 MB/min (flat) | +0.00 | ~8 cores
d3d12 | pure gz world (no app code) | +15.47 MB/min | +0.00 | ~1.4 cores

<p>So the leak is <strong>still present on the newest Mesa</strong>, and slightly faster ⇒ it is not a regression
that a version bump fixes.</p><p>One observation that <strong>contradicts</strong> the earlier comment about kisak: on my machine Mesa 26.2.3
<strong>does</strong> drive d3d12 under WSLg (the gui process loads <code>libgallium-26.2.3</code>, d3d12 symbols are
present, <code>swrast</code> is absent — no llvmpipe fallback). The catch is that it only works once its build
dependencies are present (<code>libLLVM.so.21.1</code>, <code>libz3.so.4</code>, <code>libdisplay-info.so.1</code>); without them it
dies with <code>Could not initialize GLX</code>, which can easily look like "kisak is broken under WSLg".</p><p>I also tried the <strong>zink (GL→Vulkan) + dzn (Vulkan→D3D12)</strong> path, which would bypass the d3d12
Gallium driver entirely. It does <strong>not</strong> work under WSLg: zink needs an EGL <strong>DRI2/DRI3</strong> screen for
buffer sharing, and WSLg provides neither. Under X11 it fails with <code>DRI3 not available</code>; under
Wayland (<code>QT_QPA_PLATFORM=wayland</code>) dzn <em>is</em> loaded, but <code>egl: failed to create dri2 screen</code> is
followed by a SIGSEGV.</p><p><strong>Ask</strong></p><p>Could a maintainer confirm whether the d3d12 leak is known, and whether it is tracked upstream in
Mesa? So far d3d12 appears to be the <strong>only</strong> hardware-accelerated GL path available under WSLg
(llvmpipe is the software fallback and is 8–80× slower), so WSLg users currently have no way to get
both hardware acceleration and stable memory.</p><sc-clipboard block-data="[{&quot;blockId&quot;:&quot;aU0hWeTiIiv9zoZxY4DVVM&quot;,&quot;blockSubtree&quot;:{&quot;block&quot;:{&quot;aU0hWeTiIiv9zoZxY4DVVM&quot;:{&quot;role&quot;:&quot;editor&quot;,&quot;value&quot;:{&quot;id&quot;:&quot;aU0hWeTiIiv9zoZxY4DVVM&quot;,&quot;parentId&quot;:&quot;&quot;,&quot;type&quot;:&quot;header3&quot;,&quot;props&quot;:{&quot;title&quot;:[[&quot;Additional quantitative data: the leak is d3d12-specific, scales with rendering load, and reproduces without any application code&quot;,null]]},&quot;children&quot;:[],&quot;parentTable&quot;:&quot;block&quot;,&quot;enabled&quot;:true}}}}},{&quot;blockId&quot;:&quot;GLgQQBWduYRZMYwFzSj4vT&quot;,&quot;blockSubtree&quot;:{&quot;block&quot;:{&quot;GLgQQBWduYRZMYwFzSj4vT&quot;:{&quot;role&quot;:&quot;editor&quot;,&quot;value&quot;:{&quot;id&quot;:&quot;GLgQQBWduYRZMYwFzSj4vT&quot;,&quot;parentId&quot;:&quot;&quot;,&quot;type&quot;:&quot;text&quot;,&quot;props&quot;:{&quot;title&quot;:[[&quot;I can reproduce this on a very similar setup and would like to add numbers, plus a minimal\nreproduction that involves no third-party application code at all.&quot;,null]]},&quot;children&quot;:[],&quot;parentTable&quot;:&quot;block&quot;,&quot;enabled&quot;:true}}}}},{&quot;blockId&quot;:&quot;q5yap06WAr6IZ1PIJQnJf1&quot;,&quot;blockSubtree&quot;:{&quot;block&quot;:{&quot;q5yap06WAr6IZ1PIJQnJf1&quot;:{&quot;role&quot;:&quot;editor&quot;,&quot;value&quot;:{&quot;id&quot;:&quot;q5yap06WAr6IZ1PIJQnJf1&quot;,&quot;parentId&quot;:&quot;&quot;,&quot;type&quot;:&quot;text&quot;,&quot;props&quot;:{&quot;title&quot;:[[&quot;Environment&quot;,[[&quot;b&quot;]]]]},&quot;children&quot;:[],&quot;parentTable&quot;:&quot;block&quot;,&quot;enabled&quot;:true}}}}},{&quot;blockId&quot;:&quot;9qRaPv6O9h0ytzfkIMhjXZ&quot;,&quot;blockSubtree&quot;:{&quot;block&quot;:{&quot;9qRaPv6O9h0ytzfkIMhjXZ&quot;:{&quot;role&quot;:&quot;editor&quot;,&quot;value&quot;:{&quot;id&quot;:&quot;9qRaPv6O9h0ytzfkIMhjXZ&quot;,&quot;parentId&quot;:&quot;&quot;,&quot;type&quot;:&quot;bulleted_list&quot;,&quot;props&quot;:{&quot;title&quot;:[[&quot;Windows 11 + WSL2 (WSLg), Ubuntu 24.04&quot;,null]]},&quot;children&quot;:[],&quot;parentTable&quot;:&quot;block&quot;,&quot;enabled&quot;:true}}}}},{&quot;blockId&quot;:&quot;qTKfOACMCPpMZifDwvbs5v&quot;,&quot;blockSubtree&quot;:{&quot;block&quot;:{&quot;qTKfOACMCPpMZifDwvbs5v&quot;:{&quot;role&quot;:&quot;editor&quot;,&quot;value&quot;:{&quot;id&quot;:&quot;qTKfOACMCPpMZifDwvbs5v&quot;,&quot;parentId&quot;:&quot;&quot;,&quot;type&quot;:&quot;bulleted_list&quot;,&quot;props&quo [... cut ...] the d3d12 leak is known, and whether it is tracked upstream in\nMesa? So far d3d12 appears to be the &quot;,null],[&quot;only&quot;,[[&quot;b&quot;]]],[&quot; hardware-accelerated GL path available under WSLg\n(llvmpipe is the software fallback and is 8–80× slower), so WSLg users currently have no way to get\nboth hardware acceleration and stable memory.&quot;,null]]},&quot;children&quot;:[],&quot;parentTable&quot;:&quot;block&quot;,&quot;enabled&quot;:true}}}}}]"></sc-clipboard>
</body>
</html>