gazebosim/gz-sim #2876: Visual plugin tutorial

redvinaa, 2025-04-25:
## Desired behavior

I'm trying to migrate my robot model from Gazebo Classic, and I can't find a solution for altering visuals from a plugin. I have two similar problems:
1) Set color of visual from gazebo plugin. We use this to simulate led strips changing color on the simulated robot. In Classic this could also be achieved by using a light source and changing the colors of that, for which I also haven't found a solution.
2) Rendering textures dynamically on a visual. We used this to generate arbitrary Aruco markers by specifying marker ID in the SDF. The migration guide mentions that textures can be added as "albedo map" from SDF, which also hasn't worked for me (though that might be my mistake). It would be nice to set this texture from code.

## Alternatives considered

I could bake the aruco patterns into the models and generate a number of them. Likewise I could create led strip models with N different colors. Both sound like a horrible solution to a task that's supposed to be basic.

## Implementation suggestion

Could you please give some pointers on any of the above problems? I think it would also be useful for other community members to have a tutorial on this, as this is something extremely useful in robotics simulation.

## Additional context

iche033, 2025-04-25:
Option 1: Using a visual plugin:

you can take a look at the [shader_param](https://github.com/gazebosim/gz-sim/blob/gz-sim9/src/systems/shader_param/ShaderParam.cc) as an example.

Some notes:
* Rendering operations should be done in the rendering thread by connecting to [one of the render events](https://github.com/gazebosim/gz-sim/blob/db31641f7a07fb9d179994442d090477d9848396/src/systems/shader_param/ShaderParam.cc#L249-L251)
* Not very elegant method but [here](https://github.com/gazebosim/gz-sim/blob/db31641f7a07fb9d179994442d090477d9848396/src/systems/shader_param/ShaderParam.cc#L281-L306) is one way to get a pointer to the visual 
* See [rendering::Visual](https://github.com/gazebosim/gz-rendering/blob/gz-rendering9/include/gz/rendering/Visual.hh) and [rendering::Material](https://github.com/gazebosim/gz-rendering/blob/414735f8353d3ab8f09b84bf7bfce8c98472b3cb/include/gz/rendering/Material.hh#L226) classes for APIs related to setting material texture 

Option 2: Publishing material color msg 

There is a `/world/<world_name>/material_color` topic when a world is launched. You can published a message using gz-transport to change the material color of an entity
* Here is a [sample code](https://github.com/gazebosim/gz-sim/blob/3765579cfc017de813a5780c4dda2a39d979b6d4/test/integration/user_commands.cc#L1423-L1436) in our test

redvinaa, 2025-04-30:
Thank you for your answer. I managed to change colors from the plugin, but as mentioned [here](https://gazebosim.org/api/sim/8/rendering_plugins.html), it only changes either the server side (rendered camera image) or gui. Is there a way to change both from the same plugin? If not, what was the reasoning for this?

Using the above mentioned `material_color` topic it also works, and that changes both server and gui. In that, [this issue](https://github.com/gazebosim/ros_gz/issues/615) also helped. Is it implemented through [SceneBroadcaster](https://github.com/gazebosim/gz-sim/blob/6bbf5031cabb5b9652186e6791e6434181489a44/src/systems/scene_broadcaster/SceneBroadcaster.cc)? I tried but failed to understand that code, I'm not even sure it's relevant here.

What's weird for me is that both server and gui plugins get the scene object the same way, from the rendering engine, yet they change different things. I couldn't find a lot of info explaining this.

Regarding setting texture to an image, I'm also a bit stuck.
```
    gz::rendering::MaterialPtr material = scene_->CreateMaterial();
    material->SetTexture("model://docking_station/aruco0.png");
    visual_->SetMaterial(material);
```
This is what I have so far, but it doesn't seem to change anything. Am I missing something?

iche033, 2025-04-30:
> I managed to change colors from the plugin, but as mentioned [here](https://gazebosim.org/api/sim/8/rendering_plugins.html), it only changes either the server side (rendered camera image) or gui.

Is this from a visual plugin? As of https://github.com/gazebosim/gz-sim/pull/1275, the visual plugin should be loaded on both the server and gui side so it should affect both. For example, if you run `gz sim -v 4 shader_param.sdf`, you should see that the visual appearance changes in the gui camera view and also in the camera images.

> Using the above mentioned material_color topic it also works, and that changes both server and gui. In that, https://github.com/gazebosim/ros_gz/issues/615 also helped. Is it implemented through [SceneBroadcaster](https://github.com/gazebosim/gz-sim/blob/6bbf5031cabb5b9652186e6791e6434181489a44/src/systems/scene_broadcaster/SceneBroadcaster.cc)? I tried but failed to understand that code, I'm not even sure it's relevant here.

The command is received by the `user_command` system and most of the logic is in [this function](https://github.com/gazebosim/gz-sim/blob/6bbf5031cabb5b9652186e6791e6434181489a44/src/systems/user_commands/UserCommands.cc#L1865). In summary, it creates a `VisualCmd` component which is then processed by the [rendering code](https://github.com/gazebosim/gz-sim/blob/6bbf5031cabb5b9652186e6791e6434181489a44/src/rendering/RenderUtil.cc#L759) to set the visual material

> Regarding setting texture to an image, I'm also a bit stuck.

The code looks fine. Maybe it can't find the texture? If you call `SetDiffuse`, does it change color? Running gz sim with `-v 4` should give you more verbose output. The `~/.gz/rendering/ogre2.log` may also give some hints of rendering errors.

redvinaa, 2025-05-16:
In the end I managed to make changing colors work both ways. The problem was just a few misunderstandings, like that in the published topic, the world name should be set, or that the visual plugin only means that the plugin tag is placed within the visual tag in the sdf.

As for setting textures, I decided to use aruco models with the texture baked into the collada files. I gave up trying to set textures dynamically, as I didn't have any more time for it.

Thank you for the help!

azeey, 2025-11-24:
This is not quite answering your question, but if you're only interested in changing the material color, you can look at what's done in the `UserCommands` https://github.com/gazebosim/gz-sim/blob/b58df6c2e166836174801ecc6e4ce33bd45a45c9/src/systems/user_commands/UserCommands.cc#L1703-L1748

jasmeet0915, 2025-12-02:
@azeey A separate issue has been created #3207