gazebosim/gz-sim #2879: Contact Sensor seems to be not working correctly

Icon45, 2025-04-27:
Hi everyone,

i'm currently trying to implement a contact sensor for a robot, however when i run gz topic -e -t <correct topic name> i don't get a response, when the robot must cleary have contact with another object:

this is my implementation of the sensors:
```

     <!-- ========================================================= -->
      <!--           GAZEBO-KONTAKT-SENSOR für base_link             -->
      <!-- ========================================================= -->
      <gazebo reference="base_link">
        <sensor name="base_link_contact_sensor" type="contact">
          <always_on>1</always_on>
          <update_rate>50</update_rate>
          <visualize>true</visualize>
          <contact>
            <collision>collision</collision>
          </contact>
          <topic>/contacts/base_link</topic>
        </sensor>
      </gazebo>
      <gazebo reference="link_1_s">
        <sensor name="link_1_s_contact_sensor" type="contact">
          <always_on>1</always_on>
          <update_rate>50</update_rate>
          <contact>
            <collision>collision</collision>
          </contact>
          <topic>/contacts/link_1_s</topic>
        </sensor>
      </gazebo>
      <gazebo reference="link_2_l">
        <sensor name="link_2_l_contact_sensor" type="contact">
          <always_on>1</always_on>
          <update_rate>50</update_rate>
          <contact>
            <collision>collision</collision>
          </contact>
          <topic>/contacts/link_2_l</topic>
        </sensor>
      </gazebo>
      <gazebo reference="link_3_u">
        <sensor name="link_3_u_contact_sensor" type="contact">
          <always_on>1</always_on>
          <update_rate>50</update_rate>
          <contact>
            <collision>collision</collision>
          </contact>
          <topic>/contacts/link_3_u</topic>
        </sensor>
      </gazebo>
      <gazebo reference="link_4_r">
        <sensor name="link_4_r_contact_sensor" type="contact">
          <always_on>1</always_on>
          <update_rate>50</update_rate>
          <contact>
            <collision>collision</collision>
          </contact>
          <topic>/contacts/link_4_r</topic>
        </sensor>
      </gazebo>
      <gazebo reference="link_5_b">
        <sensor name="link_5_b_contact_sensor" type="contact">
          <always_on>1</always_on>
          <update_rate>50</update_rate>
          <contact>
            <collision>collision</collision>
          </contact>
          <topic>/contacts/link_5_b</topic>
        </sensor>
      </gazebo>
      <gazebo reference="link_6_t">
        <sensor name="link_6_t_contact_sensor" type="contact">
          <always_on>1</always_on>
          <update_rate>50</update_rate>
          <contact>
            <collision>collision</collision>
          </contact>
          <topic>/contacts/link_6_t</topic>
        </sensor>
      </gazebo>
      <gazebo reference="link_7_m">
        <sensor name="link_7_m_contact_sensor" type="contact">
          <always_on>1</always_on>
          <update_rate>50</update_rate>
          <contact>
            <collision>collision</collision>
          </contact>
          <topic>/contacts/link_7_m</topic>
        </sensor>
      </gazebo>
      <gazebo reference="tool0">
        <sensor name="tool0_contact_sensor" type="contact">
          <always_on>1</always_on>
          <update_rate>50</update_rate>
          <contact>
            <collision>collision</collision>
          </contact>
          <topic>/contacts/tool0</topic>
        </sensor>
      </gazebo>

```

Console output:

 ```
   parallels@ubuntu:~/Desktop/yhc10_ws$ gz topic -l | grep contact 
    /world/sortieranlage/model/yaskawa_hc10/link/base_link/sensor/base_link_contact_sensor/contact
    /world/sortieranlage/model/yaskawa_hc10/link/link_1_s/sensor/link_1_s_contact_sensor/contact
    /world/sortieranlage/model/yaskawa_hc10/link/link_2_l/sensor/link_2_l_contact_sensor/contact
    /world/sortieranlage/model/yaskawa_hc10/link/link_3_u/sensor/link_3_u_contact_sensor/contact
    /world/sortieranlage/model/yaskawa_hc10/link/link_4_r/sensor/link_4_r_contact_sensor/contact
    /world/sortieranlage/model/yaskawa_hc10/link/link_5_b/sensor/link_5_b_contact_sensor/contact
    /world/sortieranlage/model/yaskawa_hc10/link/link_6_t/sensor/link_6_t_contact_sensor/contact
    /world/sortieranlage/model/yaskawa_hc10/link/link_6_t/sensor/link_7_m_contact_sensor/contact
    /world/sortieranlage/model/yaskawa_hc10/link/link_6_t/sensor/tool0_contact_sensor/contact
    /world/sortieranlage/model/yaskawa_hc10/link/link_1/sensor/link_1_s_contact_sensor/contact
    /world/sortieranlage/model/yaskawa_hc10/link/link_2/sensor/link_2_l_contact_sensor/contact
    /world/sortieranlage/model/yaskawa_hc10/link/link_3/sensor/link_3_u_contact_sensor/contact
    /world/sortieranlage/model/yaskawa_hc10/link/link_4/sensor/link_4_r_contact_sensor/contact
    /world/sortieranlage/model/yaskawa_hc10/link/link_5/sensor/link_5_b_contact_sensor/contact
    /world/sortieranlage/model/yaskawa_hc10/link/link_6/sensor/link_6_t_contact_sensor/contact
    /world/sortieranlage/model/yaskawa_hc10/link/tool0/sensor/tool0_contact_sensor/contact
    parallels@ubuntu:~/Desktop/yhc10_ws$ gz topic -e -t /world/sortieranlage/model/yaskawa_hc10/link/tool0/sensor/tool0_contact_sensor/contact
```

iche033, 2025-04-28:
you can debug by turning on contact visualization. Go to top right menu, search for `Visualize Contact`, set a reasonable radius (for the blue contact visuals)  and enable `Show Contacts`. See if you are getting contacts on the robot. 

If there are indeed contacts, it could be that the value of the `<collision>` sdf element does not point to a valid collision. This could happen if the urdf ->sdf conversion process changes the name of the collision. Try saving the generated SDF file: Go to top left menu, select `Save world as`, save and inspect the SDF file to see what the SDF looks like for your robot.

DumbleDuck, 2025-09-29:
Did we solve the issue? I am currently facing the same problem.