gazebosim/gz-sim #2906: Restore Stack Traces in jetty

arjo129, 2025-05-16:
## Environment
* OS Version: Ubuntu 24.04
* Source: main
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: [ogre | ogre2]. 
      - [ ] Sensor rendering error. 
      - [ ] GUI rendering error. 
    - EGL headless mode:
      - [ ] Running in EGL headless mode 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [ ] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [ ] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [ ] running on real hardware
      - [ ] running in virtual machine
      - [ ] running in Docker/Singularity
      - [ ] running remotely (e.g. via SSH)
      - [ ] running in a cloud
      - [ ] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
      - [ ] GPU is concurrently used for other tasks
        - [ ] desktop acceleration
        - [ ] video decoding (i.e. a playing Youtube video)
        - [ ] video encoding
        - [ ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [ ] multiple simulators running at the same time
      - [ ] other...
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
          ```bash
          LANG=C lspci -nn | grep VGA  # might require installing pciutils
          echo "$DISPLAY"
          LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
          ps aux | grep Xorg
          sudo env LANG=C X -version  # if you don't have root access, try to tell the version of Xorg e.g. via package manager
          ```
      - On Windows, run `dxdiag` and report the GPU-related information.
      - On Mac OS, open a terminal and type `system_profiler SPDisplaysDataType`. Copy the output here.
        
    - [ ] Please, attach the ogre.log or ogre2.log file from  `~/.gz/rendering` 

## Description
We used to generate stack traces in older versions of gazebo using `backward.cpp`. These are crucial for debugging end user problems and save us developers a lot of time. In jetty it seems like we are not linking against backward correctly as I am getting a segfault without stack traces.

## Steps to reproduce

1. Run any bug that will segfault gazebo.

## Output

iche033, 2025-05-16:
I think it needs https://github.com/gazebosim/gz-sim/pull/2812 to be merged?

arjo129, 2025-05-16:
Nope. That needs to be merged for tests. This is happening when I call `gz sim`.

arjo129, 2025-05-20:
https://github.com/gazebosim/gz-sim/pull/2849/files#diff-b015f79ecbc65c367765bef669120629ddcc069576b2854009b494b0aa085d60

I think it has to do with this PR. We need to link backward to the `gz` executable.

iche033, 2025-05-20:
> https://github.com/gazebosim/gz-sim/pull/2849/files#diff-b015f79ecbc65c367765bef669120629ddcc069576b2854009b494b0aa085d60
> 
> I think it has to do with this PR. We need to link backward to the `gz sim` executable.

cc @j-rivero

j-rivero, 2025-05-21:
@arjo129 were the stack traces previously working on Jetty for the gz binaries?  I don't see any reference in the standalone PR  nor I remember that we have added it.

Anyway, @arjo129 could you test the branch https://github.com/gazebosim/gz-sim/tree/jrivero/backwards_standalone . I'm waiting to #2910 before creating the PR since it builds on top.

arjo129, 2025-06-03:
Addressed by #2919

PR #2919 (2025-05-30): backwards-cpp support for gz-sim-main and gz-sim-gui
# 🎉 New feature

Closes #2906 

## Summary

The PR adds support for backwards-cpp in gz-sim-main and gz-sim-gui. To do so:

 * Bump backwards vendor to 1.6 5913812bb6d382fa2c8e124f1a9aca9009221fe8
 * Relocate the backwards vendor directory outside of the root to be used by CLI 965d46370d963dcb59ccee282377aad9a3b6dbc5
 * Build binaries with backwards support fd0edc98262394385bd9a4440022f350a81c1136

## Checklist
- [x] Signed all commits for DCO
- [ ] Added tests
- [ ] Added example and/or tutorial
- [ ] Updated documentation (as needed)
- [ ] [... cut]
closed 2025-06-03 by arjo129