gazebosim/gz-sim #2918: gz-sim-model is broken with CLI11 < 2.5.0 (gz-sim abusing of option and flags)

j-rivero, 2025-05-21:
Coming from https://github.com/gazebosim/gz-sim/pull/2910#issuecomment-2896056931.

When using CLI11 on Ubuntu Noble (version 2.4.2) the system is not able to handle our CLI options groups properly so the `--list` option always requires `--model` option.

I think that the root of the problem is that we are abusing of "options" and "flags" in CLI11 where a more natural approach would be to use "subcommand". The problem is that subcommands in CLI11 do not accept dashes so we can not keep our current parameters. 

A subcommand possible implementation for gz-sim-model with two subcommands (list and model) would be:
https://gist.github.com/j-rivero/b4e332833440a85b463f66f9ee87524f 

There is a breakage in the code when mixing options and flags and make one of the flags to require one of the options. This is the case now with the option "--model" that is being required by "--pose" so when trying to use the CLI executable without parameters or with "--list" the result is:

"--pose requires --model".

A quick workaround would be to stop making the flag to require the option and improve the description to reflect that.

Seems like CLI11 2.5.0 is able to deal with the current scenario.

scpeters, 2025-05-21:
gz-sim-model also works properly with the cli11 version vendored in gz-utils, [version 1.9.1](https://github.com/gazebosim/gz-utils/blob/78da42891806366329df9a5fbb7548381b22f899/cli/include/vendored-cli/gz/utils/cli/Version.hpp#L14).

Version | Works
------- | -----
1.9.1  (gz-utils vendor) | ✅ yes
2.4.2  (24.04) | ❌ no
2.5.0  (brew) | ✅ yes

scpeters, 2025-05-21:
temporarily restoring cli11 vendoring in https://github.com/gazebosim/gz-utils/pull/178

scpeters, 2025-05-23:
I bisected the failure to https://github.com/CLIUtils/CLI11/commit/f7d26f26b2ca3e150d359d00bfde9b248bd3a7b8, which was first released in CLI v2.2.0

scpeters, 2025-05-23:
I bisected the bug fix to https://github.com/CLIUtils/CLI11/commit/f7600953d43b80c536dd3e722544a3076ed86e90

scpeters, 2025-05-27:
workaround ready for review in https://github.com/gazebosim/gz-sim/pull/2921, in conjunction with https://github.com/gazebosim/gz-utils/pull/179

PR #2921 (2025-05-29): gz-sim-model: workaround cli11 bug
# 🦟 Bug fix

Fixes #2918

## Summary

There is a bug in cli11 for versions >= 2.2 and < 2.5. This works around the bug for the affected versions, similar to the approached identified by @j-rivero in https://github.com/gazebosim/gz-sim/pull/2910#issuecomment-2898112530.

To test, I will run a colcon CI job on 24.04 (which has cli11 version 2.4.1) with vending of cli11 disabled by reverting https://github.com/gazebosim/gz-utils/pull/178.