gazebosim/gz-sim #2952: Segmentation fault on running gz sim -v 4 -s on MacOS M1

ArchakGoel, 2025-06-16:
## Environment
* OS Version: MacOS Sequoia 15.5 on M1
* Source or binary build?
 binary, harmonic via brew install

     - [o] Running on laptop directly.
    - Rendering system info:

      - On Mac OS, open a terminal and type `system_profiler SPDisplaysDataType`. Copy the output here.
        
      Chipset Model: Apple M2
      Type: GPU
      Bus: Built-In
      Total Number of Cores: 10
      Vendor: Apple (0x106b)
      Metal Support: Metal 3
      Displays:
        DELL P2715Q:
          Resolution: 3840 x 2160 (2160p/4K UHD 1 - Ultra High Definition)
          UI Looks like: 1920 x 1080 @ 30.00Hz
          Main Display: Yes
          Mirror: Off
          Online: Yes
          Rotation: Supported
        Color LCD:
          Display Type: Built-in Liquid Retina Display
          Resolution: 2880 x 1864 Retina
          Mirror: Off
          Online: Yes
          Automatically Adjust Brightness: Yes
          Connection Type: Internal

    - [ ] Please, attach the ogre.log or ogre2.log file from  `~/.gz/rendering` 

<details>

```
# paste log here
```

</details>

## Description
* Expected behavior: gazebo server should launch!
* Actual behavior: segmentation fault. 

## Steps to reproduce

1. Install on macos via brew
2. Run the gz sim -v 4 -s

## Output
[Msg] Gazebo Sim Server v8.9.0
Stack trace (most recent call last) in thread 6168637440:
#10   Object "libsystem_pthread.dylib", at 0x18e6dec0b, in _pthread_start + 135
#9    Object "libgz-transport13.13.4.1.dylib", at 0x101f8cef3, in void* std::__1::__thread_proxy[abi:ne180100]<std::__1::tuple<std::__1::unique_ptr<std::__1::__thread_struct, std::__1::default_delete<std::__1::__thread_struct>>, void (gz::transport::v13::Discovery<gz::transport::v13::ServicePublisher>::*)(), gz::transport::v13::Discovery<gz::transport::v13::ServicePublisher>*>>(void*) + 71
#8    Object "libgz-transport13.13.4.1.dylib", at 0x101f8b4bf, in gz::transport::v13::Discovery<gz::transport::v13::ServicePublisher>::RecvMessages() + 115
#7    Object "libgz-transport13.13.4.1.dylib", at 0x101f8b683, in gz::transport::v13::Discovery<gz::transport::v13::ServicePublisher>::RecvDiscoveryUpdate() + 371
#6    Object "libgz-transport13.13.4.1.dylib", at 0x101f8bd7f, in gz::transport::v13::Discovery<gz::transport::v13::ServicePublisher>::DispatchDiscoveryMsg(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, char*, unsigned short) + 83
#5    Object "libprotobuf.29.3.0.dylib", at 0x1056372b7, in google::protobuf::MessageLite::ParseFromArray(void const*, int) + 27
#4    Object "libprotobuf.29.3.0.dylib", at 0x10563580b, in bool google::protobuf::internal::MergeFromImpl<false>(std::__1::basic_string_view<char, std::__1::char_traits<char>>, google::protobuf::MessageLite*, google::protobuf::internal::TcParseTableBase const*, google::protobuf::MessageLite::ParseFlags) + 191
#3    Object "libprotobuf.29.3.0.dylib", at 0x1055eb133, in char const* google::protobuf::internal::TcParser::MpMessage<false>(google::protobuf::MessageLite*, char const*, google::protobuf::internal::ParseContext*, google::protobuf::internal::TcFieldData, google::protobuf::internal::TcParseTableBase const*, unsigned long long) + 463
#2    Object "libsystem_platform.dylib", at 0x18e718623, in _sigtramp + 55
#1    Object "libgz-tools2-backward.2.0.2.dylib", at 0x100d45643, in backward::SignalHandling::sig_handler(int, __siginfo*, void*) + 19
#0    Object "libgz-tools2-backward.2.0.2.dylib", at 0x100d456bf, in backward::SignalHandling::handleSignal(int, __siginfo*, void*) + 95

daveismith, 2025-07-05:
@ArchakGoel - I looked at your stack trace and it made me wonder if the bug was really in `gz-transport`. In looking over there, I found https://github.com/gazebosim/gz-transport/issues/602 which suggested rebuilding protobufs. Once I did that I was able to start both the server & gui apps successfully.

I do think that to close the ticket the appropriate bugs need to be logged to either gz-transport or the homebrew to get this working without a source rebuild of protobufs. In the meantime, it may be worth including this in the instructions for gazebo on Mac.

zhankins, 2025-07-19:
> [@ArchakGoel](https://github.com/ArchakGoel) - I looked at your stack trace and it made me wonder if the bug was really in `gz-transport`. In looking over there, I found [gazebosim/gz-transport#602](https://github.com/gazebosim/gz-transport/issues/602) which suggested rebuilding protobufs. Once I did that I was able to start both the server & gui apps successfully.
> 
> I do think that to close the ticket the appropriate bugs need to be logged to either gz-transport or the homebrew to get this working without a source rebuild of protobufs. In the meantime, it may be worth including this in the instructions for gazebo on Mac.

I had the same issue and your solution worked. I was able to rebuild protobufs and then it ran with no issues. To quote the solution there, the command to rebuild is:
```
brew reinstall protobuf --build-from-source --debug-symbols
```
Note: Homebrew raises a warning saying that build from source is not allowed and if you have a failure not to open an issue, but it will proceed as you requested.

On 2024 MacBook Pro with M4 Apple Silicon and Sequoia 15.5

Lanot, 2025-07-22:
> I had the same issue and your solution worked. I was able to rebuild protobufs and then it ran with no issues. To quote the solution there, the command to rebuild is:
>
> ```
> brew reinstall protobuf [...]

Thanks, it worked also for MacBook Pro M1 Apple Silicon with OSX Sequoia 15.5

scpeters, 2025-07-22:
> ```
> brew reinstall protobuf --build-from-source --debug-symbols
> ```

> [@scpeters](https://github.com/scpeters) do we need to rebuild bottles?

the solution people have reported is to rebuild protobuf from source with debug symbols, which is different than our normal bottle rebuild process. I don't know what is going on here; I need to find an ARM-based mac to test this with