gazebosim/gz-sim #2961: No waring for a bad reference link for sensor

Demonmasterlqx, 2025-07-05:
## Environment
* OS Version: 22.04
* Source or binary build?
  binary fortress

</details>

## Description
I found that the gazebo lack some warming for a woring reference link for senser when load a urdf robot

```

  <link name="lidar_sensor_link">
  </link>

  <joint name="lidar_sensor_joint" type="fixed">
    <parent link="dummy"/>
    <child link="lidar_sensor_link"/>
    <origin
      xyz="0 0 0.5"
      rpy="0 0 0" />
  </joint>

  <gazebo reference="lidar_sensor_joint">
    <plugin filename="ignition-gazebo-sensors-system"
            name="ignition::gazebo::systems::Sensors">
        <render_engine>ogre2</render_engine>
    </plugin>
    <sensor name='gpu_lidar' type='gpu_lidar'>
      <pose relative_to="lidar_sensor_link">0 0 0 0 0 0</pose>
      <topic>lidar</topic>
      <update_rate>10</update_rate>
        <ray>
          <scan>
              <horizontal>
              <samples>640</samples>
              <resolution>1</resolution>
              <min_angle>-1.396263</min_angle>
              <max_angle>1.396263</max_angle>
              </horizontal>
              <vertical>
              <samples>1</samples>
              <resolution>0.01</resolution>
              <min_angle>0</min_angle>
              <max_angle>0</max_angle>
              </vertical>
          </scan>
          <range>
              <min>0.08</min>
              <max>10.0</max>
              <resolution>0.01</resolution>
          </range>
        </ray>
        <alwaysOn>1</alwaysOn>
        <visualize>true</visualize>
    </sensor>
  </gazebo>
```

It's a part of my urdf file for gazebo sensors config. As you can see, the reference of sensor is wrong. But when I load the robot by run `` ign service -s /world/empty/create --reqtype ignition.msgs.EntityFactory --reptype ignition.msgs.Boolean --timeout 1000 --req 'sdf_filename:xxx, name: "" ' `` there is no warming with the sensor loaded fail.

## Steps to reproduce

1. run `` ign gazebo empty.sdf`` and start scence.
2. load robot
3. run `` ign gazebo topic --list``

## Output

Demonmasterlqx, 2025-07-05:
Thanks for help. By the way, I want to know where I can find these plugin name. I only find a website that tell me how to config urdf but no one tell how to write a right plugin name

arjo129, 2025-07-06:
The best docs we currently have is [this](https://gazebosim.org/api/sim/9/namespacegz_1_1sim_1_1systems.html). Its not great cause we do not have example system names and configs there. For examples your best bet is to look in [here](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds). We really should document and catalog all these somewhere.

Demonmasterlqx, 2025-07-06:
okk