gazebosim/gz-sim #2962: Consider spawning a scene broadcaster automatically once the gui is launched

arjo129, 2025-07-05:
## Description:
When new users add custom plugins or modify the default configuration in Gazebo, they may unintentionally end up in a state where the `SceneBroadcaster` is not running. As a result, the minimalscene view appears completely empty. This leads to a poor first-time experience, with many users concluding that "Gazebo is not working" when in fact the issue is a missing system plugin.

## Steps to Reproduce:

Create a custom Gazebo world with modified system plugins.
Launch the world without including `SceneBroadcaster`.
Open minimalscene or a remote viewer.

## Expected Behavior:
A warning or fallback should indicate that `SceneBroadcaster` is missing, or it should be enabled by default to avoid confusion.

## Actual Behavior:
The viewer opens with an empty scene, with no feedback explaining the issue.

## Impact:
High for new users — this can lead to significant confusion and the false impression that Gazebo is broken or improperly installed.

## Suggested Fix:
Consider:
- Including `SceneBroadcaster` by default in example and minimal configurations.
- Emitting a clear warning if the scene cannot be visualized due to missing broadcasters. 
- Possibly even automatically spawn the scene broadcaster plugin via the add plugin service.

arjo129, 2026-02-12:
I think since gz-sim9 this is no longer an issue. But in older versions it probably would be good to have.

arjo129, 2026-02-12:
My thought process is when we load the minimal scene system plugin from the gui we perform a check.

naitikpahwa18, 2026-02-21:
Hi @arjo129, I’ve implemented a simpler fix that auto-detects and injects `SceneBroadcaster` in `CreateEntities()` if it’s missing.
This avoids modifying its internals and keeps the change localized. Would appreciate your feedback on this approach.

arjo129, 2026-02-25:
I believe thats what #3334 was trying to do (rather than button we just  directly make the call).