gazebosim/gz-sim #3011: Null Pointer Dereference in KeyPublisher::eventFilter with Null QEvent Argument

XINJIANGMO, 2025-08-04:
## Environment
* OS Version: Ubuntu 24.04
* Source or binary build?
  Source,gz-sim9

## Description
* Expected behavior: gazebo did not crash
* Actual behavior: gazebo crashed

## Steps to reproduce

1. define a derived class to expose `eventFilter` in `KeyPublisher_TEST` 
```
class KeyPublisherTestable : public KeyPublisher
{
public:
  using KeyPublisher::eventFilter;
};
```
3. add a test in `KeyPublisher_TEST` 
```
/////////////////////////////////////////////////
TEST_F(KeyPublisherTest, EventFilterNullKeyEvent)
{
  QEvent *event = nullptr;
  bool result = this->plugin->eventFilter(this->win->QuickWindow(), event);
  EXPECT_FALSE(result);
}
```

## Output

```
(gdb) bt
#0  0x00007ffff7f8acb8 in QEvent::type (this=0x0) at /usr/include/x86_64-linux-gnu/qt5/QtCore/qcoreevent.h:301
#1  0x00007ffff7f89dc4 in gz::gui::plugins::KeyPublisher::eventFilter (this=0x555556420020, _obj=0x5555566c5340, _event=0x0)
    at /home/momo/w_s/src/gz-gui/src/plugins/key_publisher/KeyPublisher.cc:80
#2  0x00005555555c564a in KeyPublisherTest_EventFilterNullEvent_Test::TestBody (this=0x555555f733c0) at /home/momo/w_s/src/gz-gui/src/plugins/key_publisher/KeyPublisher_TEST.cc:144
#3  0x0000555555670645 in testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void>
    (object=0x555555f733c0, method=&virtual testing::Test::TestBody(), location=0x55555569ea6b "the test body") at /home/momo/w_s/src/gz-gui/test/gtest_vendor/src/gtest.cc:2599
#4  0x0000555555660136 in testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>
    (object=0x555555f733c0, method=&virtual testing::Test::TestBody(), location=0x55555569ea6b "the test body") at /home/momo/w_s/src/gz-gui/test/gtest_vendor/src/gtest.cc:2635
#5  0x000055555560ea5b in testing::Test::Run (this=0x555555f733c0) at /home/momo/w_s/src/gz-gui/test/gtest_vendor/src/gtest.cc:2674
#6  0x0000555555610340 in testing::TestInfo::Run (this=0x55555581d090) at /home/momo/w_s/src/gz-gui/test/gtest_vendor/src/gtest.cc:2853
#7  0x00005555556117c7 in testing::TestSuite::Run (this=0x5555557b47f0) at /home/momo/w_s/src/gz-gui/test/gtest_vendor/src/gtest.cc:3012
#8  0x0000555555635738 in testing::internal::UnitTestImpl::RunAllTests (this=0x555555818a50) at /home/momo/w_s/src/gz-gui/test/gtest_vendor/src/gtest.cc:5870
#9  0x00005555556722d3 in testing::internal::HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>
    (object=0x555555818a50, method=(bool (testing::internal::UnitTestImpl::*)(testing::internal::UnitTestImpl * const)) 0x555555634e42 <testing::internal::UnitTestImpl::RunAllTests()>, location=0x55555569f4e0 "auxiliary test code (environments or event listeners)") at /home/momo/w_s/src/gz-gui/test/gtest_vendor/src/gtest.cc:2599
#10 0x0000555555662619 in testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>
    (object=0x555555818a50, method=(bool (testing::internal::UnitTestImpl::*)(testing::internal::UnitTestImpl * const)) 0x555555634e42 <testing::internal::UnitTestImpl::RunAllTests()>, location=0x55555569f4e0 "auxiliary test code (environments or event listeners)") at /home/momo/w_s/src/gz-gui/test/gtest_vendor/src/gtest.cc:2635
#11 0x0000555555631e2d in testing::UnitTest::Run (this=0x55555571b780 <testing::UnitTest::GetInstance()::instance>) at /home/momo/w_s/src/gz-gui/test/gtest_vendor/src/gtest.cc:5444
#12 0x0000555555698c40 in RUN_ALL_TESTS () at /home/momo/w_s/src/gz-gui/test/gtest_vendor/include/gtest/gtest.h:2293

```

azeey, 2025-08-04:
The code that is triggering this is https://github.com/gazebosim/gz-gui/blob/e0c95585919d95a01fbf3af9a33c7fcd653ab154/src/plugins/key_publisher/KeyPublisher.cc#L78-L86

We can ofcourse add a nullcheck before line 80 there, but I understand correctly, these `eventFilter` functions are going to be called by Qt itself, so I'm not sure if its worth going through the entire codebase to add nullchecks.

azeey, 2025-08-05:
I'd say it's not worth the effort.